zephyrproject-rtos/zephyr #34667: posix_apis:mimxrt685_evk_cm33 timeout in test_posix_realtime

hakehuang, 2021-04-28:
**Describe the bug**
posix related cases hangs when run
test_posix_realtime

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt685_evk_cm33
3. make
4. make flash

**Expected behavior**
test pass

**Impact**
no clear.

**Logs and console output**
```
Running test suite posix_apis
===================================================================
START - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 &amp; priority 15
Thread 1 starting with scheduling policy 0 &amp; priority 15
Thread 2 starting with scheduling policy 0 &amp; priority 15
Bounce test OK
Barrier test OK
PASS - test_posix_pthread_execution in 0.718 seconds
===================================================================
START - test_posix_pthread_error_condition
PASS - test_posix_pthread_error_condition in 0.1 seconds
===================================================================
START - test_posix_pthread_termination
Thread 0 starting with a priority of 4
Cancelling thread 0
Thread 1 starting with a priority of 3
Cancelling thread 1
Thread 1 could not be cancelled
Thread 2 starting with a priority of 2
Cancelling thread 2
Thread 3 starting with a priority of 1
Cancelling thread 3
Thread 3 could not be cancelled
PASS - test_posix_pthread_termination in 1.14 seconds
===================================================================
START - test_posix_multiple_threads_single_key
Different threads set different values to same key:
thread 0: set value = 806886024 and retrieved value = 806886024
thread 1: set value = 806886248 and retrieved value = 806886248
PASS - test_posix_multiple_threads_single_key in 0.17 seconds
===================================================================
START - test_posix_single_thread_multiple_keys
Single thread associates its value with different keys:
key 0: set value = 806886448 and retrieved value = 806886448
key 1: set value = 806886448 and retrieved value = 806886448
PASS - test_posix_single_thread_multiple_keys in 0.17 seconds
===================================================================
START - test_posix_clock
POSIX clock APIs
POSIX clock APIs test done
PASS - test_posix_clock in 2.4 seconds
===================================================================
START - test_posix_semaphore
PASS - test_posix_semaphore in 5.1 seconds
===================================================================
START - test_posix_normal_mutex
mutex lock is taken
PASS - test_posix_normal_mutex in 0.102 seconds
===================================================================
START - test_posix_recursive_mutex
recrusive mutex lock is taken
PASS - test_posix_recursive_mutex in 0.3 seconds
===================================================================
START - test_posix_mqueue
PASS - test_posix_mqueue in 0.11 seconds
===================================================================
START - test_posix_realtime
POSIX clock set APIs
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-2637-ga3469a049785

hakehuang, 2021-04-28:
@MaureenHelm , @dleach02

mmahadevan108, 2021-05-10:
This was fixed as part of PR https://github.com/zephyrproject-rtos/zephyr/pull/34743

dleach02, 2021-05-10:
I confirm that it works...