zephyrproject-rtos/zephyr #34687: intel_adsp_cavs15: run tests/kernel/semaphore/semaphore/ failed on ADSP

KangJianX, 2021-04-29:
**Describe the bug**
run semaphore testcase failed.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/semaphore/semaphore/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/config,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Logs and console output**

> DEBUG   - DEVICE: START - test_sem_measure_timeout_from_thread
DEBUG   - DEVICE: PASS - test_sem_measure_timeout_from_thread in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sem_multiple_take_and_timeouts
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:1008: test_sem_multiple_take_and_timeouts: (timeout not equal to QSEC2MS(i + 1))
DEBUG   - DEVICE: timeout did not occur properly: -1107201456 != 250
DEBUG   - DEVICE: FAIL - test_sem_multiple_take_and_timeouts in 0.252 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sem_multi_take_timeout_diff_sem
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:973: sem_multiple_take_and_timeouts_helper: (diff_ticks >= timeout is false)
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:1038: sem_multi_take_timeout_diff_sem_helper: (diff_ticks >= timeout is false)
DEBUG   - DEVICE: time mismatch - expected at least 500, got 252time mismatch - expected at least 3000, got 0
DEBUG   - DEVICE:
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs15

**Environment (please complete the following information):**
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.3
Commit id: 9aa0f212ae

nashif, 2021-05-11:
clearer log message:

```
===================================================================
START - test_sem_multiple_take_and_timeouts

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:1010: test_sem_multiple_take_and_timeouts: (timeout not equal to QSEC2MS(i + 1))
timeout did not occur properly: -1107201456 != 250
 FAIL - test_sem_multiple_take_and_timeouts in 0.252 seconds
===================================================================
START - test_sem_multi_take_timeout_diff_sem

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:975: sem_multiple_take_and_timeouts_helper: (diff_ticks >= timeout is false)

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:1040: sem_multi_take_timeout_diff_sem_helper: (diff_ticks >= timeout is false)
time mismatch - expected at least 500, got 251time mismatch - expected at least 3000, got 0

```

nashif, 2021-05-11:
@enjiamai  this was your commit

andyross, 2021-05-11:
Yeah, we turned SMP on and exposed a bug in an SMP platform that isn't part of CI.  Not really that surprising I guess.  Hopefully a race in the test and new a kernel bug.  Will take a look.

KangJianX, 2021-05-12:
Double confirmed. This bug can be reproduced in commit 8e69dafccf25bc5b4015a607cdba114216c4ea0e.

PR #35306 (2021-05-14): tests: kernel: fix two semaphroe testcases failed on ADSP
Two testcases of semaphore failed in ADSP due to the timeout value
we got back from the child thread is invalid. We put the variable in
the bss instead of in a stack, trying to avoid this.

Fixes #34687
commit 05d8c6fc (2021-05-14): tests: kernel: fix two semaphroe testcases failed on ADSP

Two testcases of semaphore failed in ADSP due to the timeout value
we got back from the child thread is invalid. We put the variable in
the bss instead of in a stack, trying to avoid this.

Fixes #34687
closed 2021-05-14 by nashif