zephyrproject-rtos/zephyr #34706: RFC: API Change: flash_area

de-nordic, 2021-04-29:
### Problem description
Current flash_area structures contain much more fields than needed for Zephyr operation out of compatibility with mcuboot and rely on device_get_binding to get device name at compile time prior to any operation on flash.

### Proposed change

The proposed change is to replace flash_area access within mcuboot and other external projects with getter functions and make flash_area black-box object that is being operated by such getter functions and other flash_area functions without knowledge on internal structure of the flash_area object.
The proposed change should also maintain compatibility with previous solution to reduce impact on external code and reduce need for translation layer.

## Detailed RFC

The change should introduce flash_area_get_ functions that should provide access to flash_area objects that the Zephyr will be only source of, Fields that are not needed within zephyr will be removed and instead flash_area_get functions will generate proper data.
The flash_area_get interface should be compatible with mcuboot required flash_area interface.
The code within Zephyr should be able to directly access flash_area objects as internally the structure of object is known.

### Proposed change (Detailed)

Proposed change has been implemented as RFC:
 - for mcuboot https://github.com/mcu-tools/mcuboot/pull/982
 - for zephyr https://github.com/zephyrproject-rtos/zephyr/pull/34530

### Dependencies

Changes are dependent on agreement with mcuboot project for getter functions and flash_area public access API.

### Concerns and Unresolved Questions

List any concerns, unknowns, and generally unresolved questions etc.

## Alternatives

To rewrite flash_area as only mcyboot layer and add internal implementation of partitions but this will increase flash footprint and may impact mcuboot size.

Laczen, 2021-04-30:
@de-nordic, the aim of this change seems to be to reduce the rom size of a zephyr application that is not mcuboot. Could you add some figures showing the difference in rom size of an application using the old API and the new API?

de-nordic, 2021-06-23:
Currently the aim of this feature has changed to move flash map API to compatibility layer; the flash map  API has been taken into Zephyr, from other project, to quickly allow adaptation of some feature, so it carries a some characteristics that has been required by that project,
Since we are willing to rewrite it in Zephyr it is good moment to select something that will be more Zephyr-native and push the flash map API int compatibility layer, and deprecate it internally.
The compatibility layer will probably be pushed to mcuboot, with deprecation, and since the mcumgr will, probably, be taken under Zephyr tree, then it is possible to rewrite internal services to use the new APPI.

The current goal is to obtain feature replacement of the API, so that the work on compatibility layer could be started, and input on impact on external users could gathered and assessed as early as possible.

de-nordic, 2021-07-15:
List of available solutions:
 - ~~Rework of current flash map API internals https://github.com/zephyrproject-rtos/zephyr/pull/34530~~
 - MTD https://github.com/zephyrproject-rtos/zephyr/pull/34494
 - https://github.com/zephyrproject-rtos/zephyr/pull/36487
 - https://github.com/zephyrproject-rtos/zephyr/pull/36979

carlescufi, 2021-07-20:
@Laczen  and @de-nordic could we measure the footprint of the solutions proposed in [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/34706#issuecomment-880679465) to help us gather data and select an option?

mbolivar-nordic, 2021-07-20:
@carlescufi OK. I've been keeping quiet, but since you've asked, I will share my opinions.

Re: @de-nordic 's approach:

- con: device model usage
- pro: data structure

I don't think the value in making flash partitions into emulated devices outweighs the costs of misusing the device model. I think footprint data would be helpful, but I didn't get the impression at the latest dev-review discussion that we have really thought through the implications.

The data structures themselves look great. A partition is a pointer to a flash memory device, offset, and size, along with read/write/etc functions. That's exactly what I was hoping to see at the beginning of this, so I'm happy about that.

Re: @Laczen 's approach:

- con: trying to do too many things at once
- pro: clearer device model interaction

This approach avoids the issues with using the device model in a nonstandard way, which I like. But it's also adding a bunch of other features that I personally do not want to see in a first draft replacement for the flash map API.

In particular, making partitions a hierarchy right away feels like a bad idea.

To me, a flash partition should mean a partition of the sectors in the flash memory device in the mathematical sense of a [partition of a set](https://en.wikipedia.org/wiki/Partition_of_a_set#Definition). This means flash partitions should cover the entire device and be disjoint. This is a simple and straightforward starting out point we can refine later for well defined use cases.

We don't have this hierarchical data structure in the flash map API right now, though, and I think it should not be added yet.

I feel the same way about adding attributes like 'read_only'. Replacing flash_map is a big enough problem on its own that this feels like scope creep.

However, I think @carlescufi 's point from dev-review that since this is a contentious issue, the everyday users of the API should be able to decide among themselves, with @nvlsianpu as maintainer making the final call if necessary.

nvlsianpu, 2021-07-20:
@mbolivar Thanks for your opinion - It is really valuable to me. Regard @Laczen solution: to which one you was referencing to (#36487, #34494)?

Worth to notice that what I perceived as `+` (emulated partition device) you saw as the `-`, of course from other reason (I omitted) than I did.

@de-nordic @Laczen Is it possible to join pros of both solutions?

mbolivar-nordic, 2021-07-20:
> Regard @Laczen solution: to which one you was referencing to (#36487, #34494)?

#36979 -- that's the one included in the list of possible solutions in [de-nordic's comment](https://github.com/zephyrproject-rtos/zephyr/issues/34706#issuecomment-880679465) from 5 days ago. Did I misunderstand the status?

de-nordic, 2021-07-20:
> 
> #36979 -- that's the one included in the list of possible solutions in [de-nordic's comment](https://github.com/zephyrproject-rtos/zephyr/issues/34706#issuecomment-880679465) from 5 days ago. Did I misunderstand the status?

You did not misunderstood.

Laczen, 2021-07-20:
> Re: @Laczen 's approach:
> 
> * con: trying to do too many things at once
> * pro: clearer device model interaction
>

@mbolivar, thank you for sharing your opinions. I value your comments a lot. I understand your concerns and also agree with some of them.
 
> In particular, making partitions a hierarchy right away feels like a bad idea.
>
I was thinking you were referring to the older PR (mtd support), but from the comment I above I conclude you do not. In the proposal partitions are not a hierarchy, for present dts definitions they are exactly as flat as before. 
 
> To me, a flash partition should mean a partition of the sectors in the flash memory device in the mathematical sense of a [partition of a [...]
OK,  but this doesn't really seem in accordance with the fixed partition definition that is used as compatible, there is no mentioning that the whole device should be covered. I would definively make them disjoint, just to avoid problems. 

> We don't have this hierarchical data structure in the flash map API right now, though, and I think it should not be added yet.
> 
> I feel the same way about adding attributes like 'read_only'. Replacing flash_map is a big enough problem on its own that this feels like scope creep.
> 
The proposal is not only targeting the flash map API, it is targeting support for partitions. Regardless of what is said there was no real support for partitions in zephyr, As soon as support for partitions is added it was my idea that this should support allready defined properties (like read-only: see yaml) as well as subpartitions. This may seem like scope creep, but I wouldn't dare to propose a solution that provides support for working with partitions without these properties.

From your comments I would conclude that zephyr is using the fixed-partition compatible different from linux. This is OK, but it would be better not to use the name fixed-partition then. And to avoid confusion it would be best to document the difference.

Laczen, 2021-07-20:
from fixed-partitions.yaml in linux (examples section): both subpartitions and read-only are shown.

```
partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;
        partition@0 {
            label = "bootloader";
            reg = <0x000000 0x100000>;
            read-only;
        };
        firmware@100000 {
            compatible = "brcm,trx";
            label = "firmware";
            reg = <0x100000 0xe00000>;
        };
        calibration@f00000 {
            compatible = "fixed-partitions";
            label = "calibration";
            reg = <0xf00000 0x100000>;
            ranges = <0 0xf00000 0x100000>;
            #address-cells = <1>;
            #size-cells = <1>;
            partition@0 {
                label = "wifi0";
                reg = <0x000000 0x080000>;
            };
            partition@80000 {
                label = "wifi1";
                reg = <0x080000 0x080000>;
            };
        };
    };
```

Laczen, 2021-07-20:
> @mbolivar Thanks for your opinion - It is really valuable to me. Regard @Laczen solution: to which one you was referencing to (#36487, #34494)?
>
> Worth to notice that what I perceived as `+` [...]

@nvlsianpu, it is a 2 minute change to add pointers to the routines in the flp_struct. I have seen in the linux mtd-code that the mtd-info structure has `_write(), _read(),...` methods, but it is preferred not to use these directly, but instead use a `mtd_write(),...` method that calls these internal routines. Does zephyr use a similar approach?

nvlsianpu, 2021-07-21:
I missed the PR numbers. Thanks you for correction.

> The data structures themselves look great. A partition is a pointer to a flash memory device, offset, and size, along with read/write/etc functions. That's exactly what I was hoping to see at the beginning of this, so I'm happy about that.

@mbolivar-nordic So you think at this point new API will be implemented as a driver implemented for operating on flash-partition, and in the future we can extend it to operate on any else low-level memory driver partitions kind? In general this gives more flexibility in new feature development.
I really want to understand rationale you are thinking about.

> don't think the value in making flash partitions into emulated devices outweighs the costs of misusing the device model. I think footprint data would be helpful, but I didn't get the impression at the latest dev-review discussion that we have really thought through the implications.

One of implication which comes to me that this won't scale if nor-flash drivers API won't scale to support . And it won't.

@laczen #36979 looks to me very similar to what was done by you at the beginning under #34494. Nested partition looks like something we should add after, also read-only is similar think (isn't it a partition state party)

de-nordic, 2021-07-21:
> I don't think the value in making flash partitions into emulated devices outweighs the costs of misusing the device model.

I do not really understand what you mean by misusing, can you give some more details on the topic?
Any developer can define device at run-time, which is not even a hack just the structure use: I use this in  https://github.com/zephyrproject-rtos/zephyr/pull/36484 to implement API test, which does not use any specific device, only checks whether API calls are correctly passed to device.

> I think footprint data would be helpful, but I didn't get the impression at the latest dev-review discussion that we have really thought through the implications.

I will work to gather that stats. Regarding your insights on the  dev-review, you are right that there has not been any input on the implications, unfortunately, but the interest in topic has diverged from what it supposed to be to what extra features are needed.

Laczen, 2021-07-21:
> @Laczen #36979 looks to me very similar to what was done by you at the beginning under #34494. 

You are correct, only the "overruling" of sector size has been removed and nested partitions added. And some general clean-up to improve readability.

> Nested partition looks like something we should add after, also read-only is similar think (isn't it a partition state party)

Thank you for the remark on read-only, it could indeed be moved into the state part.

Regarding the addition of read-only, there are about 20 boards that use read-only in the partition definition and the yaml defines it. Why not add support for it now ?

Regarding nested partition support:  is the method wrong ? I really do not understand the problem with this.

nvlsianpu, 2021-07-21:
> Regarding nested partition support: is the method wrong ? I really do not understand the problem with this.

I think we should go via specification phase first, and then implement what will be specified. The method maybe is fine but I see this as separate topic. I have some thought on requirements which can be scribed.

> Regarding the addition of read-only, there are about 20 boards that use read-only in the partition definition and the yaml defines it. Why not add support for it now ?

This is one of concerns on your draft - It adds to much in comparison to what is expected at once. It might be prepare to be extended (and it is) so what you want, can be added easily after.

nandojve, 2021-07-21:
> I think we should go via specification phase first, and then implement what will be specified.

I'm missing a draw about this memory related things. For me, it has been difficult to understand the big picture and how everything will be connected. The draw could be good for the LTS documentation too and could help to talk about each point.

For instance,:

1) How internal flash, external flash (NOR, NAND), eMMC or even a future USB flash disk are/should be layered ?
2) How bootloader interact with above devices ? Is a unique interface ?
3) How all storage options interact with above topics ?

Than it will be easy to understand why flash layout is important , partitions , mount points etc. and what are the problems that you folks are trying to solve.

Is there such kind of documentation?

Sorry about my lack of information.

mbolivar-nordic, 2021-07-22:
@Laczen thanks for your reply! Hopefully you can help me clear up some confusion I'm having.

> I was thinking you were referring to the older PR (mtd support), but from the comment I above I conclude you do not. In the proposal partitions are not a hierarchy, for present dts definitions they are exactly as flat as before.

I am confused, sorry. I see this in the PR that I mentioned my comments were based on, #36979:

```c
/**
 * @brief flp_info_cfg represents the configuration of a partition. A partition
 * can be the child of a partition (then the parent pointer is defined and the
 * device pointer is NULL) or a child of a device (then the parent pointer is
 * NULL and the device pointer is defined).
 */
struct flp_info_cfg {
	const struct device *device;
	const struct flp_info *parent;
	const off_t off;
	const size_t size;
	const bool read_only;
};
```

This looks like a hierarchical representation to me; isn't it? Your comments later seem to confirm that.

>  OK, but this doesn't really seem in accordance with the fixed partition definition that is used as compatible, there is no mentioning that the whole device should be covered. I would definively make them disjoint, just to avoid problems.

Fair enough. Holes or gaps in the DT seem OK to me too -- as long as we avoid hierarchy and keep them disjoint, any missing sectors can be added to "imaginary" partitions of the sector set, that just aren't addressable via DT.

> From your comments I would conclude that zephyr is using the fixed-partition compatible different from linux. This is OK, but it would be better not to use the name fixed-partition then. And to avoid confusion it would be best to document the difference.

Thanks for this comment. I looked at https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/mtd/partitions/fixed-partitions.yaml again to see if we are really doing anything different in Zephyr.

I see somewhat contradictory information in there, though. The binding description says:

```
  The partition table should be a node named "partitions". Partitions are then
  defined as subnodes.
```

But to your point, there is one example where there is a partitions subnode that has some 'sub-partitions':

https://github.com/torvalds/linux/blob/3d5895cd351757f69c9a66fb5fc8cf19f454d773/Documentation/devicetree/bindings/mtd/partitions/fixed-partitions.yaml#L103

Those two pieces of information contradict each other, so it's not really clear to me which is "right".

Do you have some pointers to users in Linux that handle this hierarchy? Or is the example in the binding incorrect?

mbolivar-nordic, 2021-07-22:
@Laczen sorry I see you pasted the same example from the Linux binding already in a following comment -- my question about how that contradicts the description and which is right stands.

mbolivar-nordic, 2021-07-22:
@de-nordic 

> > I don't think the value in making flash partitions into emulated devices outweighs the costs of misusing the device model.
> 
> I do not really understand what you mean by misusing, can you give some more details on the topic?

Yeah, I think my comment here depends on some assumptions which might not be correct. I am particularly assuming that we are allowing people to create and destroy these at runtime.

> Any developer can define device at run-time, which is not even a hack just the structure use: I use this in #36484 to implement API test,

Are you sure about this? I don't understand clearly how run-time devices are meant to work when `CONFIG_USERSPACE=y` in particular. If we make a device at runtime, the hash tables used by the userspace system call verifiers don't know about it. So how can they check that it's a flash device? I looked at your PR but I don't understand that part.

mbolivar-nordic, 2021-07-22:
@nvlsianpu 

> @mbolivar-nordic So you think at this point new API will be implemented as a driver implemented for operating on flash-partition, and in the future we can extend it to operate on any else low-level [...]

What I understand is that the main goal is to replace the flash map API with something that is native to Zephyr. That is what I remember @de-nordic's original PR was trying to do way back at the beginning of this project -- put compile-time devices into the flash map.

At the beginning, during one of the dev-review meetings on the subject, I mentioned that I think all we really need to describe a flash area in zephyr is a pointer to the device, an offset, and a size. That seems like a minimal data representation that is "zephyr native", which does not use the flash ID numbers or things like that which are from mynewt. We can then map that to whatever mcuboot needs internally inside mcuboot itself.

So I'm not really thinking about MTD or other future extensions like that. I mostly just want to try to figure out what the smallest API is that solves this main problem. I still think that device + offset + size is all we really need and so that's why I'm glad to see @de-nordic was able to make that work.

de-nordic, 2021-07-22:
> Are you sure about this? I don't understand clearly how run-time devices are meant to work when `CONFIG_USERSPACE=y` in particular. If we make a device at runtime, the hash tables used by the [...]

That is great remark! I will check this, thanks for pointing this out.

de-nordic, 2021-07-22:
> What I understand is that the main goal is to replace the flash map API with something that is native to Zephyr. That is what I remember @de-nordic's original PR was trying to do way back at the beginning of this project -- put compile-time devices into the flash map.

That is correct.

> At the beginning, during one of the dev-review meetings on the subject, I mentioned that I think all we really need to describe a flash area in zephyr is a pointer to the device, an offset, and a [...]

That is correct. We have followed that path at the beginning, as an example here is definition https://github.com/zephyrproject-rtos/zephyr/blob/1ae189806f8787905b89526d23e4b2f152fb14ef/include/storage/flash_map.h#L56-L60 of the flash_area from the now defunct #34530.

At the same dev-review, or one of later ones, it has been agreed that the Flash Map API is a bad name, and confusing one, taking into account that it is mistakenly often called "Flash Area API", by the name of the structure. It has been agreed that the Flash Partition API is a better name, and it was already used by @Laczen proposal, which, mind some sneaky additions, was basically implementing the same solution. We have decided to give it a chance, as renaming Flash Map API to Flash Partition API would be a mess in the context of #34530 (now I think that it just could be done).

Now we are in spiral of a feature creep and a race of inventing "missing features", that would distinguish one solution for an other, and give it short advantage, till the other one catches up.
Wash, rinse, repeat.

Laczen, 2021-07-22:
> At the same dev-review, or one of later ones, it has been agreed that the Flash Map API is a bad name, and confusing one, taking into account that it is mistakenly often called "Flash Area API", by the name of the structure. It has been agreed that the Flash Partition API is a better name, and it was already used by @Laczen proposal, which, mind some sneaky additions, was basically implementing the same solution. We have decided to give it a chance, as renaming Flash Map API to Flash Partition API would be a mess in the context of #34530 (now I think that it just could be done).
> 
> Now we are in spiral of a feature creep and a race of inventing "missing features", that would distinguish one solution from an other, and give it a short advantage, till the other one catches up.
> Wash, rinse, repeat.

@de-nordic, will you please stop implying that I'm trying to do sneaky additions. From the start I have been stating that there are 2 users of partitions: the first ones are using the flash_map API and these seem to be the only ones you care about, the second ones are flash_disk, nvs, and possibly others. I am starting from the second ones, how can these be converted into using a API that uses the flash partition information that is in the dts.

What are the main reasons they are not using flash_map:
a. Because it always was a quite hidden api,
b. Because they where using ram for creating an array that is not used. Now this would be converted into rom, but this problem still stands.

So I started from the dts and the definition of the partition.yaml, what are the properties there that need to be a part of the structure that supports partitions (this is where the read-only comes from, nothing sneaky). The next thing I added (and later removed) was a possibility to "overrule" the erase-block-size used in a partition. This was added because it seemed a good idea to allow partitions to define there own "sector-size".

In defining it I have always kept in mind the present flash_map users and have made sure that there would be nothing blocking them, as it possible to put the structs describing the partitions in a new structure for flash_map there is no problem at all. And as the API is added as an extra, the conversion between flash_map and the new API can be done smoothly without the need to create on big PR that risks breaking everything.

After this it occured to me that there is a problem that we can add a partitions node under a partition node, so I added support for this as well. Nothing sneaky about it, just solving a potential future problem. I realized pretty soon that this also solves a existing problem: "one dts file is being used for several (hardware?) setups, a setup using a mcuboot setup and a setup that does not use mcuboot". Does the support for subpartitions come at a cost, yes it does, one extra pointer is added to distinguish between a partition that is a child of a flash device and a partition that is part of a partition. And in the case a subpartition is used the parent partition will also take up space. In the proposal there is no obligation to use it to solve the problem of one dts file for multiple setups, so when applied to the present dts the only cost is the addition of one (NULL) pointer per partition. And this possible disadvantage could be removed by adding a config setting "SUBPARTITION_SUPPORT".

Now let's look at you proposal, trying to reuse the flash API is a good idea, if this however comes at the cost of being forced to have all partitons taking up (rom) space as a device this creates exactly the same problem as flash_map. I washed and rinsed your solution only to find out it is still dirty.

Laczen, 2021-07-22:
@mbolivar, thank you. I am happy that you are looking at it starting from the binding definitions.

> @Laczen thanks for your reply! Hopefully you can help me clear up some confusion I'm having.
> I see somewhat contradictory information in there, though. The binding description says:
> 
> ```
>   The partition table should be a node named "partitions". Partitions are then
>   defined as subnodes.
> ```
I read this just as the partition table should be a node named "partitions", wether this is describing a partition table of a flash device or a partition is not specified. I don't think it is correct to interprete this as "THE" partition table, there are allready multiple partition tables in case multiple flash devices are used.

Regarding the hierarchy, what is done is converting the hierarchy defined in the dts into a hierarchy in the flash partition structs. This of course is somewhat fighting with the idea of a flattened devicetree. It could be possible (but probably very complex) to do the translation to a completely flat devicetree. I don't think the complete flattening is done for any of the devices (e.g. spi flash). The hierarchy is moved into the device definition. This is just using the same idea (altough not by defining a device). 
 
> Do you have some pointers to users in Linux that handle this hierarchy? Or is the example in the binding incorrect?

I think the example in the binding is correct. The mtd_info structure is using exactly the same method to distinguish between a master device and a parent device (I copied the idea from there).

nvlsianpu, 2021-07-22:
> I'm missing a draw about this memory related things. For me, it has been difficult to understand the big picture and how everything will be connected. The draw could be good for the LTS documentation too and could help to talk about each point.
...

@nandojve You didn't miss anything - big picture wasn't defined so fare. Probably we need to make a such.

nvlsianpu, 2021-07-22:
>What I understand is that the main goal is to replace the flash map API with something that is native to Zephyr. That is what I remember @de-nordic's original PR was trying to do way back at the beginning of this project -- put compile-time devices into the flash map.

At the beginning, during one of the dev-review meetings on the subject, I mentioned that I think all we really need to describe a flash area in zephyr is a pointer to the device, an offset, and a size. That seems like a minimal data representation that is "zephyr native", which does not use the flash ID numbers or things like that which are from mynewt. We can then map that to whatever mcuboot needs internally inside mcuboot itself.

So I'm not really thinking about MTD or other future extensions like that. I mostly just want to try to figure out what the smallest API is that solves this main problem. I still think that device + offset + size is all we really need and so that's why I'm glad to see @de-nordic was able to make that work.

@mbolivar @Laczen @de-nordic I remember this as well. Can we stick to simplicity now?

de-nordic, 2021-07-22:
@mbolivar
> 
> > Are you sure about this? I don't understand clearly how run-time devices are meant to work when `CONFIG_USERSPACE=y` in particular. If we make a device at runtime, the hash tables used by the userspace system call verifiers don't know about it.
> 
> That is great remark! I will check this, thanks for pointing this out.
> 

OK, so I have checked that, everything seem to work correctly. The verifiers find they way properly.

Laczen, 2021-07-22:
> At the beginning, during one of the dev-review meetings on the subject, I mentioned that I think all we really need to describe a flash area in zephyr is a pointer to the device, an offset, and a [...]

@nvlsianpu, OK as said I can live with a simpler definition of "flash area", but then live with the consequences and do not use the fixed-partitions compatible. Define what a flash area is, what the limitations are, make a compatible for it and change all dts's (Don't forget to remove the mcuboot partition in each of them because otherwise it is OK to write to it. And then don't forget to add it to compile mcuboot itself). Make sure a error is raised when someone creates a flash_area under a flash_area as this was intentionally made unsupported.

@nvlsianpu, and regarding simplicity exactly how much change is needed for each of the PR's ?

nvlsianpu, 2021-07-22:
> @nvlsianpu, OK as said I can live with a simpler definition of "flash area", but then live with the consequences and do not use the fixed-partitions compatible. Define what a flash area is, what the [...]

> @nvlsianpu, and regarding simplicity exactly how much change is needed for each of the PR's ?

You exaggerate a bit with these consequence.
DTS are in current shape from a while and we are still alive.
You might open PRs with features mentioned in whole discussion as "new ones" right after simple solution will be merged - I want to have this issue solved.

My <b>"decision"</b> for this RFC now:
I expect solution which will introduce the new API which will replace flash_area_xx API. Such api should be able to use build-time defined structures for describe fixed-partition we want to access (what is only needed discussed above), such structures should be provided, run time structures also should be allowed.

The fixed-partition might be defined on other memory technologies than only nor-flash so this shouldn't be extension only to flash driver implementation.
I also think that the flash drive API (both implementation API and user interface) should be kept simple as it can be. It shouldn't be extend to support all the memories - It implements RAW access to the memory which can be inherited (by usage) by a feature NAND flash driver etc. Need to have in mind that people are likely to do "magic stuff" with flash drivers (like panic memory to flash dump) etc. Zephyr-Rtos is the part of embedded devices word. Simplicity in adding a new implementation is also the value.

Therefor fixed-partition support on level of flash driver won't scale. Virtual flash partitions driver is a very elegant solution, I like it personally but consequence is that it does not fit to bigger picture.

Right now #36979 is closer to the expectations and might go if we come to the agreement.

I also think new API name should be decoupled form 'flash'. I would use `fixed_partition` or `fixed_mem_partition`.

Laczen, 2021-07-22:
@nvlsianpu, it is good that you have taken a decision about this.

The standards I set for myself do not allow me to create a fixed_partitions support solution without the proposed additions. I wish you all the luck with adding this to zephyr. Please do not copy any ideas or solution from #36979 in your solution.

nvlsianpu, 2021-07-22:
> The standards I set for myself do not allow me to create a fixed_partitions support solution without the proposed additions.

Ok. I don't care your personal standards - It's mater of your taste.

> I wish you all the luck with adding this to zephyr. Please do not copy any ideas or solution from #36979 in your solution.

You copied flash_area API approach and add a few new things - any new ideas in real. Also the PR code is under apache 2.0 license already. Pleas do not disturb project efforts if you don't want to cooperate. Forcing us to accept everything you want is not a way to fruitful cooperation. We are in this discussion since a while - Draw your conclusions, please.

nandojve, 2021-07-22:
Folks,

I don't want to be on this type of unproductive discussion. However, ...

>> I'm missing a draw about this memory related things. For me, it has been difficult to understand the big picture and how everything will be connected. The draw could be good for the LTS documentation too and could help to talk about each point.

> @nandojve You didn't miss anything - big picture wasn't defined so fare. Probably we need to make a such.

I don't have any problem to see a maintainer take a decision. This is one of many big motives why a maintainer is so important. However, based on the confirmation that we don't have a well defined goal on paper, it seems dictatorial ( red flag ) such a strong decision. That is not good for an open source project like Zephyr. This is the why it is so important have things on paper. This is why I'm asking about clear picture from what we have and what Zephyr desire. Then, when maintainer need to take a very difficult action ( decision ) it is supported by what project is looking for, not on individuals, not on group opinions or even blind interests.

I've been observing this area for a time and still have difficulty to understood all decisions. I pretty sure that this group is working hard to give best to project and I support group decisions. I've been seen it.  However, it is necessary after today a clear and concise documentation to avoid any future misunderstandings. With a clear and concise doc anyone can understand and propose changes, otherwise Zephyr frustrate people. I already saw a lot of persons give up because need accept something that no one can see and understood.

@Laczen and @nvlsianpu don't go to personal. It is clear that we should step back to desk to move forward. However, it is necessary take a difficult decision for LTS and the maintainer were obligated to do it.

Laczen, 2021-07-22:
@nandojve, I don't mind a dictatorial decision when it needs to  be taken. It is better to take a wrong decision than not to take a decision. A wrong decision can be corrected, no decision cannot be corrected.

You are talking about clear and concise documents, this is an exceptional case where we have the documentation we need. This documentation is the fixed-partition compatible (or partition.yaml). What is being implemented is support for this compatible. Altough there might be discussion to understand some of this specification (this is the discussion with @mbolivar-nordic), this is good because it brings knowledge and understanding to zephyr.

However, the decision that is taken is: we have a document, but we will not do what this document says. I cannot support such a decision.

nvlsianpu, 2021-07-23:
@nandojve @Laczen As I saw there are some doubts on fixed-partition implementation and documentation.

> However, the decision that is taken is: we have a document, but we will not do what this document says. I cannot support such a decision.

I just asked to do the implementation in a few iteration. I think this is usually way effective way. I never minded to tell that this documentation is bad, or what @Laczen propose is bad - I see that people want to discuss thing separately (at last @de-nordic 
 and @mbolivar ), otherwise we are stacking.

Can we discuss #36979 more. @de-nordic @mbolivar Are additional features (read-only property and sub-partition) possible to be considered.

Big picture: @nandojve Definitely we need to make architectural draft on how memory related things in zephyr should looks like, form low level flash drivers, SDHC driver, to block-device (common) API and higher if needed. It result might be a zephyr mtd common interface layer. I would drive such a initiative. I think we can start after summer vacation period (September?).

nandojve, 2021-07-23:
> Big picture: @nandojve Definitely we need to make architectural draft on how memory related things in zephyr should looks like, form low level flash drivers, SDHC driver, to block-device (common) [...]

This is great!

Laczen, 2021-07-27:
@mbolivar, I am reposting this as it might have been hurried in the message stream.

> @mbolivar, thank you. I am happy that you are looking at it starting from the binding definitions.
> 
> > @Laczen thanks for your reply! Hopefully you can help me clear up some confusion I'm having.
> > I see somewhat contradictory information in there, though. The binding description says:
> > ```
> >   The partition table should be a node named "partitions". Partitions are then
> >   defined as subnodes.
> > ```
> 
> I read this just as the partition table should be a node named "partitions", wether this is describing a partition table of a flash device or a partition is not specified. I don't think it is correct to interprete this as "THE" partition table, there are allready multiple partition tables in case multiple flash devices are used. I think the above should be read as: "The partition table of a node should be a node named "partitions". Partitions are then defined as subnodes.".
> 
> Regarding the hierarchy, what is done is converting the hierarchy defined in the dts into a hierarchy in the flash partition structs. This of course is somewhat fighting with the idea of a flattened devicetree. It could be possible (but probably very complex) to do the translation to a completely flat devicetree. I don't think the complete flattening is done for any of the devices (e.g. spi flash). The hierarchy is moved into the device definition. This is just using the same idea (altough not by defining a device).
> 
> > Do you have some pointers to users in Linux that handle this hierarchy? Or is the example in the binding incorrect?
> 
> I think the example in the binding is correct. The mtd_info structure is using exactly the same method to distinguish between a master device and a parent device (I copied the idea from there).

Laczen, 2021-07-27:
@de-nordic, #36979 now supports read-only partitions without doing any checks for writing to non-read-only partitions.

mbolivar-nordic, 2021-07-29:
> @mbolivar-nordic , I am reposting this as it might have been burried in the message stream.

I saw it, but this is actually part of my point:

> I read this just as the partition table should be a node named "partitions"

If you look closely at:

https://github.com/torvalds/linux/blob/4010a528219e01dd02e768b22168f7f0e78365ce/Documentation/devicetree/bindings/mtd/partitions/fixed-partitions.yaml#L103

Then you will see that the only example with hierarchy is using a node named `calibration@f00000`, not "partitions". That is part of what is contradictory about the text in the binding.

> The mtd_info structure is using exactly the same method to distinguish between a master device and a parent device (I copied the idea from there).

That's interesting to hear and I think something we could definitely evolve towards over time. But I still think that replacing the existing API with something more zephyr native is a big enough problem to solve on its own, before introducing hierarchy in the C structures representing partitions themselves. You may be totally right that hierarchy is needed for something else, but my personal opinion is this is trying to do too much at once.

Laczen, 2021-07-29:
> > @mbolivar-nordic , I am reposting this as it might have been burried in the message stream.
> 
> I saw it, but this is actually part of my point:
> 
> > I read this just as the partition table should be a node named "partitions"
> 
> If you look closely at:
> 
> https://github.com/torvalds/linux/blob/4010a528219e01dd02e768b22168f7f0e78365ce/Documentation/devicetree/bindings/mtd/partitions/fixed-partitions.yaml#L103
> 
> Then you will see that the only example with hierarchy is using a node named `calibration@f00000`, not "partitions". That is the main contradictory text in the binding. "Should" is not "must", but it's not clear to me just from the binding what is "right".

You are right, to be completely according to the definition the node calibration@f00000 should contain a node partitions, and it does not.

I will remove the hierarchy from the structure for now.

henrikbrixandersen, 2022-01-24:
I will remove the "area: CAN" label from this issue as it is strictly not CAN related. The CANopenNode module integration code uses the flash map internally, but we should be able to migrate that to a new flash map API once ready.