zephyrproject-rtos/zephyr #34707: drivers/flash: extend API by write/erase with flash content check and re-write

nvlsianpu, 2021-04-29:
Usually flash drivers/controllers doesn't check whether what was requested to be  write was written. As flash driver API implements just write, it doesn't mean that corruption of written data will be detected as error by the driver.

Most of FS and storage system implementation assumes that write/erase operations are reliable (and they usually are).
If corrupted write or corrupted erase occurred then this is not visible for the storage layer. So storage need to read and check the content o its own, which actually is common problem. Also If a flash driver (or flash controller under it) already does the check, the read operation will be just a waste of time. Looks like suitable will be to move that functionality to layer bellow.

**solution part 1**
<b>Write with check</b> and <b>erase with check</b>  functions API can be introduced.
For flash drivers without write/erase verification build-in this API should read back procedures for check the content.
For drivers which already support checks under write API almost nothing needs to be done. This especially preferred for flash controller with check build-in.

flash driver glue layer might be able to decide which way is the case for the driver and works accordingly basing of a new callbacks fields in API table structures provided by shims.

Current `flash_write()` API shouldn't be extended by write check silently - read operation might be time costly as for seral flashes are.

**solution part 2**
<b>Write with check and rewrite on corruption attempt</b> function API
This would be extension to the <b>Write with check </b> API which perform another try of data write if flash content doesn't match (and the flash controller allows rewrites to same location).

nvlsianpu, 2021-04-29:
cc @Laczen @d3zd3z @de-nordic @anangl @carlescufi 
This idea came with review of my draft PR for fixing issue which looks like results of silently unwritten flash.
See the comment: https://github.com/zephyrproject-rtos/zephyr/pull/33947#issuecomment-822227983

Laczen, 2021-04-29:
@nvlsianpu, maybe I was a bit too quick in saying that this belongs in the flash driver. It seems that littlefs does a check of a write even if the flash driver provides a write error return.

nvlsianpu, 2021-04-29:
^^ that's nothing. It is RFC for query what can be done and what can't.

thedjnK, 2021-05-04:
My comment on this is that it's a nice idea but not something all or even most devices are going to require so there should be an option to enable it which should be off by default for the reason that this will bloat the flash size and execution speed (in the case of the nRF52840 with a QSPI chip which is already slow in comparison to the internal flash, each write will now be half the speed and even worse than that would be a full erase which takes upwards of 2 minutes on an MX25R6435F - this is then going to require the MCU to read out the whole 8MB contents of the QSPI chip to ensure it's all erased, that is going to cause a huge detrimental impact to time required). This would also not be relevant to all modules, e.g. STM32's where they have the checksum in the flash.

nvlsianpu, 2021-05-04:
^^ @thedjnK I'm are aware of all limitation. I'm taking them into account. Thanks for input!.

ycsin, 2021-12-02:
The SPI flash erase/write with SPI_ASYNC APIs sometimes fails silently, this seems like something that can discover the failure as it happens.

d3zd3z, 2021-12-07:
There are several devices where both of these operations can really only be done within the driver (and aren't necessarily done by write/erase themselves). For example, there may be an erase check command that adjusts voltages to distinguish erase vs decayed (or partially written) data. Or a readback may use ECC to verify the data.

So, I think this could be useful, but a little challenge since we would need to get into the APIs for the different platforms. I suppose one way forward would be to add MCUboot implementations (that just read data), and gradually add platform-specific support for the more advanced versions.