zephyrproject-rtos/zephyr #34732: stm32h747i_disco: Wrong Power supply setting LDO

ABOSTM, 2021-04-30:
**Describe the bug**
Clock_control driver for STM32H7 set LDO as powersupply source.
https://github.com/zephyrproject-rtos/zephyr/blob/29c838ed1dc9aa2520051b670d387b3d4d02187c/drivers/clock_control/clock_stm32_ll_h7.c#L262
https://github.com/zephyrproject-rtos/zephyr/blob/29c838ed1dc9aa2520051b670d387b3d4d02187c/drivers/clock_control/clock_stm32_ll_h7.c#L278

But on stm32h747i_disco board, both power supply, SMPS or LDO, are supported.
And SMPS is the default hardware configuration (can be changed by some solder bridge).
See https://www.st.com/resource/en/user_manual/dm00504240-discovery-kit-with-stm32h747xi-mcu-stmicroelectronics.pdf
chapter **5.2.3 SMPS/LDO power supply**

As a consequence, after running zephyr firmware on this board out of the box,
a deadlock occurs ( see chapter 5.2.3 SMPS/LDO power supply):
> Caution: A deadlock occurs if the board SMPS/LDO firmware PWR configuration does not match the
> hardware configuration: after the reset, the ST-LINK cannot connect the target anymore

Thankfully a method is explained to recover the situation.
But driver should be changed to avoid entering this deadlock situation.

affrinpinhero-2356, 2021-05-20:
HI @ABOSTM,

I havnt seen any deadlock with stm32h747i_disco board. Test methods was as follows. 

1. Used GDB to put breakpoint in all function that called LL_PWR_ConfigSupply. And I have observed LDO settings were called second. For reference I have pasted debugger log below:
```
(gdb) b prepare_regulator_voltage_scale
Breakpoint 1 at 0x8001318: file /home/myown/zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_ll_pwr.h, line 1000.
(gdb) b optimize_regulator_voltage_scale
Breakpoint 2 at 0x8001490: file /home/myown/zephyrproject/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_ll_pwr.h, line 1000.
(gdb) b stm32h7_init
Breakpoint 3 at 0x8000dbc: file ../include/arch/arm/aarch32/asm_inline_gcc.h, line 56.
(gdb) c
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 3, stm32h7_init (arg=0x0) at ../include/arch/arm/aarch32/asm_inline_gcc.h:56
56              __asm__ volatile(
(gdb) c
Continuing.
Info : halted: PC: 0x08000dbe
halted: PC: 0x08000dbe

Breakpoint 1, prepare_regulator_voltage_scale () at /home/myown/zephyrproject/zephyr/drivers/clock_control/clock_stm32_ll_h7.c:262
262             LL_PWR_ConfigSupply(LL_PWR_LDO_SUPPLY);
(gdb) c
Continuing.
Info : halted: PC: 0x0800131a
halted: PC: 0x0800131a

Breakpoint 2, optimize_regulator_voltage_scale (sysclk_freq=400000000) at /home/myown/zephyrproject/zephyr/drivers/clock_control/clock_stm32_ll_h7.c:278
278             LL_PWR_ConfigSupply(LL_PWR_LDO_SUPPLY);
(gdb) q
A debugging session is active.

        Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /home/myown/zephyrproject/zephyr/build/zephyr/zephyr.elf, Remote target

```
2. I have comented LL_PWR_ConfigSupply(LL_PWR_LDO_SUPPLY); were ever it was called and run.

ABOSTM, 2021-05-20:
OK, I don't know exactly how to reproduce, I was excepting that your Test methods should be sufficient.
Nevertheless I think there is still a latent issue, as behavior is described in board's user manual,
And there is still an incoherence between LDO setting in `clock_stm32_ll_h7.c` and SMPS setting in `soc_m7.c`
Also, as the board can support both LDO and SMPS, it could be interesting to have the capability to select which one to use from software point of view (with Kconfig for example, or DT).
So I removed 2.6.0 milestone, and we could come back on this issue issue later

erwango, 2021-11-22:
@ABOSTM Can it be closed, now that  #39555 is merged ?

ABOSTM, 2021-11-22:
No, as said in #39555 

> This is for sure a good step, but we still need to define this switch in board stm32h747i_disco

So this current PR is now aim to define the switch for  board stm32h747i_disco

PR #41105 (2021-12-25): boards: arm: stm32h7: select direct SMPS for both disco boards
Similarly to #41087,  configure direct SMPS for both STM32H7 disco boards

Direct SMPS is the default configuration out of the box of:
* stm32h474i_disco
* stm32h735g_disco

Fixes #34732
commit 607a40e7 (2021-12-25): boards: arm: stm32h7: select direct SMPS for both disco boards

Direct SMPS is the default configuration out of the box of:
* stm32h474i_disco
* stm32h735g_disco

Fixes #34732
commit 956bb90b (2022-01-06): boards: arm: stm32h7: select direct SMPS for both disco boards

Direct SMPS is the default configuration out of the box of:
* stm32h474i_disco
* stm32h735g_disco

Fixes #34732
closed 2021-12-25 by cfriedt