zephyrproject-rtos/zephyr #34737: Can't compile CIVETWEB with CONFIG_NO_OPTIMIZATIONS & CONFIG_DEBUG

mglettig, 2021-04-30:
**Describe the bug**
When setting either CONFIG_NO_OPTIMIZATIONS =y or CONFIG_DEBUG=y CIVETWEB server has compile errors. The compile errors say that the POSIX functions ferror and fileno were undefined. These are POSIX functions that are not supported by Zephyr according to the documentation.

**To Reproduce**
Steps to reproduce the behavior:
Build https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/civetweb/http_server

**Expected behavior**
No compile error.

**Impact**
Due to this bug I can't debug other issues I observed with civetweb. --> Showstopper.

**Logs and console output**
```
/opt/toolchains/gcc-arm-none-eabi-8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: ../modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `push_inner':
/opt/zephyrproject/modules/lib/civetweb/src/civetweb.c:6267: undefined reference to `ferror'
/opt/toolchains/gcc-arm-none-eabi-8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: ../modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `pull_inner':
/opt/zephyrproject/modules/lib/civetweb/src/civetweb.c:6426: undefined reference to `fileno'
collect2: error: ld returned 1 exit status
make[3]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:130: zephyr/zephyr_prebuilt.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:3009: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2934: zephyr/CMakeFiles/zephyr_final.dir/rule] Error 2
make: *** [Makefile:385: zephyr_final] Error 2
```

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr 2.5.0 using civetweb e6903b80c09d17cd1a8bb32e40080005558aad29

**Additional context**

Nukersson, 2021-05-01:
I think, we have either to wait for Implementation or you can just create stubs for this functions.

de-nordic, 2021-05-19:
@carlescufi, @ioannisg  According to this bug description

>  The compile errors say that the POSIX functions ferror and fileno were undefined. These are POSIX functions that are not supported by Zephyr according to the documentation.

Which means that compilation fails not due to bug but rather lack of feature.
I think this should be reclassified as feature request or enhancement.

mglettig, 2021-05-21:
@de-nordic Sorry but I don't agree. No longer being able to set `CONFIG_DEBUG=y` in my application when using CIVETWEB is a no-go. In my opinion the least that should be done is adding the missing functions to libc_extensions in the civetweb sample.

de-nordic, 2021-05-21:
> 
> (...) No longer being able to set `CONFIG_DEBUG=y` in my application when using CIVETWEB is a no-go. 

So it used to work? If it used to work previously then it is bug indeed.

de-nordic, 2021-05-21:
Actually, I have focused on missing function but the fact that turning on debugging breaks sample is a bug. Sorry @mglettig, you are right.

MaureenHelm, 2021-05-26:
> > (...) No longer being able to set `CONFIG_DEBUG=y` in my application when using CIVETWEB is a no-go.
> 
> So it used to work? If it used to work previously then it is bug indeed.

When did this work before (which version tag or SHA)? Can you try a git bisect to determine when it broke?

Nukersson, 2021-06-02:
> @carlescufi, @ioannisg According to this bug description
> 
> > The compile errors say that the POSIX functions ferror and fileno were undefined. These are POSIX functions that are not supported by Zephyr according to the documentation.
> 
> Which means that compilation fails not due to a bug but rather lack of feature.
> I think this should be reclassified as feature request or enhancement.

I agree with it. It is done.

ibirnbaum, 2021-08-12:
I'm able to reproduce this issue on the MIMXRT1050-EVK board when building the civetweb HTTP server sample project.

To me, it looks like a lucky coincidence that the binary can acutally be built using the default project settings, based on compiler optimization: the build *only* works when building with the default compiler optimization settings, optimize for size. I wanted to build the project unoptimized for single-step debugging, but ran into the unresolved symbol problem as described above, which also applied to the other optimization presets except optimize for size.

Even with optimize for size set, the problem can be enforced by, for example, removing the static qualifier from the function push_inner in civetweb.c, which references ferror() if the FILE* parameter of the function is NULL. In that case, the now global function creates an external reference which remains unresolved. There's no complaint regarding fileno though, which is not referenced by this particular function. 

This all seemed a bit odd, so I did a full disassembly of the working size-optimized code using objdump. Based on this, it looks like the respective branches where ferror and fileno are referenced are being optimized out! The attached file contains the relevant code from the function push_inner, which starts at line 6208 in civetweb.c. Below the code, you'll find the disassembly of this particular section.

The disassembly shows that push_inner has been inlined, which is not too surprising. Code-wise, it adds up with the presence of the first instruction after the variable declarations (line 6226, if (timeout > 0)), and the if(ssl) statement from line 6237 is also present. Surprisingly, we're jumping directly to line 6274 from there, and therefore directly into the else-branch of the if-statement in line 6265. Therefore, if the if-branch isn't present, there's also impossible to ever end up calling ferror (line 6267).

So how can the compiler be sure that the size of those few lines of code can be saved? How can the compiler be sure that fp will always be NULL? Here's my guess: push_inner is called by push_all. push_all receives fp as a parameter, doesn't check its value and just passes it on. There's a total of 4 calls to push_all in the civetweb code, 3 of which pass a constant NULL as parameter value for fp. The only case where push_all is called with an actual variable as fp is from forward_body_data. Yet, this function already exits prior to calling push_all if fp is not NULL. Therefore, this function will also only ever call push_all with fp=NULL,

Assuming a high level of inlining being performed when using the optimize for size setting, the compiler might have been clever enough to recognize the circumstance that fp will always be NULL, thus ferror will never be called. I won't dig further into what's the deal with fileno or other occurrences of ferror, but I guess that it's always going to be the same phenomenon.

So from my point of view, being able to build at only a certain optimization level which happens to optimize out the open ends is a bug. I'd like to be able to build this sample without any optimization.

[civetweb_code_vs_dump.txt](https://github.com/zephyrproject-rtos/zephyr/files/6976166/civetweb_code_vs_dump.txt)

ibirnbaum, 2021-12-30:
This issue is still pending (at least for the fileno() function, as described in my post from Aug 12th) - please remove the 'stale' tag.

closed 2022-03-16 by github-actions[bot]