zephyrproject-rtos/zephyr #34739: tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading fails to build on a number of platforms

galak, 2021-04-30:
`./scripts/twister -s tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading -l`

Gets:

```
ERROR   - segger_trb_stm32f407      tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/segger_trb_stm32f407/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:    7/ 276   2%  skipped:   83, failed:    1
ERROR   - serpente                  tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/serpente/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   28/ 276  10%  skipped:   85, failed:    2
ERROR   - ruuvi_ruuvitag            tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/ruuvi_ruuvitag/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   29/ 276  10%  skipped:   85, failed:    3
ERROR   - thingy52_nrf52832         tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/thingy52_nrf52832/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   30/ 276  10%  skipped:   85, failed:    4
ERROR   - sam4l_ek                  tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/sam4l_ek/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   49/ 276  17%  skipped:   89, failed:    5
ERROR   - pinetime_devkit0          tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/pinetime_devkit0/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   57/ 276  20%  skipped:   89, failed:    6
ERROR   - nrf9160_innblue22         tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nrf9160_innblue22/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   63/ 276  22%  skipped:   89, failed:    7
ERROR   - nrf9160_innblue21         tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nrf9160_innblue21/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   88/ 276  31%  skipped:   92, failed:    8
ERROR   - lpcxpresso11u68           tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/lpcxpresso11u68/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   91/ 276  32%  skipped:   93, failed:    9
ERROR   - faze                      tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/faze/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   92/ 276  33%  skipped:   93, failed:   10
ERROR   - mec1501modular_assy6885   tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mec1501modular_assy6885/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   95/ 276  34%  skipped:   93, failed:   11
ERROR   - mimxrt1010_evk            tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1010_evk/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   96/ 276  34%  skipped:   93, failed:   12
ERROR   - mimxrt1024_evk            tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mimxrt1024_evk/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   97/ 276  35%  skipped:   93, failed:   13
ERROR   - mm_swiftio                tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/mm_swiftio/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:   98/ 276  35%  skipped:   93, failed:   14
ERROR   - google_kukui              tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/google_kukui/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  104/ 276  37%  skipped:   93, failed:   15
ERROR   - frdm_k82f                 tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/frdm_k82f/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  109/ 276  39%  skipped:   93, failed:   16
ERROR   - ip_k66f                   tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/ip_k66f/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  113/ 276  40%  skipped:   93, failed:   17
ERROR   - decawave_dwm1001_dev      tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/decawave_dwm1001_dev/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  121/ 276  43%  skipped:   95, failed:   18
ERROR   - bcm958402m2_m7            tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/bcm958402m2_m7/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  125/ 276  45%  skipped:   95, failed:   19
ERROR   - bt510                     tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/bt510/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  129/ 276  46%  skipped:   95, failed:   20
ERROR   - bl653_dvk                 tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/bl653_dvk/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  130/ 276  47%  skipped:   95, failed:   21
ERROR   - bbc_microbit_v2           tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/bbc_microbit_v2/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
[... 67 lines cut ...]
ERROR   - frdm_kw41z                tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/frdm_kw41z/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  267/ 276  96%  skipped:   97, failed:   45
ERROR   - bbc_microbit              tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/bbc_microbit/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  269/ 276  97%  skipped:   97, failed:   46
ERROR   - 96b_carbon_nrf51          tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/96b_carbon_nrf51/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/build.log
INFO    - Total complete:  276/ 276  100%  skipped:   97, failed:   47
INFO    - 212 of 259 test configurations passed (81.85%), 47 failed, 97 skipped with 0 warnings in 287.93 seconds
```

logs look like:
```
[ 93%] Linking C executable zephyr_prebuilt.elf
[... cut ...]
/home/galak/pr-336/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld: libzephyr.a(rtt_console.c.obj): in function `k_sleep':
/home/galak/git/zephyr/twister-out/segger_trb_stm32f407/tests/arch/arm/arm_no_multithreading/arch.arm.no_multithreading/zephyr/include/generated/syscalls/kernel.h:86: undefined reference to `z_impl_k_sleep'
collect2: error: ld returned 1 exit status
gmake[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:120: zephyr/zephyr_prebuilt.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2914: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
```

ioannisg, 2021-04-30:
I wonder how this was not caught by CI @nordic-krch

galak, 2021-05-04:
decided to merge fix to unblock CI issues, and open a new high priority issue to track adding hardware platforms to test this on beyond QEMU platforms.

PR #34749 (2021-05-04): tests: arch: arm: no_multithreading: Limit platforms allowed
When multithreading is off, kernel source files like sem.c (samphore
implementation) are not present in the build. Some platforms by default
fetch modules or drivers that are using multithreading primitives and
because of that fails to compile when multithreading is off.

Limit the test to only qemu platforms since test is arch specific.

Fixes #34739.
closed 2021-05-04 by ioannisg