zephyrproject-rtos/zephyr #34748: Native posix: Segmentation fault in case of allocations without explicit heap assignment

prathje, 2021-05-01:
**Describe the bug**
I am trying to dynamically allocate memory (for e.g. a message queue) in the native POSIX architecture.
Without explicitly assigning the main thread to a heap, I get a segmentation fault. I assume that the resource pool of the thread is not properly initialized which might result in a segmentation fault when dereferencing. 

**To Reproduce**
Steps to reproduce the behavior:
1. (Optional) `docker run --rm -it prathje/babble-sim-docker:latest /bin/bash` then `cd $ZEPHYR_BASE && git fetch && git checkout 180e61a32ed5002331d8d4b417f3b3938e921ff8 && west update`
2. Copy gist files: [https://gist.github.com/prathje/0667c337c5fd6423420e5f7074657220](https://gist.github.com/prathje/0667c337c5fd6423420e5f7074657220) 
3. `cd $ZEPHYR_BASE && west build -b native_posix --pristine auto <path-to-project>`
4. run `./build/zephyr/zephyr.exe` 
5. See uninitialized memory address

**Expected behavior**
*k_aligned_alloc* directly uses the shared _SYSTEM_HEAP and works as expected while methods such as *k_msgq_alloc_init* use the thread's assigned resource pool. For me, the expected default behavior would either be that bg_thread_main initially also uses the _SYSTEM_HEAP or none at all resulting in gracefully failing allocations.

**Impact**
I first tried to search for mistakes in my application code and did not assume that this could be directly related to Zephyr's resource management. As an explicit resource pool assignment solves this issue, the goal would be to save some time for others by either informing them of this behavior or eliminating the segmentation fault.

**Logs and console output**
Non-null initial resource pool
```bash
*** Booting Zephyr OS build zephyr-v2.5.0-3008-g180e61a32ed5  ***
Initial resource pool before assignment 0x14
Resource pool after system pool assignment 0x56637f10
```
Resulting Segmentation Error (example):
```bash
(gdb) run
Starting program: /zephyr/zephyr/build/zephyr/zephyr.exe 
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0xf7c0ab40 (LWP 20021)]
[New Thread 0xf72ffb40 (LWP 20022)]
[New Thread 0xf6afeb40 (LWP 20023)]
*** Booting Zephyr OS build zephyr-v2.5.0-3008-g180e61a32ed5  ***

Thread 3 "zephyr.exe" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xf72ffb40 (LWP 20022)]
0x565cd04a in sys_heap_aligned_alloc (heap=0x14, align=4, bytes=36) at /zephyr/zephyr/lib/os/heap.c:248
248		struct z_heap *h = heap->heap;
(gdb) backtrace
#0  0x565cd04a in sys_heap_aligned_alloc (heap=0x14, align=4, bytes=36) at /zephyr/zephyr/lib/os/heap.c:248
#1  0x565d0e45 in k_heap_aligned_alloc (h=0x14, align=4, bytes=36, timeout=...) at /zephyr/zephyr/kernel/kheap.c:39
#2  0x565d0c9b in z_heap_aligned_alloc (heap=0x14, align=<optimized out>, size=<optimized out>) at /zephyr/zephyr/kernel/mempool.c:29
#3  0x565d0d2d in z_thread_aligned_alloc (align=0, size=32) at /zephyr/zephyr/kernel/mempool.c:107
#4  0x565d00ef in z_thread_malloc (size=32) at ../kernel/include/kernel_internal.h:82
#5  z_impl_k_msgq_alloc_init (msgq=0x565d5458 <kheap.system_heap+48>, msg_size=8, max_msgs=4) at /zephyr/zephyr/kernel/msg_q.c:87
#6  0x565cca1b in k_msgq_alloc_init (msg_size=8, max_msgs=4, msgq=0x565d5458 <kheap.system_heap+48>) at zephyr/include/generated/syscalls/kernel.h:823
#7  zephyr_app_main () at /app/segmentation_fault_test/main.c:16
#8  0x565cfdfa in bg_thread_main (unused1=0x0, unused2=0x0, unused3=0x0) at /zephyr/zephyr/kernel/init.c:186
#9  0x565ccb00 in z_thread_entry (entry=0x565cfdc3 <bg_thread_main>, p1=0x0, p2=0x0, p3=0x0) at /zephyr/zephyr/lib/os/thread_entry.c:29
#10 0x565cdf12 in posix_thread_starter (arg=0x0) at /zephyr/zephyr/arch/posix/core/posix_core.c:305
#11 0xf7e04635 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#12 0xf7d158fa in clone () from /lib/i386-linux-gnu/libc.so.6
```

**Environment (please complete the following information):**
 - OS: MacOS (BigSur and Catalina)
 - Toolchain: Docker with [Dockerfile](https://github.com/prathje/BabbleSimDocker)
 - Commit SHA: 180e61a32ed5002331d8d4b417f3b3938e921ff8

**Additional context**
The [line which could reset the threads resource pool](https://github.com/zephyrproject-rtos/zephyr/blob/bfb1634af595207aa98e4a7233d12edb9b6769d7/kernel/thread.c#L575) is not executed because *_current* is not null.
This issue should also be tested in a native execution environment.

Thanks in advance!

aescolar, 2021-05-05:
@prathje this would seem to not be native_posix specific, but heap specific
=> @andrewboie , @andyross ^^

maksimmasalski, 2021-05-17:
@prathje Hello, I was assigned to look at that issue. Are you still experiencing it? 
Unfortunately, I can't check on MacOS, I have Fedora 28. I used the latest Zephyr version 41e885947e4f48aae0866df9e5fee657afb7c1cc
I tried that code, which, as I understand, must fail, because main thread not explicitly assigned to a heap:
```
#include <zephyr.h>
#include <stdlib.h>

#define MSGQ_MSG_SIZE 4
#define MSGQ_MAX_MSGS 16

void main(void) {
    struct k_thread *thread = k_current_get();
    printk("Initial resource pool before assignment %p\n", thread->resource_pool);
    //k_thread_system_pool_assign(k_current_get());
    printk("Resource pool after system pool assignment %p\n", thread->resource_pool);

    // TODO: without the explicit pool assignment, the k_msgq_alloc_init results in a segmentation fault
    struct k_msgq * queue = k_malloc(sizeof(struct k_msgq));
    //struct k_msgq * queue;
    k_msgq_alloc_init(queue, MSGQ_MSG_SIZE, MSGQ_MAX_MSGS);
}
```
No segmentation fault happens. Am I right with the code or you have another code for test?

prathje, 2021-05-17:
@maksimmasalski Hello and thank you for your support!
Your code seems correct. Can you confirm that the shared resource pool was correctly initialized even before the explicit assignment? If so, it might depend on the used platform.
I also confirmed the problem running the official CI image ([https://hub.docker.com/r/zephyrprojectrtos/ci](https://hub.docker.com/r/zephyrprojectrtos/ci)) using Docker+M1 Mac. As this

aescolar, 2021-05-17:
Can you run your test in valgrind? (`valgrind zephyr/zephyr.exe`)

prathje, 2021-05-17:
Sure, valgrind execution:
```
==13002== Memcheck, a memory error detector
==13002== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==13002== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==13002== Command: ./build/zephyr/zephyr.exe
==13002== 
--13002-- WARNING: unhandled x86-linux syscall: 403
--13002-- You may be able to write your own handler.
--13002-- Read the file README_MISSING_SYSCALL_OR_IOCTL.
--13002-- Nevertheless we consider this a bug.  Please report
--13002-- it at http://valgrind.org/support/bug_reports.html.
*** Booting Zephyr OS build zephyr-v2.5.0-3884-g878043aff19c  ***
==13002== Thread 3:
==13002== Conditional jump or move depends on uninitialised value(s)
==13002==    at 0x10C7A9: cbvprintf (cbprintf_complete.c:1640)
==13002==    by 0x10BA92: vprintk (printk.c:153)
==13002==    by 0x10BABE: printk (printk.c:218)
==13002==    by 0x10BA07: zephyr_app_main (main.c:9)
==13002==    by 0x10EC5A: bg_thread_main (init.c:224)
==13002==    by 0x10BAD9: z_thread_entry (thread_entry.c:29)
==13002==    by 0x10CD72: posix_thread_starter (posix_core.c:305)
==13002==    by 0x4962634: start_thread (pthread_create.c:477)
==13002==    by 0x4A858F9: clone (clone.S:108)
==13002== 
==13002== Conditional jump or move depends on uninitialised value(s)
==13002==    at 0x10C1B7: encode_uint (cbprintf_complete.c:796)
==13002==    by 0x10C7C4: cbvprintf (cbprintf_complete.c:1641)
==13002==    by 0x10BA92: vprintk (printk.c:153)
==13002==    by 0x10BABE: printk (printk.c:218)
==13002==    by 0x10BA07: zephyr_app_main (main.c:9)
==13002==    by 0x10EC5A: bg_thread_main (init.c:224)
==13002==    by 0x10BAD9: z_thread_entry (thread_entry.c:29)
==13002==    by 0x10CD72: posix_thread_starter (posix_core.c:305)
==13002==    by 0x4962634: start_thread (pthread_create.c:477)
==13002==    by 0x4A858F9: clone (clone.S:108)
==13002== 
==13002== Conditional jump or move depends on uninitialised value(s)
==13002==    at 0x10C1D5: encode_uint (cbprintf_complete.c:799)
==13002==    by 0x10C7C4: cbvprintf (cbprintf_complete.c:1641)
==13002==    by 0x10BA92: vprintk (printk.c:153)
==13002==    by 0x10BABE: printk (printk.c:218)
==13002==    by 0x10BA07: zephyr_app_main (main.c:9)
==13002==    by 0x10EC5A: bg_thread_main (init.c:224)
==13002==    by 0x10BAD9: z_thread_entry (thread_entry.c:29)
==13002==    by 0x10CD72: posix_thread_starter (posix_core.c:305)
==13002==    by 0x4962634: start_thread (pthread_create.c:477)
==13002==    by 0x4A858F9: clone (clone.S:108)
==13002== 
==13002== Conditional jump or move depends on uninitialised value(s)
==13002==    at 0x49FAF69: _IO_file_overflow@@GLIBC_2.1 (fileops.c:783)
==13002==    by 0x49F0E48: putchar (putchar.c:28)
==13002==    by 0x10C0D3: outs (cbprintf_complete.c:1319)
==13002==    by 0x10C997: cbvprintf (cbprintf_complete.c:1805)
==13002==    by 0x10BA92: vprintk (printk.c:153)
==13002==    by 0x10BABE: printk (printk.c:218)
==13002==    by 0x10BA07: zephyr_app_main (main.c:9)
==13002==    by 0x10EC5A: bg_thread_main (init.c:224)
==13002==    by 0x10BAD9: z_thread_entry (thread_entry.c:29)
==13002==    by 0x10CD72: posix_thread_starter (posix_core.c:305)
==13002==    by 0x4962634: start_thread (pthread_create.c:477)
==13002==    by 0x4A858F9: clone (clone.S:108)
==13002== 
==13002== Syscall param write(buf) points to uninitialised byte(s)
==13002==    at 0x4A72D03: __libc_write (write.c:26)
==13002==    by 0x4A72D03: write (write.c:24)
==13002==    by 0x49F99BA: _IO_file_write@@GLIBC_2.1 (fileops.c:1181)
==13002==    by 0x49F8AC6: new_do_write (fileops.c:449)
==13002==    by 0x49FAB7E: _IO_new_do_write (fileops.c:426)
==13002==    by 0x49FAB7E: _IO_do_write@@GLIBC_2.1 (fileops.c:423)
[... 122 lines cut ...]
=================================================================
==13598==ERROR: AddressSanitizer: SEGV on unknown address 0x001ac130 (pc 0x56629d0f bp 0xf46ff118 sp 0xf46ff0f0 T2)
==13598==The signal is caused by a READ memory access.
    #0 0x56629d0e in sys_heap_aligned_alloc /zephyr/zephyr/lib/os/heap.c:259

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /zephyr/zephyr/lib/os/heap.c:259 in sys_heap_aligned_alloc
Thread T2 created by T1 here:
    #0 0xf7a86d7a in __interceptor_pthread_create (/lib32/libasan.so.5+0xfbd7a)
    #1 0x5662d7d9 in posix_new_thread /zephyr/zephyr/arch/posix/core/posix_core.c:380
    #2 0x56631c51 in z_setup_new_thread /zephyr/zephyr/kernel/thread.c:548
    #3 0x56631977 in prepare_multithreading /zephyr/zephyr/kernel/init.c:303
    #4 0x56631977 in z_cstart /zephyr/zephyr/kernel/init.c:453
    #5 0x5662e02b in zephyr_wrapper /zephyr/zephyr/soc/posix/inf_clock/soc.c:188
    #6 0xf79be856  (/lib32/libasan.so.5+0x33856)
    #7 0xf777c8f9 in __clone (/lib/i386-linux-gnu/libc.so.6+0x1088f9)

Thread T1 created by T0 here:
    #0 0xf7a86d7a in __interceptor_pthread_create (/lib32/libasan.so.5+0xfbd7a)
    #1 0x5662e19f in posix_boot_cpu /zephyr/zephyr/soc/posix/inf_clock/soc.c:209
    #2 0x5662fa98 in main /zephyr/zephyr/boards/posix/native_posix/main.c:67
    #3 0xf7692ee4 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x1eee4)

==13598==ABORTING
```

maksimmasalski, 2021-05-18:
> @maksimmasalski Hello and thank you for your support!
> Your code seems correct. Can you confirm that the shared resource pool was correctly initialized even before the explicit assignment? If so, it might depend on the used platform.
> I also confirmed the problem running the official CI image ([hub.docker.com/r/zephyrprojectrtos/ci](https://hub.docker.com/r/zephyrprojectrtos/ci)) using Docker+M1 Mac. As this

Sorry for my question, but how to confirm that the shared resource pool was correctly initialized?

prathje, 2021-05-18:
@maksimmasalski No worries :) Initialized is actually misleading, it just needs to be correctly assigned to the thread which you can check by using the two print statements:
```c
    printk("Initial resource pool before assignment %p\n", thread->resource_pool);
    k_thread_system_pool_assign(k_current_get()); // this should have no effect, if the thread already was assigned to the system pool
    printk("Resource pool after system pool assignment %p\n", thread->resource_pool);
```
I.e. you just test if the explicit assignment changed the thread's resource pool. Before the explicit assignment, the thread should either have no pool (0x00) or the shared system one (the one after execution of ` k_thread_system_pool_assign(k_current_get());`).

maksimmasalski, 2021-05-18:
> @maksimmasalski No worries :) Initialized is actually misleading, it just needs to be correctly assigned to the thread which you can check by using the two print statements:
>
> ```c [...]

Understood) I tried that. It was changed, I tried code from your comment.
```
*** Booting Zephyr OS build v2.6.0-rc1-121-g8a42a73d3eb0  ***
Initial resource pool before assignment 0xf7c65358
Resource pool after system pool assignment 0x8056004
```

prathje, 2021-05-18:
And I think this means that the thread's resource pool was not correctly assigned. I am however not entirely sure. For me, the value was 0x14 (which directly resulted in a segmentation fault). I am not sure if 0xf7c65358 is a valid reference to a resource pool as there should normally just be this one shared system pool at 0x8056004, correct?

maksimmasalski, 2021-05-19:
> And I think this means that the thread's resource pool was not correctly assigned. I am however not entirely sure. For me, the value was 0x14 (which directly resulted in a segmentation fault). I am [...]

I didn't receive segmentation fault in my program. It was completed.

maksimmasalski, 2021-05-19:
> Can you run your test in valgrind? (`valgrind zephyr/zephyr.exe`)
> and/or build it with the address sanitizer enabled? (`CONFIG_ASAN`)

Hello, are you using Linux? On my Fedora I experience that kind of error when try to run valgrind
`valgrind: failed to start tool 'memcheck' for platform 'amd64-linux': Not a directory`

aescolar, 2021-05-19:
@maksimmasalski Yes, I do use Linux :), and never saw that problem you mention. Maybe https://stackoverflow.com/questions/64813183/ubuntu-valgrind-failed-to-start-tool-memcheck-for-platform-amd64-linux-no helps

maksimmasalski, 2021-05-19:
> @maksimmasalski Yes, I do use Linux :), and never saw that problem you mention. Maybe [stackoverflow.com/questions/64813183/ubuntu-valgrind-failed-to-start-tool-memcheck-for-platform-amd64-linux-no](https://stackoverflow.com/questions/64813183/ubuntu-valgrind-failed-to-start-tool-memcheck-for-platform-amd64-linux-no) helps

Ok, I will try. Are you working on that issue? If yes, can I assign you? Because seems you are more familiar with that.

aescolar, 2021-05-19:
> Ok, I will try. Are you working on that issue?

I'm not. From a quick look, this issue does not seem related to the posix arch/native_posix, or to even be a bug anywhere(?) but a misunderstanding of the usage of the heap. Ideally this "bug" would be reformulated as a question, and somebody else, more involved in the heap side, would answer it.

maksimmasalski, 2021-05-19:
> I'm not. From a quick look, this issue does not seem related to the posix arch/native_posix, or to even be a bug anywhere(?) but a misunderstanding of the usage of the heap. Ideally this "bug" would [...]

I understand that part can be documented better and that behavior can be described in the documentation.

maksimmasalski, 2021-05-20:
Current status:
On my Fedora 34 code didn't cause a Segmentation fault.
```
#include <zephyr.h>
#include <stdlib.h>

void main(void) {
    struct k_thread *thread = k_current_get();
    printk("Initial resource pool before assignment %p\n", thread->resource_pool);
    k_thread_system_pool_assign(k_current_get()); // this should have no effect, if the thread already was assigned to the system pool
    printk("Resource pool after system pool assignment %p\n", thread->resource_pool);
}
```
However, when I run in valgrind I see next result and errors:
```
==111281== Memcheck, a memory error detector
==111281== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==111281== Using Valgrind-3.17.0 and LibVEX; rerun with -h for copyright info
==111281== Command: build/zephyr/zephyr.exe
==111281== 
*** Booting Zephyr OS build v2.6.0-rc1-122-gc029bae3db0a  ***
==111281== Thread 3:
==111281== Conditional jump or move depends on uninitialised value(s)
==111281==    at 0x804ACB8: cbvprintf (cbprintf_complete.c:1640)
==111281==    by 0x80498B5: vprintk (printk.c:153)
==111281==    by 0x80498D6: printk (printk.c:218)
==111281==    by 0x8049823: zephyr_app_main (main.c:6)
==111281==    by 0x804E044: bg_thread_main (init.c:224)
==111281==    by 0x8049917: z_thread_entry (thread_entry.c:29)
==111281==    by 0x804B50A: posix_thread_starter (posix_core.c:305)
==111281==    by 0x41433D7: ??? (in /usr/lib/libpthread-2.33.so)
==111281==    by 0x4254E29: clone (in /usr/lib/libc-2.33.so)
==111281== 
==111281== Conditional jump or move depends on uninitialised value(s)
==111281==    at 0x804A569: encode_uint (cbprintf_complete.c:796)
==111281==    by 0x804ACDE: cbvprintf (cbprintf_complete.c:1641)
==111281==    by 0x80498B5: vprintk (printk.c:153)
==111281==    by 0x80498D6: printk (printk.c:218)
==111281==    by 0x8049823: zephyr_app_main (main.c:6)
==111281==    by 0x804E044: bg_thread_main (init.c:224)
==111281==    by 0x8049917: z_thread_entry (thread_entry.c:29)
==111281==    by 0x804B50A: posix_thread_starter (posix_core.c:305)
==111281==    by 0x41433D7: ??? (in /usr/lib/libpthread-2.33.so)
==111281==    by 0x4254E29: clone (in /usr/lib/libc-2.33.so)
==111281== 
==111281== Conditional jump or move depends on uninitialised value(s)
==111281==    at 0x804A5C4: encode_uint (cbprintf_complete.c:799)
==111281==    by 0x804ACDE: cbvprintf (cbprintf_complete.c:1641)
==111281==    by 0x80498B5: vprintk (printk.c:153)
==111281==    by 0x80498D6: printk (printk.c:218)
==111281==    by 0x8049823: zephyr_app_main (main.c:6)
==111281==    by 0x804E044: bg_thread_main (init.c:224)
==111281==    by 0x8049917: z_thread_entry (thread_entry.c:29)
==111281==    by 0x804B50A: posix_thread_starter (posix_core.c:305)
==111281==    by 0x41433D7: ??? (in /usr/lib/libpthread-2.33.so)
==111281==    by 0x4254E29: clone (in /usr/lib/libc-2.33.so)
==111281== 
==111281== Conditional jump or move depends on uninitialised value(s)
==111281==    at 0x41DBAE9: _IO_file_overflow (in /usr/lib/libc-2.33.so)
==111281==    by 0x41D1E98: putchar (in /usr/lib/libc-2.33.so)
==111281==    by 0x804988F: char_out (printk.c:118)
==111281==    by 0x804A6BD: outs (cbprintf_complete.c:1319)
==111281==    by 0x804AF5D: cbvprintf (cbprintf_complete.c:1805)
==111281==    by 0x80498B5: vprintk (printk.c:153)
==111281==    by 0x80498D6: printk (printk.c:218)
==111281==    by 0x8049823: zephyr_app_main (main.c:6)
==111281==    by 0x804E044: bg_thread_main (init.c:224)
==111281==    by 0x8049917: z_thread_entry (thread_entry.c:29)
==111281==    by 0x804B50A: posix_thread_starter (posix_core.c:305)
==111281==    by 0x41433D7: ??? (in /usr/lib/libpthread-2.33.so)
[... 27 lines cut ...]
==111281== 
Initial resource pool before assignment 0x4f1f338
Resource pool after system pool assignment 0x8055004
^C
Stopped at 48.510s
==111281== 
==111281== HEAP SUMMARY:
==111281==     in use at exit: 136 bytes in 1 blocks
==111281==   total heap usage: 11 allocs, 10 frees, 4,292 bytes allocated
==111281== 
==111281== LEAK SUMMARY:
==111281==    definitely lost: 0 bytes in 0 blocks
==111281==    indirectly lost: 0 bytes in 0 blocks
==111281==      possibly lost: 136 bytes in 1 blocks
==111281==    still reachable: 0 bytes in 0 blocks
==111281==         suppressed: 0 bytes in 0 blocks
==111281== Rerun with --leak-check=full to see details of leaked memory
==111281== 
==111281== Use --track-origins=yes to see where uninitialised values come from
==111281== For lists of detected and suppressed errors, rerun with: -s
==111281== ERROR SUMMARY: 23 errors from 5 contexts (suppressed: 0 from 0)
```
Next steps:
I recommend someone who has M1 Apple Mac try it and repeat the same behaviour which had @prathje 
I asked on Slack community try that code on their M1 Macs if they are using them.

maksimmasalski, 2021-05-20:
I tried one more time to run code. But commented out all printk.
```
#include <zephyr.h>
#include <stdlib.h>

void main(void) {
    struct k_thread *thread = k_current_get();
    //printk("Initial resource pool before assignment %p\n", thread->resource_pool);
    k_thread_system_pool_assign(k_current_get()); // this should have no effect, if the thread already was assigned to the system pool
    //printk("Resource pool after system pool assignment %p\n", thread->resource_pool);
    printf("Success\n");
}
```
This time, valgrind didn't complain much:
```
==123882== Memcheck, a memory error detector
==123882== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==123882== Using Valgrind-3.17.0 and LibVEX; rerun with -h for copyright info
==123882== Command: build/zephyr/zephyr.exe
==123882== 
*** Booting Zephyr OS build v2.6.0-rc1-122-gc029bae3db0a  ***
Success
^C
Stopped at 1.520s
==123882== 
==123882== HEAP SUMMARY:
==123882==     in use at exit: 136 bytes in 1 blocks
==123882==   total heap usage: 11 allocs, 10 frees, 4,292 bytes allocated
==123882== 
==123882== LEAK SUMMARY:
==123882==    definitely lost: 0 bytes in 0 blocks
==123882==    indirectly lost: 0 bytes in 0 blocks
==123882==      possibly lost: 136 bytes in 1 blocks
==123882==    still reachable: 0 bytes in 0 blocks
==123882==         suppressed: 0 bytes in 0 blocks
==123882== Rerun with --leak-check=full to see details of leaked memory
==123882== 
==123882== For lists of detected and suppressed errors, rerun with: -s
==123882== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

maksimmasalski, 2021-05-20:
@prathje I think your Segmentation fault is platform related. On Fedora build native_posix no seg fault as I can see.

prathje, 2021-05-20:
@maksimmasalski Thank you very much for your support! You are still encountering an uninitialized resource_pool, correct? Valgrind essentially tells you that the memory address (i.e., the pointer thread->resource_pool) has not been initialized. If this is the case (and I do not see any reason why the resource_pool should be correct as it is not the shared heap?), we would still need to initialize the pointer to the resource_pool. I am still not sure if this is a common problem or just for the native_posix architecture. However, we could initialize the resource_pool at https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/arch/posix/core/swap.c#L98 as follows: 

``` C
#if (CONFIG_HEAP_MEM_POOL_SIZE > 0)
        k_thread_system_pool_assign(_current);
#else
        _current->resource_pool = NULL;
#endif
```

This fix is at the moment very specific to native_posix but results in a valid resource_pool (the system one):
``` bash
*** Booting Zephyr OS build v2.6.0-rc1-167-gfa836cc00d4f  ***
Initial resource pool before assignment 0x5660bf10
Resource pool after system pool assignment 0x5660bf10
```
I set the shared resource pool to the main thread but a null assignment would also be valid (this just needs to be mentioned in the docs).
That said, I have no deep knowledge about Zephyr's threads, initialization, or heap handling - so this is rather a wild guess...
If this would be the way to go, I'll happily create a pull-request 👍

maksimmasalski, 2021-05-21:
> @maksimmasalski Thank you very much for your support! You are still encountering an uninitialized resource_pool, correct? Valgrind essentially tells you that the memory address (i.e., the pointer thread->resource_pool) has not been initialized. If this is the case (and I do not see any reason why the resource_pool should be correct as it is not the shared heap?), we would still need to initialize the pointer to the resource_pool. I am still not sure if this is a common problem or just for the native_posix architecture. However, we could initialize the resource_pool at
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/arch/posix/core/swap.c#L98
> 
> as follows:
> ```c
> #if (CONFIG_HEAP_MEM_POOL_SIZE > 0)
>         k_thread_system_pool_assign(_current);
> #else
>         _current->resource_pool = NULL;
> #endif
> ```
> 
> This fix is at the moment very specific to native_posix but results in a valid resource_pool (the system one):
> 
> ```shell
> *** Booting Zephyr OS build v2.6.0-rc1-167-gfa836cc00d4f  ***
> Initial resource pool before assignment 0x5660bf10
> Resource pool after system pool assignment 0x5660bf10
> ```
> 
> I set the shared resource pool to the main thread but a null assignment would also be valid (this just needs to be mentioned in the docs).
> That said, I have no deep knowledge about Zephyr's threads, initialization, or heap handling - so this is rather a wild guess...
> If this would be the way to go, I'll happily create a pull-request 👍

Please submit the PR.

aescolar, 2021-05-21:
I'm not too versed on the resource pool initialization, but in other architectures this does not seem to be initialized in the arch code. This initialization would seem to be the task of the kernel (`z_setup_new_thread()` ), or?
@andyross could you please bring some light here?

maksimmasalski, 2021-05-24:
> I'm not too versed on the resource pool initialization, but in other architectures this does not seem to be initialized in the arch code. This initialization would seem to be the task of the kernel (`z_setup_new_thread()` ), or?
> @andyross could you please bring some light here?

@andyross Could you help us?

maksimmasalski, 2021-06-02:
Closed it, because it is not reproducable and no any updates from the developers.

closed 2021-06-02 by maksimmasalski