zephyrproject-rtos/zephyr #34752: west flash: nrfjprog: using --pinresetenable by default may lead to bricking development boards

hubertmis, 2021-05-01:
**Is your enhancement proposal related to a problem? Please describe.**
I'm using a development board including nRF52840. Because of limited number of available pins, the pin assigned by default as RESET (0.18) on this board is connected to a peripheral. I expected that setting `CONFIG_GPIO_AS_PINRESET=n` in the project configuration would be enough to easily use Zephyr on this board. Unfortunately, there is another independent feature enabling reset pin: `west flash` by default calls `nrfjprog --pinresetenable`. If I use `west flash` I brick the device, because peripheral connected to reset pin causes reset loop and I cannot event connect a programmer to reflash the device. The only recovery method I found is to desolder the peripheral connected to pin 18. 

I can use `west flash --softreset` and it works correctly, because it does not reconfigure pin 18. However, I can easily by mistake type `west flash` command and brick my board again.

**Describe the solution you'd like**
I would like `west flash` not to call `nrfjprog --pinresetenable` if `CONFIG_GPIO_AS_PINRESET=n` is in the project configuration.

**Describe alternatives you've considered**
Alternatively, `west flash` could be modified to call system reset by default and replace command line argument `--softreset` with `--pinreset`, because system reset is safer (it cannot brick any device).

Swij, 2021-05-02:
I just encountered this issue as well on my board. I think it's unacceptable that west modifies the UICR as a default behavior when it resets the board. Pin reset should only be enabled if the user requests it using `--pinresetenable`.

joicetm, 2021-05-05:
I am also facing a similar issue, I could not control the p0.18 even after adding "CONFIG_GPIO_AS_PINRESET=n" to the config file.

I have  tried   _west flash --softreset_  and also uploaded the .hex using the nrfjprog directly but it doesn't seem to work.(for me, it may not be configuring the pin as normal gpio in the first place) 

I have used the blinky examples from zephyr samples to test it by adding CONFIG_GPIO_AS_PINRESET=n to project config as well as setting it through menuconfig. 
I have also went through zephyr.dts file to make sure this pin is not configured for any other peripheral. 
I have tested the board and pin by testing it with nrf5 sdk code.

@hubertmis @Swij Have you guys done anything else to make it work as a GPIO pin? Could find out what the issue is.

hubertmis, 2021-05-05:
> Have you guys done anything else to make it work as a GPIO pin? Could find out what the issue is.

Try

```
nrfjprog -e
west flash --softreset
```

It worked for me. 

`nrfjprog -e` clears memory, including UICR holding pin reset configuration.

joicetm, 2021-05-05:
@hubertmis Thanks man,
It is working.