zephyrproject-rtos/zephyr #34753: Building and Debugging Zephyr for Native Platform on Linux using VSCode and/or QtCreator

bob2oneil, 2021-05-01:
I am trying to use a standard IDE on Linux as a build and development platform for Zephyr, with the board targets to native POSIX builds.

I have had success with the process in VS Code, where I am able to single step and debug into the Zephyr.exe generated using the standard "west build -p auto -b native_posix_64 samples/string" build tool, and then specifying the "west" built binary in the build subfolder.  I added the following 
kernel configuration to generated debug info in the binary.

CONFIG_DEBUG_OPTIMIZATIONS=y

But how do I integrate the build process more tightly with VSCode?

And what about tight integration with the QtCreator IDE?

It allows for the creation of a non-Qt Cmake project, but does not seem to do the import of the 
Zephyr Cmake content well.

Has anybody has success with tighter VSCode and/or QtCreator integration for native platform building?

aunsbjerg, 2021-05-03:
#21119 and #17645 may have some useful information for using vscode for zephyr development.

You can add a default build task in vscode that invokes west build:

```
        {
            "label": "build",
            "type": "shell",
            "group": {
                "isDefault": true,
                "kind": "build"
            },
            "command": "source zephyr/zephyr-env.sh && west build -b native_posix_64 path/to/application",
            "problemMatcher": {
                "owner": "cpp",
                "fileLocation": ["relative", "${workspaceFolder}"],
                "pattern": {
                    "regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
                    "file": 1,
                    "line": 2,
                    "column": 3,
                    "severity": 4,
                    "message": 5
                }
            },
        },
```

Then you can use `ctrl+b` to quickly build. You can also add task inputs to make the build target a bit more generic.

bob2oneil, 2021-05-12:
Thanks, works as you described well.

closed 2021-05-12 by bob2oneil