zephyrproject-rtos/zephyr #34772: Mixed usage of signed/unsigned integer by the logging subsystem

gmarull, 2021-05-03:
**Describe the bug**

If `-Wextra` is enabled on an application using the logging subsystem a few compilers warnings related to signed/unsigned comparisons are reported

**To Reproduce**
Steps to reproduce the behavior:
1. Take any application that makes uses of the logging subsystem, e.g. `LOG_ERR(...)`.
2. Enable `-Wextra`, `target_compile_options(app PRIVATE -Wextra)`
3. Observe reported warnings

**Expected behavior**
I expect that logging subsystem makes consistent usage of signed/unsigned integers

**Impact**
`-Wextra` may be useful in applications that want to enforce stricter rules (e.g. MISRA)

**Logs and console output**
```
zephyr/include/logging/log_msg2.h:297:39: warning: operand of ?: changes signedness from 'int' to 'unsigned int' due to unsignedness of other operand [-Wsign-compare]
  297 |   CBPRINTF_STATIC_PACKAGE(_msg->data, _plen, _plen, \
zephyr/include/sys/cbprintf_internal.h:230:19: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'unsigned int'} [-Wsign-compare]                                                                                                            
  230 |  if (_buf && _idx < _max) { \  
zephyr/include/sys/cbprintf_internal.h:320:22: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'unsigned int'} [-Wsign-compare]
  320 |  _outlen = (_pkg_len > _pmax) ? -ENOSPC : _pkg_len; \

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.12.4
 - Commit SHA or Version used 5d2909654fb572bc059bfc530a669d086797fe6a

**Additional context**
Add any other context about the problem here.

nordic-krch, 2021-05-06:
I don't see it on master anymore. Fixed by #34697. @gmarull can you confirm?

gmarull, 2021-05-06:
@nordic-krch seems fixed after a `west update`, thanks

closed 2021-05-06 by gmarull