zephyrproject-rtos/zephyr #34794: LIS2DH Hard Fault when INT2 is not defined

jaredwolff, 2021-05-04:
**Describe the bug**
I noticed that since pulling a newer version of Zephyr through Nordic's nRF Connect SDK my accelerometer code stopped working. After some research and adding these

```
CONFIG_LOG=y
CONFIG_RESET_ON_FATAL_ERROR=n
```

(See logs below)

My best guess is that the NULL reference of `cfg->irq2_dev_name` in `lis2dh_trigger.c` is causing this to happen. A workaround right now is to check if `irq2_dev_name` is NULL before trying to run `device_get_binding` on it.

**To Reproduce**

1. Edit the `prj.conf` in the `lis2dh` driver example and add

```
CONFIG_LIS2DH_TRIGGER_GLOBAL_THREAD=y

CONFIG_LOG=y
CONFIG_RESET_ON_FATAL_ERROR=n
```

2. Build for your board. Make sure there's only one interrupt pin configured. Here's an example of my overlay for the nRF9160 Feather

```
&i2c1 {
	lis2dh: lis2dh@18 {
		compatible = "st,lis2dh";
		label = "LIS2DH";
		reg = <0x18>;
		irq-gpios = <&gpio0 29 GPIO_ACTIVE_HIGH>;
	};
};
```

3. Flash your board 
4. Connect UART0 so you can view the output. (if not already)

**Expected behavior**
Runtime should not fault or have an error condition.

**Impact**
Annoyance after figuring it out. 

**Logs and console output**

```
Â©¹jÑZÂÅ[ubÒåÂZÿòvÛIÙûÖÅ¥ÃxÉßi[00:00:00.002,075] [0m<inf> lis2dh: int1 on GPIO_0.29[0m
2021-05-04T02:45:56.460Z DEBUG modem << [00:00:00.002,105] [1;31m<err> os: Exception occurred in Secure State[0m
2021-05-04T02:45:56.465Z DEBUG modem << [00:00:00.002,105] [1;31m<err> os: ***** HARD FAULT *****[0m
2021-05-04T02:45:56.467Z DEBUG modem << [00:00:00.002,105] [1;31m<err> os:   Fault escalation (see below)[0m
2021-05-04T02:45:56.479Z DEBUG modem << [00:00:00.002,136] [1;31m<err> os: ***** BUS FAULT *****[0m
2021-05-04T02:45:56.481Z DEBUG modem << [00:00:00.002,136] [1;31m<err> os:   Precise data bus error[0m
2021-05-04T02:45:56.483Z DEBUG modem << [00:00:00.002,136] [1;31m<err> os:   BFAR Address: 0x50008158[0m
2021-05-04T02:45:56.484Z DEBUG modem << [00:00:00.002,136] [1;31m<err> os: r0/a1:  0x00000000  r1/a2:  0x00024501  r2/a3:  0x00000000[0m
2021-05-04T02:45:56.485Z DEBUG modem << [00:00:00.002,136] [1;31m<err> os: r3/a4:  0xffffffff r12/ip:  0x00000000 r14/lr:  0x00021227[0m
2021-05-04T02:45:56.487Z DEBUG modem << [00:00:00.002,166] [1;31m<err> os:  xpsr:  0x21000000[0m
2021-05-04T02:45:56.488Z DEBUG modem << [00:00:00.002,166] [1;31m<err> os: Faulting instruction address (r15/pc): 0x00023462[0m
2021-05-04T02:45:56.490Z DEBUG modem << [00:00:00.002,166] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
2021-05-04T02:45:56.491Z DEBUG modem << [00:00:00.002,166] [1;31m<err> os: Current thread: 0x20010218 (unknown)[0m
2021-05-04T02:45:56.493Z DEBUG modem << [00:00:00.090,087] [1;31m<err> os: Halting system[0m
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain NCS 1.5.0

avisconti, 2021-05-13:
I'm not able to reproduce this issue unfortunately, but I'm on a different h/w though. 
Initially I thought the issue was the one highlighted by you (and maybe still is), so the ```cfg->irq2_dev_name``` not initialized.
But then I tried with following configuration:
  1x instance on iks02a1 shield with drdy irq only
  1x instance on DIL24 with both drdy and AnyMotion irq

```  
&arduino_i2c {
        lsm303agr-accel@19 {
                compatible = "st,lis2dh", "st,lsm303agr-accel";
                reg = <0x19>;
                label = "LSM303AGR-ACCEL";
                irq-gpios = <&arduino_header 2 GPIO_ACTIVE_HIGH>;>      /* A2 - MODIFY defaults on JP9/JP10 */
        };

        lis2dh@18 {
                compatible = "st,lis2dh";
                reg = <0x18>;
                label = "LIS2DH";
                irq-gpios = <&arduino_header 5 GPIO_ACTIVE_HIGH>,  <&arduino_header 4 GPIO_ACTIVE_HIGH>;       /* A5 */
        };
};
```

Logs are as below:
```
X-NUCLEO-IKS01A2 sensor dashboard

LSM303AGR: Accel (m.s-2): x: 0.0, y: 0.0, z: 0.0
lis2dh: Accel (m.s-2): x: 0.0, y: 0.0, z: 0.0
1:: lsm303agr_a trig 2
1:: lis2dh trig drdy 6
1:: lis2dh trigs delta2 0
[00:00:00.000,000] <inf> lis2dh: int1 on GPIOC.00
[00:00:00.000,000] <inf> lis2dh: int2 on GPIOC.01
[00:00:00.001,000] <inf> lis2dh: bus=I2C_1 fs=2, odr=0x4 lp_en=0x0 scale=9576
[00:00:00.001,000] <inf> lis2dh: int1 on GPIOA.04
[00:00:00.001,000] <inf> lis2dh: Cannot get pointer to irq2_dev_name
[00:00:00.001,000] <inf> lis2dh: bus=I2C_1 fs=2, odr=0x4 lp_en=0x0 scale=9355
[00:00:00.009,000] <inf> lis2dh: int2_ths=0x41 range_g=2 ums2=9999999
[00:00:00.009,000] <inf> lis2dh: int2_dur=0x5
```
So, in my case it seems that device_get_binding() does not crash.
Anyway, I will align this code with latest drivers, where some checks with device_is_ready are performed, and let's see.

jaredwolff, 2021-05-13:
@avisconti thanks for taking a look. nRF Connect SDK is behind Zephyr's master HEAD but not by much. I wouldn't think much has changed between the two. 

I'd be curious if you changed the second one whereas you only had one interrupt pin. I wonder since the lsm303agr is not my exact use case.

I'm also tooling around with adding the option of enabling INT1 for AnyMotion. I only have one GPIO connected for interrupts so having that option is super useful.

avisconti, 2021-05-13:
@jaredwolff Can you try #35261? Not sure it solves the issue though, as I was not able to replicate it on my side.

jaredwolff, 2021-05-13:
I'll give it a whirl and get back to you shortly!

jaredwolff, 2021-05-13:
@avisconti I tried your patch but it looks like it's not 100% in sync with the latest on nRF Connect SDK. I can confirm by using the latest master of NCS this problem does not occur with the following overlay:

```
&i2c1 {
	lis2dh@18 {
		compatible = "st,lis2dh";
		label = "LIS2DH";
		reg = <0x18>;
		irq-gpios = <&gpio0 29 0>;
	};
};
```

I'll close this out. Thanks again for your help!

avisconti, 2021-05-13:
> I'd be curious if you changed the second one whereas you only had one interrupt pin. I wonder since the lsm303agr is not my exact use case.

@jaredwolff It is crashing infact. Not sure why it happens only in this way. But I can debug it right now...

avisconti, 2021-05-13:
@jaredwolff I re-open the issue because I was able to reproduce the issue removing the second int from the second instance. So, I would like to understand why it is happening...

jaredwolff, 2021-05-13:
@avisconti no problem. Glad you were able to reproduce it!

jaredwolff, 2021-05-13:
@avisconti  one thing I did do on my end which I think would address it is adding this in `list2dh_init_interrupt` before working on INT2:

```
	/*
	 * Setup INT2 (for Any Motion) if defined in DT
	 */

	if (cfg->irq2_dev_name == NULL) {
		LOG_INF("Cannot get pointer to irq2_dev_name");
		status = 0;
		goto end;
	}

```

Whereas `device_get_binding` seems to be the culprit.

avisconti, 2021-05-13:
> @avisconti no problem. Glad you were able to reproduce it!

I got the issue. I still was setting from application the AnyMotion trigger. So, I now added some code to the PR to avoid this cases. I will close again this bug.

commit 282a644b (2021-06-02): drivers/sensor: lis2dh: use gpio_dt_spec oriented macros

Use gpio_dt_spec structure and related macros for both drdy
and AnyMotion interrupts, to have a more compat, readable and
safe code. Moreover, skip setting DRDY or AnyMotion trigger
from application if the corresponding irq-gpios has not been set.

(This commit also fixes #34794)
closed 2021-05-13 by avisconti