zephyrproject-rtos/zephyr #34827: tests: power management: test_power_state_trans fails on nrf boards

PerMac, 2021-05-04:
**Describe the bug**
The test test_power_state_trans from tests/subsys/power/power_mgmt/ fails on-target tests (at least on nrf and reel boards)

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk connected
2. go to your zephyr dir
3. call `/scripts/twister -T /tests/subsys/power/power_mgmt -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See error

**Expected behavior**
The test passes

**Impact**
Not clear

**Logs and console output**
```
START - test_power_state_trans
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt/src/main.c:109: notify_pm_state_exit: (notify_app_exit == true is false)
Notification to leave suspend was not sent to the App
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt/src/main.c:112: notify_pm_state_exit: state not equal to PM_STATE_RUNTIME_IDLE
FAIL - test_power_state_trans in 0.133 seconds
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK 0.12.2
 - Commit SHA or Version used: zephyr-v2.5.0-3011-gb61c23943c6f

PerMac, 2021-05-06:
I think the same bug is causing more power mgmt tests/samples to fail, as I see also this one failing from the same time: sample.power.ospm.dev_idle_pm

closed 2021-05-13 by galak

PerMac, 2021-05-17:
@gmarull The fail of test_power_state_trans  is gone, thanks. Do you know if sample.power.ospm.dev_idle_pm (samples/subsys/pm) is related? Should I raise a separate issue? This test is still failing with the fix (nrf52840 and nrf52832). It hangs at:
```
ERROR   - *** Booting Zephyr OS build zephyr-v2.5.0-3821-g41afd838aabd  ***
Device PM sample app start
open()
parent resuming..
Async wakeup request queued
child resuming..
Dummy device resumed
write()
transfer()
read()
transfer()
close()
child suspending..
Device PM sample app complete
parent suspending..
```

gmarull, 2021-05-17:
@ceolin do you know if related to any recent change?

gmarull, 2021-05-17:
@PerMac https://github.com/zephyrproject-rtos/zephyr/pull/35315 may fix the problem, but I haven't reviewed yet

PerMac, 2021-05-17:
unluckily, #35315 doesn't change the test behavior. I will create a separate issue for ease of tracking.

PR #35233 (2021-05-13): pm: device: check for ENOSYS
If feature is not available -ENOSYS is used instead of -ENOTSUP.

Fixes #34827