zephyrproject-rtos/zephyr #34832: Coding Guideline - MISRA rule 14.4 not applied properly

Thalley, 2021-05-04:
**Is your enhancement proposal related to a problem? Please describe.**
MISRA rule 14.4 basically enforces strong typing, such that <expr> in `if (<expr>)` shall be a binary expression. It works for some examples such as 
```
void foo(void *pointer)
{
        if (pointer) { ... } 
}
``` 
which should be `if (pointer != NULL) { ... } `

But seemingly don't work for any of the following examples (and probably more):
```
void foo(void *pointer_a, void *pointer_b, int val)
{
        if (!pointer_a) { ... } 
        if (pointer_a && pointer_b) { ... }
        if (val) { ... }
}
```

**Describe the solution you'd like**
The script should be updated to check that expressions used for binary boolean operators (e.g. `&&` and `||`) are also booleans, as well as other integer values (including but not limited to int, uintX_t, long). 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2021-05-04:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/34823

ceolin, 2021-05-05:
Yep, the coccinelle script does not catch these and probably other examples. There are a lot of room for improvements.
Just one thing, the idea with these scripts is catch more common errors fast but not replace the static analysis tool.

Thalley, 2021-05-05:
@ceolin @nashif I assume that the plan is to follow the rule fully, right? So we should actually have boolean expressions for boolean checks?

ceolin, 2021-05-05:
> @ceolin @nashif I assume that the plan is to follow the rule fully, right? So we should actually have boolean expressions for boolean checks?

Yep, that is the plan.

ceolin, 2021-07-07:
The project is using Parasoft static analysis tool to catch these code guideline violations.

closed 2021-07-07 by ceolin