zephyrproject-rtos/zephyr #34837: Unstable multi connections between NRF52840

hf81-ble, 2021-05-04:
**Describe the bug**
I want to run a stable simultaneous multi connection between 1 NRF52840 and all 20 Peripherals. With DLE to 251 and 2 MBit

I calc a maximal Event Time from from ~ 1,5ms. =>  1,5 * 20 => 30ms needed for a connection interval. 

To use 10ms for scanning => 40ms connection intervall should be sufficient. 

Is this basic calculation correct?

The problem is, that connected peripherals get disconnected during a read from the peripherals.

connection is created with min=max = 40ms. 

scan interval is smaller than 10ms

What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
```
1. connect with connection interval 40ms + 10 ms scan interval
2. peripherals are connection step by step
3. from 6 upwards timing gets corrupted and sporadic some peripherals gets disconnected
4. sometimes a stable connection is possible, if no new peripherals are added

**Expected behavior**
 There BLE Stack scheduling should be fixed after adding a new connection to the scheduling. Even if a peripheral is not instant responding, there should be no impact on the internal scheduling. 

stability decreases on increasing amount of connections. 

**Impact**
Connection are disconnected. Unstable connections on longterm 

**Logs and console output**

One connection interval without data transfer. It is ack transfer. should you see the ACK in master packet?!

![grafik](https://user-images.githubusercontent.com/83449580/117031664-1a22e880-ad01-11eb-8efb-136387d6334c.png)

After Blocktransfer no slave responses:

![grafik](https://user-images.githubusercontent.com/83449580/117032507-d2e92780-ad01-11eb-97a7-5ef5e4018dae.png)

Disconnect 
![grafik](https://user-images.githubusercontent.com/83449580/117035046-2eb4b000-ad04-11eb-869b-909ebb321fb8.png)

**Environment (please complete the following information):**
 - OS: ( Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - zeyphyr 2.4.99

**Additional context**

cvinayak, 2021-05-05:
1. When using 251 bytes, connection event length has to be set with correct time in microseconds. This is not implemented in the Data Length Update Procedure. Refer to, https://github.com/zephyrproject-rtos/zephyr/issues/21995
2. Please upload air traffic log files, screenshots can not be used to analyze packet contents.

hf81-ble, 2021-05-05:
just added the log files

cvinayak, 2021-05-05:
> The timing offset for the event length in the ble LL scheduler for the connections are not yet ported?!

Correct.

As a quick test, please change 328 to 2120 here: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/ull_master.c#L324

You have to place your connections 5 ms apart, and use connection interval of 110ms.

Please upload sniffer logs of the results.

cvinayak, 2021-05-06:
@hf81-ble  Do you have any results? 

If there is no more queries/issue, please close this github issue.

hf81-ble, 2021-05-06:
is it recommended switch between different connection intervals in a connected state?

for example:

i have one active transmission and want to increase throughput and setting the connection intervall to 7,5ms. After that, is switch to another peripheral and set previous to connection intervall 100ms and the next peripheral to 7,5ms. So i use a roundtrip process to request data from each peripheral with a kind of burst mode.

cvinayak, 2021-05-07:
> Another question to multiconnection communication. is it recommended switch between different connection intervals in a connected state?

Yes, if you want lower latency on one connection at a time, you can switch to lower connection interval on that connection and increase connection interval on others. When you put back all connections to same connection interval, the Zephyr controller scheduling to group the connections back into non-overlapping time space.

Do revert back if you see any scheduling issues.

carlescufi, 2021-05-07:
Closing since this is informational. The conversation for advice can continue even with the issue closed.

closed 2021-05-07 by carlescufi