zephyrproject-rtos/zephyr #34838: tests/subsys/logging/log_msg2 failes on qemu_cortex_a53

dcpleung, 2021-05-04:
**Describe the bug**
There is build error with `tests/subsys/logging/log_msg2` on `qemu_cortex_a53` due to uninitialized variable. However, after setting a default value to `lli`, the test would fail.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -d /tmp/build -p -b qemu_cortex_a53 tests/subsys/logging/log_msg2/

**Expected behavior**
Should build without errors.

**Impact**
Prevents verifying local changes via twister.

**Logs and console output**
```
zephyr/rtos/zephyr/subsys/logging/log_core.c: In function 'log_hexdump':
zephyr/rtos/zephyr/subsys/logging/log_core.c:341:6: note: parameter passing for argument of type 'struct log_msg_ids' changed in GCC 9.1
  341 | void log_hexdump(const char *str, const void *data, uint32_t length,
      |      ^~~~~~~~~~~
[118/129] Building C object CMakeFiles/app.dir/src/main.c.obj
In file included from zephyr/rtos/zephyr/tests/subsys/logging/log_msg2/src/main.c:12:
zephyr/rtos/zephyr/tests/subsys/logging/log_msg2/src/main.c: In function 'test_log_msg2_fp':
zephyr/rtos/zephyr/include/logging/log_msg2.h:420:3: warning: 'lli' is used uninitialized in this function [-Wuninitialized]
  420 |   z_log_msg2_runtime_create(_domain_id, (void *)_source, \
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/rtos/zephyr/tests/subsys/logging/log_msg2/src/main.c:311:12: note: 'lli' was declared here
  311 |  long long lli;
      |            ^~~
In file included from zephyr/rtos/zephyr/include/logging/log_msg2.h:655,
                 from zephyr/rtos/zephyr/tests/subsys/logging/log_msg2/src/main.c:12:
zephyr/include/generated/syscalls/log_msg2.h: In function 'z_log_msg2_static_create.constprop':
zephyr/include/generated/syscalls/log_msg2.h:25:20: note: parameter passing for argument of type 'const struct log_msg2_desc' changed in GCC 9.1
   25 | static inline void z_log_msg2_static_create(const void * source, const struct log_msg2_desc desc, uint8_t * package, const void * data)
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK 0.12.4
 - commit 556baa2fd216227b67d5097ceb9a158787ef5c66

nixchun, 2021-05-05:
I didn't have this compilation problem on Zephyr 0.12.4.

user@ubuntu_20_04:~/projects/open/zephyr/zephyr$ west build -d build/log_msg2/ -p -b qemu_cortex_a53 tests/subsys/logging/log_msg2/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/user/projects/open/zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/user/projects/open/zephyr/zephyr/tests/subsys/logging/log_msg2
-- Zephyr version: 2.5.99 (/home/user/projects/open/zephyr/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: qemu_cortex_a53
-- Cache files will be written to: /home/user/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.4 (/home/user/.local/zephyr-sdk-0.12.4)
-- Found dtc: /home/user/.local/zephyr-sdk-0.12.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/user/projects/open/zephyr/zephyr/boards/arm64/qemu_cortex_a53/qemu_cortex_a53.dts
-- Generated zephyr.dts: /home/user/projects/open/zephyr/zephyr/build/log_msg2/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/user/projects/open/zephyr/zephyr/build/log_msg2/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/user/projects/open/zephyr/zephyr/build/log_msg2/zephyr/include/generated/device_extern.h
Parsing /home/user/projects/open/zephyr/zephyr/Kconfig
Loaded configuration '/home/user/projects/open/zephyr/zephyr/boards/arm64/qemu_cortex_a53/qemu_cortex_a53_defconfig'
Merged configuration '/home/user/projects/open/zephyr/zephyr/tests/subsys/logging/log_msg2/prj.conf'
Configuration saved to '/home/user/projects/open/zephyr/zephyr/build/log_msg2/zephyr/.config'
Kconfig header saved to '/home/user/projects/open/zephyr/zephyr/build/log_msg2/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.local/zephyr-sdk-0.12.4/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc
CMake Deprecation Warning at /home/user/projects/open/zephyr/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1251 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1196 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/user/projects/open/zephyr/zephyr/build/log_msg2
-- west build: building application
[1/129] Preparing syscall dependency handling

[21/129] Building C object CMakeFiles/app.dir/src/main.c.obj
In file included from /home/user/projects/open/zephyr/zephyr/include/logging/log_msg2.h:655,
                 from /home/user/projects/open/zephyr/zephyr/tests/subsys/logging/log_msg2/src/main.c:12:
zephyr/include/generated/syscalls/log_msg2.h: In function 'z_log_msg2_static_create.constprop':
zephyr/include/generated/syscalls/log_msg2.h:25:20: note: parameter passing for argument of type 'const struct log_msg2_desc' changed in GCC 9.1
   25 | static inline void z_log_msg2_static_create(const void * source, const struct log_msg2_desc desc, uint8_t * package, const void * data)
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
[36/129] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/logging/log_msg2.c.obj
/home/user/projects/open/zephyr/zephyr/subsys/logging/log_msg2.c: In function 'z_log_msg2_finalize':
/home/user/projects/open/zephyr/zephyr/subsys/logging/log_msg2.c:12:6: note: parameter passing for argument of type 'const struct log_msg2_desc' changed in GCC 9.1
   12 | void z_log_msg2_finalize(struct log_msg2 *msg, const void *source,
      |      ^~~~~~~~~~~~~~~~~~~
[... 174 lines cut ...]
timestamp: 287454020
-------------------end of printing message-----------------
 PASS - test_log_msg2_string_and_data in 0.3 seconds
===================================================================
START - test_log_msg2_fp
 PASS - test_log_msg2_fp in 0.1 seconds
===================================================================
START - test_mode_size_plain_string
 PASS - test_mode_size_plain_string in 0.1 seconds
===================================================================
START - test_mode_size_data_only
[... cut ...]
 PASS - test_mode_size_plain_str_data in 0.1 seconds
===================================================================
START - test_mode_size_str_with_2strings
 PASS - test_mode_size_str_with_2strings in 0.1 seconds
===================================================================
START - test_saturate
 PASS - test_saturate in 0.1 seconds
===================================================================
Test suite test_log_msg2 succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

npitre, 2021-05-05:
> @npitre this seems to be introduced with f1f63dda176f8bce7a8cebe50c30ebe01262b798.  Not sure if your #34793 might address this.

No. #34793 won't do anything here.

galak, 2021-05-05:
> No. #34793 won't do anything here.

Yeah, I was hoping it might magically fix this issue. :)

npitre, 2021-05-05:
The first warning can be reproduced with this small test:

```
struct foo {
	short a : 3;
	short b : 3;
	short c : 10;
};

void bar(struct foo f) { }
```

This produces the following:

```
$ aarch64-zephyr-elf-gcc -c test.c
test.c: In function 'bar':
test.c:7:6: note: parameter passing for argument of type 'struct foo' changed in GCC 9.1
    7 | void bar(struct foo f) { }
      |      ^~~
```

It appears that the only way to shut it up is to use `-Wno-psabi`.
However this could have unexpected impact elsewhere if people link against
libraries that might have that a mismatched ABI. Some project-wide policy
might be needed here.

dcpleung, 2021-05-05:
The warning about GCC 9.1 does not result in twister failure. It's just annoyance during build.

galak, 2021-05-05:
we're able to build, but the test fails to run correctly in qemu is what I was seeing.

npitre, 2021-05-07:
A few things look wrong here.

`Z_LOG_MSG2_CREATE2()` is passed a `char str[256]` making
`CBPRINTF_MUST_RUNTIME_PACKAGE()` to always be true. The resulting mode is
`RUNTIME` and not `ZERO_COPY` which fails the first `zassert_equal()`.

Then, removing the `str` argument makes it pass, but the test fails further
down. It turns out that the static package produces a payload of 56 bytes
whereas the runtime package is 64 bytes. The hex dump shows 8 extra zero bytes
in the middle, probably due to some padding.

I tried `twister -p qemu_cortex_a53 -T tests/lib/cbprintf_package` to
validate proper padding around doubles. This is a simpler test without the
the logging stuff. But that failed to build for yet another reason:

```
zephyr/tests/lib/cbprintf_package/src/test.inc:
In function 'void test_cbprintf_package()':
zephyr/include/sys/cbprintf_internal.h:244:15: error: non-constant condition for static assertion
  244 |  BUILD_ASSERT(!((sizeof(double) < VA_STACK_ALIGN(long double)) && \
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  245 |    Z_CBPRINTF_IS_LONGDOUBLE(_arg) && \
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  246 |    !IS_ENABLED(CONFIG_CBPRINTF_PACKAGE_LONGDOUBLE)),\
      |    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

WTF?? Here `CONFIG_CBPRINTF_PACKAGE_LONGDOUBLE` is not even enabled.
In fact the whole build.log file has 7429 lines in it!
The amount of macro recursion is simply overwelming and gcc list them all.

There is a bit too much magic in this static packing code.

@nordic-krch You are the magician. Could you please have a look?

npitre, 2021-05-07:
This appears to be fixed with #34951

PR #34951 (2021-05-07): lib: os: cbprintf: Fixes for c++ and coverity
Fixes for C++ includes:
    - avoid calling static inline template function in static
    assert on certain platforms which consider it non const expression
    - Add 0 to variable before calling C++ argument storing function
    to promote a variable.
    
Fix for coverity:
    - avoid using sizeof(n + 0) by using local variable (gcc extension)

Additionally removed intergration platform from `tests/lib/cbprintf_package` since it is platform specific test.

Fixes #34925.
Fixes #34838.

It also suppose to fix 14K coverity issues.
closed 2021-05-07 by galak