zephyrproject-rtos/zephyr #34844: qemu_cortex_a53_smp:  tests/ztest/error_hook failed after enabling the FPU context switching support

enjiamai, 2021-05-05:
**Describe the bug**
The testcase tests/ztest/error_hook failed after the FPU context switching support.  This failure is not 100% reproduced.
And after git bisect, we found the issue started after commit:   a82fff04ff9ce3bbb126e4f8e7af9c0fc4003c4f. 

**To Reproduce**
Steps to reproduce the behavior:
1. run command:   twister -T tests/ztest/error_hook/ -p qemu_cortex_a53_smp

**Expected behavior**
A test case pass as expected.

**Impact**
It blocked the CI where the submitted PR run testcase tests/ztest/error_hook. 

**Logs and console output**
```
===================================================================
START - test_catch_z_oops
```
or
```
===================================================================
START - test_catch_z_oops
E: ELR_ELn: 0x000000004001d000
E: ESR_ELn: 0x0000000000000008
                                                                                                                                                                                    
```

**Environment (please complete the following information):**
 - OS: ubuntu 18.04
 - Toolchain Zephyr SDK 0.12.4
 - Commit 86851434c8372fdd1f4f078d92a6e72a92d5ac0d, 

**Additional context**
N/A

enjiamai, 2021-05-05:
Hi @nashif, I found this is due to the assertion fail from ISR in the previous executed test case, and it might not recover all the status of the program.  So I put the test_catch_assert_in_isr() in the last to prevent it affects the following test case.

galak, 2021-05-05:
Possibly whatever is happening here might impact #34838

npitre, 2021-05-05:
> Possibly whatever is happening here might impact #34838

I doubt it. #34838 is a toolchain issue that got exposed for the first 
time when ARM64 started having CONFIG_FPU=y.

Here this is most likely something to do with the interaction between FP 
context handling code and the exception mode.

And in this case I'm unable to reproduce when building and running the 
test via west. It requires twister to fail. What's the difference?

enjiamai, 2021-05-05:
> And in this case I'm unable to reproduce when building and running the test via west. It requires twister to fail. What's the difference?

Hi, @npitre , in this case west command needs a extra -DCONFIG_TEST_USERSPACE=y to keep in concert with twister.
After looking at #34838, I'm not very sure, but it seems like a different issue, @galak, thanks!

npitre, 2021-05-06:
> Hi, @npitre , in this case west command needs a extra -DCONFIG_TEST_USERSPACE=y to keep in concert with twister.

That doesn't work here. I tried:

```
west build tests/ztest/error_hook/ -b qemu_cortex_a53_smp -t run -p always -DCONFIG_TEST_USERSPACE=y
```

And I verified that `build/zephyr/.config` does include 
`CONFIG_USERSPACE=y`.

Then I even copied 
`twister-out/qemu_cortex_a53_smp/tests/ztest/error_hook/testing.ztest.error_hook/zephyr/.config` 
onto `build/zephyr/.config` to make sure I have the same config.

I just can't make it to fail with `west build -t run` at all while it 
fails with twister.

I'm rather puzzled.

enjiamai, 2021-05-06:
Hi @npitre , I can reproduce it on my side. I just use the command:
```
west build tests/ztest/error_hook/ -b qemu_cortex_a53_smp -p always -t run -DCONFIG_TEST_USERSPACE=y
```
Did you use the same command as mine? thanks!

npitre, 2021-05-06:
On Thu, 6 May 2021, enjia mai wrote:

> Hi @npitre , I can reproduce it on my side. I just use the command:
> ```
> west build tests/ztest/error_hook/ -b qemu_cortex_a53_smp -p always -t run -DCONFIG_TEST_USERSPACE=y
> ```
> Did you use the same command as mine? thanks! 

I just did. And I get:

```
PROJECT EXECUTION SUCCESSFUL
```

enjiamai, 2021-05-06:
Excuse me, @npitre, could you please tell me the commit ID you use, let me also have a try, thanks~

npitre, 2021-05-06:
> Excuse me, @npitre, could you please tell me the commit ID you use, let me also have a try, thanks~

Commit 5f423d45cfa1 (before the test order swap workaround was merged).

enjiamai, 2021-05-06:
Here are my trials of running by "west build tests/ztest/error_hook/ -b qemu_cortex_a53_smp -p always -t run -DCONFIG_TEST_USERSPACE=y"

f5e3d89e326567c2aa345abb6d6b4f28fb497b37 (ztest order workaround)  west 20/20 pass, by twister pass
2b864edbcd1d6f7dfbf8c9797a845c05fd344456 (last before workaround)  west 20/15 pass, by twister failed
5f423d45cfa1fdfe4a543a0b773f8f8eb3048109 (commit you are using)  west 20/19 pass, by twister failed
f1f63dda176f8bce7a8cebe50c30ebe01q262b798 (enable FPU context) west 20/9 pass, by twister failed
9136c408339b95b10a5ac54dd3eeb2dd189a19d5 (last before enable FPU)  west 20/20 pass, by twister pass

It was sometimes to reproduce very easily, but sometimes not, by west.  By twister, it's very easy to reproduce.
I also compare their .config between west and twister, but they are the same.

I think this needs some time to take look at it.

npitre, 2021-05-06:
Let's keep this open until we can explain the root issue.

enjiamai, 2021-05-06:
@npitre But I also want to highlight that when an assertion happened, especially in ISR, in a normal situation program should be terminated. Recover it here then keep executing the program only for testing purposes.

npitre, 2021-05-06:
> @npitre But I also want to highlight that when an assertion happened, especially in ISR, in a normal situation program should be terminated. Recover it here then keep executing the program only for testing purposes. 

Agreed. However, in this case, the fact that it sometimes works and 
sometimes doesn't is worrisome. That is indicative of a race somewhere 
and we better find it.

enjiamai, 2021-05-17:
Hi @npitre, why the test case sometimes failed is because when we intentionally trigger an assertion in ISR, then we did not leave ISR context gracefully. Though we release some resources such as semaphore or spinlock before the test_main thread end, it is still in ISR context ztest thread until some point. Currently, our ztest error hook does not have a mechanism to return back from the ISR and recover the stack. 

For most of the platforms, this might be just enough to pass the test case. But in the SMP environment, it is possible to cause a problem if the ISR context does not exit immediately and another fatal error trigger at the same time.

Though put this kind of testcase in the last, or make the testcase run with 1cpu can avoid this but the best solution here I think is, to remove this kind of usage unless we have a good recovery mechanism for the testcases.
So I submitted a #35353 to remove this testcase, thanks.

andyross, 2021-05-19:
Came here to comment that this sounds like almost exactly the issue detailed in #35307, but it looks like @npitre is already here and aware.

enjiamai, 2021-05-19:
@andyross Thank you so much!  Like you just mentioned in #35307 , they might be the same cause. I verified the related PR #35410 will fix this issue.  @npitre .

npitre, 2021-05-21:
#35410 is now merged

PR #34846 (2021-05-06): tests: ztest: fix testcase might fail when assertion fail in ISR
Put the testcase test_catch_assert_in_isr() to execute last, to prevent
it affects other test cases. Because when we caught an assert failure
in the ISR handler, it cannot be guaranteed that all the current
program status would be recovered.

Fixes #34844.
commit f5e3d89e (2021-05-06): tests: ztest: fix testcase might fail when assertion fail in ISR

Put the testcase test_catch_assert_in_isr() to execute last, to prevent
it affects other test cases. Because when we caught an assert failure
in the ISR handler, it cannot be guaranteed that all the current
program status would be recovered.

Fixes #34844.
closed 2021-05-21 by npitre