zephyrproject-rtos/zephyr #34852: Some bluetooth advertising packages never get transmitted over-air (Bluetooth Mesh application)

isakwestin, 2021-05-05:
**Describe the bug**
My setup is a Nordic nRF52832-board running the zephyr hci_uart sample which is connected via UART with the host, a GNU/Linux aarch64. The host is running the bluetooth-mesh-daemon proveded by BlueZ and I am using the tool `mesh-cfgclient`, also provided by BlueZ, that is using the DBUS-api provided by the daemon.

I have noticed that the reliability of mesh-messages reaching its destination is quite low. And when using a sniffer to monitor the bluetooth traffic, it looks like many of the messages (i.e. advertising packages, the mesh-daemon only uses the ADV-bearer) are never actually transmitted over-air. Only about 40% of the messages are transmitted.

The HCI-communication between controller and host looks healthy and does not report any issues or failures. An example of the sequence that the mesh-daemon uses for sending a mesh-message is shown below (captured with btmon).

```
< HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15                #366 [hci0] 12.009032
        Min advertising interval: 100.000 msec (0x00a0)
        Max advertising interval: 100.000 msec (0x00a0)
        Type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Own address type: Random (0x01)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from White List Only, Allow Connect Request from White List Only (0x03)
> HCI Event: Command Complete (0x0e) plen 4                                       #367 [hci0] 12.011885
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32                      #368 [hci0] 12.012075
        Length: 25
        Mesh Data: 0b27cd18256f81ae41163d31c1bb446fea5a391bea95e4
          IVI: 0
          NID: 0x0b
        27 cd 18 25 6f 81 ae 41 16 3d 31 c1 bb 44 6f ea  '..%o..A.=1..Do.
        5a 39 1b ea 95 e4                                Z9....          
> HCI Event: Command Complete (0x0e) plen 4                                       #369 [hci0] 12.016376
      LE Set Advertising Data (0x08|0x0008) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                       #370 [hci0] 12.016514
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                       #372 [hci0] 12.020959
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                       #377 [hci0] 12.108042
        Advertising: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                       #378 [hci0] 12.109763
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15                #379 [hci0] 12.109890
        Min advertising interval: 100.000 msec (0x00a0)
        Max advertising interval: 100.000 msec (0x00a0)
        Type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Own address type: Random (0x01)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from White List Only, Allow Connect Request from White List Only (0x03)
> HCI Event: Command Complete (0x0e) plen 4                                       #380 [hci0] 12.112711
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32                      #381 [hci0] 12.112839
        Length: 25
        Mesh Data: 0b27cd18256f81ae41163d31c1bb446fea5a391bea95e4
          IVI: 0
          NID: 0x0b
        27 cd 18 25 6f 81 ae 41 16 3d 31 c1 bb 44 6f ea  '..%o..A.=1..Do.
        5a 39 1b ea 95 e4                                Z9....          
> HCI Event: Command Complete (0x0e) plen 4                                       #382 [hci0] 12.117134
      LE Set Advertising Data (0x08|0x0008) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                       #383 [hci0] 12.117259
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                       #384 [hci0] 12.120972
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                       #389 [hci0] 12.208141
        Advertising: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                       #390 [hci0] 12.209806
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                         #391 [hci0] 12.209945
        Address: FF:A9:66:35:89:DA (Static)
> HCI Event: Command Complete (0x0e) plen 4                                       #392 [hci0] 12.211977
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
```

**What I have tried:**
I have found that if the message is sent or not, basically relies on this if-statement evaluating to true:
https://github.com/zephyrproject-rtos/zephyr/blob/ebfe9be01f4da55de8f5876c64b2ca79fb04ada2/subsys/bluetooth/controller/ll_sw/ull_adv.c#L1729

 - Enable `CONFIG_BT_TICKER_LOW_LAT`: If-statement will always be true, however the function `ticker_cb` is not always entered -> ~70% success rate
 - Disable `CONFIG_BT_TICKER_LOW_LAT `and force the if-statement to true (`if(1)`): Always enters function `ticker_cb` -> ~98% success rate. This is working quite well, but obviously I don't know any side effects this might introduce.

Success rate is based on sending a mesh-message and successfully receiving a response, for a number of times.

I'm hoping that someone here can help me understand why so many of the advertising requests are actually never advertised, if there are some configurations that would improve this that I don't know of, or perhaps what potential side effects it might have to force the if-statement to true.

**Environment:**
 - SHA: ebfe9be01f4da55de8f5876c64b2ca79fb04ada2
 - Toolchain: Zephyr SDK 0.12.3
 - Host: GNU/Linux aarch64 running BlueZ bluetooth mesh daemon and mesh-cfgclient tool
 - BlueZ version 5.55
 - Controller nRF52832 running  zephyr hci_uart sample and is used only for the mesh application

cvinayak, 2021-05-05:
> Disable CONFIG_BT_TICKER_LOW_LAT and force the if-statement to true (if(1)): Always enters function ticker_cb -> ~98% success rate. This is working quite well, but obviously I don't know any side effects this might introduce.

Radio Rx utilization will be low, and hence will lower number of advertising reports will be generated. 

Please provide details of reproducing your issue, steps to build, steps and commands to use to reproduce the issue, how was the statistics calculated, if there has been changes to code etc., preferably in the format requested in the issue template that was provided when creating this issue.

trond-snekvik, 2021-05-06:
> Also, correct me, the advertisement are enabled and disabled in less than 100 ms?

Depends on the configuration, but often, yes.

isakwestin, 2021-05-06:
Thank you for the reply!

`CONFIG_BT_CTLR_SCAN_UNRESERVED=y` seems to have been the missing piece in my case. Enabling it made a big difference and for now it works as expected. I didn't notice this configuration flag until now, so thanks for mentioning it.

For curiousity:
Is there any summary or specific documentation anywhere on how to use/configure a controller-only build for mesh-only application? If there is I would be interested to know.

closed 2021-05-06 by isakwestin