zephyrproject-rtos/zephyr #34865: CONFIG_NET_SOCKETS_PACKET interferes with other network traffic (gptp, IP)

EinarHenriksen, 2021-05-05:
**Describe the bug**
I want to send Ethernet packets (AF_PACKET, SOCK_DGRAM) from a BSD socket while also running gPTP on my system, but this has proven problematic.
In trying to debug this, I have simplified my code into an empty main file where only changing one line in prj.conf is enough to trigger the issue.
Here is the content of my prj.conf file (it is very similar to the prj.conf of the gptp sample):
```
CONFIG_NETWORKING=y
CONFIG_NET_LOG=y
CONFIG_LOG=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_STATISTICS=y
CONFIG_NET_STATISTICS_USER_API=y

CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_NET_PKT_RX_COUNT=32
CONFIG_NET_PKT_TX_COUNT=32
CONFIG_NET_BUF_RX_COUNT=32
CONFIG_NET_BUF_TX_COUNT=32
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=5
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=5
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1
CONFIG_NET_MAX_CONTEXTS=10

CONFIG_INIT_STACKS=y
CONFIG_PRINTK=y
CONFIG_NET_SHELL=y

# Ethernet is needed for gPTP
CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_SETTINGS=y

CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV4_GW="169.254.202.192"

# gPTP settings
CONFIG_NET_GPTP=y
CONFIG_NET_GPTP_STATISTICS=y
CONFIG_NET_GPTP_GM_CAPABLE=y

#CONFIG_NET_GPTP_LOG_LEVEL_DBG=y
#CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y
#CONFIG_NET_CONTEXT_LOG_LEVEL_DBG=y
#CONFIG_NET_CONN_LOG_LEVEL_DBG=y
#CONFIG_NET_IF_LOG_LEVEL_DBG=y
#CONFIG_NET_L2_ETHERNET_LOG_LEVEL_DBG=y

# How many traffic classes to enable
CONFIG_NET_TC_TX_COUNT=6
CONFIG_NET_TC_RX_COUNT=6

# Enable priority support in net_context
CONFIG_NET_CONTEXT_PRIORITY=y
```

With this setup, gPTP works as expected, and my system replies to pings.
Now if _both_ of these lines are included:
```
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_PACKET=y
```
then my system answers neither gPTP requests or pings.
This is problematic because these contain the functionality I need to set up a BSD socket in my code.
It seems like a callback in the sockets code receives all the ingoing packets, instead of the modules that need them, from what I have observed so far.

**To Reproduce**
I am using a FRDM K64F connected directly to my computer with an ethernet cable
The code was compiled and flashed to the board using West.
The network traffic between the board and my computer was then observed in Wireshark.
I am running the gptp host on my computer as described here:
https://docs.zephyrproject.org/2.5.0/samples/net/gptp/README.html#setting-up-linux-host

**Expected behavior**
I expect gptp traffic and ping requests to still be handled normally after I've set my necessary socket options.

**Impact**
Setting the necessary options to set up an ethernet packet socket diverts other incoming network traffic.
This makes me unable to set up a socket like this while also running gPTP on my system.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Toolchain Zephyr v. 2.5.99
 - Platform: NXP FRDM K64F

jukkar, 2021-05-06:
@EinarHenriksen could you test the fix proposal at #34899 and report back if your use case works with it.
I did test the PR and everything seems to work as expected but packet socket part of the code is quite convoluted and it is difficult to see all the corner cases.

EinarHenriksen, 2021-05-06:
I'm testing with your code now, and in the basic example this has solved the problem for me too.
Thank you very much for the quick response :)

Though in my application where I set up a BSD socket and send ethernet packets, after a connection has been established between my linux gptp host and the gptp server on my board this warning gets printed for each incoming packet afterwards:
`net_conn: pkt cloning failed, pkt <packet pointer> dropped`

At this point the packets no longer reach their destination and the system behaves like earlier.
If I don't start the gptp host, I am able to ping the board.

jukkar, 2021-05-06:
> net_conn: pkt cloning failed, pkt <packet pointer> dropped

This means that you run out of net_pkt, you need to increase the number of `CONFIG_NET_PKT_RX_COUNT` and probably `CONFIG_NET_BUF_RX_COUNT` too.

EinarHenriksen, 2021-05-07:
ok so I increased pkt_rx_count and buf_rx_count from 32 to 512, filling up my sram to 97%, and now it fails in 30 seconds instead of 3 seconds after the gptp connection is established

jukkar, 2021-05-08:
> now it fails in 30 seconds instead of 3 seconds after the gptp connection is established

That sounds like a buffer leak. It is easy to verify that if you enable `CONFIG_NET_BUF_POOL_USAGE` and optionally also `CONFIG_NET_DEBUG_NET_PKT_ALLOC`. Then the network buffer usage can be seen in net shell by `net mem` and `net allocs` commands.

EinarHenriksen, 2021-05-11:
I've done some testing with this now, and observed a buffer leak while running my code.
It seems like all I have to do to trigger the buffer leak is to create a socket like this:
`s = zsock_socket(AF_PACKET, SOCK_DGRAM, 0x22f0);`
Then incoming gptp packets will fill up the rx buffer, so my guess would be the packets are never dereferenced for some reason?
If I do nothing in my main function, but build with the same setup otherwise, the buffer leak does not happen.

closed 2021-05-25 by nashif

EinarHenriksen, 2021-05-25:
I revisited this today, and it seems like what is needed to trigger the buffer leak is to create a socket _and_ to send at least two messages on the socket, while either pinging or sending gptp messages to the board.
Meaning that this triggers a buffer leak for me:
```
[...]
void main(void) {
        [...]
        s = zsock_socket(AF_PACKET, SOCK_DGRAM, 0x22f0);
        [...]
        zsock_sendto(s, pdu, PDU_SIZE, 0, (struct sockaddr *)&addr, sizeof(addr));
        k_sleep(K_SECONDS(10));
        zsock_sendto(s, pdu, PDU_SIZE, 0, (struct sockaddr *)&addr, sizeof(addr));
        while(1){k_sleep(K_SECONDS(10));}
        return;
}
```
While this does not:
```
[...]
void main(void) {
        [...]
        s = zsock_socket(AF_PACKET, SOCK_DGRAM, 0x22f0);
        [...]
        zsock_sendto(s, pdu, PDU_SIZE, 0, (struct sockaddr *)&addr, sizeof(addr));
        while(1){k_sleep(K_SECONDS(10));}
        return;
}
```
After the second message is sent, the incoming messages fill up the rx buffer.

jukkar, 2021-05-25:
@EinarHenriksen thanks for the info. I will investigate this.

jukkar, 2021-05-25:
This is actually the same case that in issue #34462 and the fix in #34475 (see the discussion there). In short, the application needs to call `recv()` in order to receive and "free" the received packets. If the recv is not called, the incoming packets for AF_PACKET sockets keep piling up and are never freed.

EinarHenriksen, 2021-05-26:
OK, so if I understand this correctly, every incoming packet gets sent to my socket, and then copied to where it is supposed to go?
And in order to avoid this buffer leak I am supposed to regularly call recv() in my application to free up the packet buffer?
Seems like a very backwards way of solving the problem, why would my socket recieve ping- and gptp-packets at all when I have explicitly declared my socket as an avtp packet socket?
I see you are talking about setting up a send-only socket in [#34475](https://github.com/zephyrproject-rtos/zephyr/pull/34475), which is exactly what I am trying to do, but having to call recv() regularly in my application isn't really a satisfactory solution to that problem in my opinion.
And why does the buffer leak only start happening after I have sent two packets from my socket?

jukkar, 2021-05-26:
> OK, so if I understand this correctly, every incoming packet gets sent to my socket, and then copied to where it is supposed to go?
> And in order to avoid this buffer leak I am supposed to regularly call recv() in my application to free up the packet buffer?

Yes, that is the case atm.

> Seems like a very backwards way of solving the problem, why would my socket recieve ping- and gptp-packets at all when I have explicitly declared my socket as an avtp packet socket?

Yes, there could be optimization possibilities here if you are only interested in certain type packets. But for example if the application has created ETH_P_ALL type packet socket, then it needs to call recv in order to flush the incoming socket queue because the core IP does not know atm if the application is only sending data or is listening too.

> And why does the buffer leak only start happening after I have sent two packets from my socket?

I was not able to replicate the issue in your example code. I modified the packet socket sample app according to your example above and only saw the "leak" if I sent one packet, if I sent two packets, the "leak" disappeared. This could be related to my modifications to the sample app.

> I see you are talking about setting up a send-only socket in [#34475](https://github.com/zephyrproject-rtos/zephyr/pull/34475), which is exactly what I am trying to do, but having to call recv() regularly in my application isn't really a satisfactory solution to that problem in my opinion.

The packet socket code is a bit convoluted and would need an overhaul in order to avoid these weird issues. The code is improving all the time, it was not long time ago when after enabling packet socket, one was not able to receive normal IP packets at all. If you have good ideas how to improve the packet socket handling, or working PR would be even better, it would be great.

EinarHenriksen, 2021-05-26:
Then I will try using recv() as a temporary solution for now.
It seems like most of these problems would be solved by only recieving the specified protocol on a socket though.
I would very much like to know what causes the leak to stop and start when sending packets if you find out, which seems to be the only undefined behaviour/actual bug in my application right now.

PR #34899 (2021-05-25): net: packet: Do not drop net_pkt immediately
If there are no sockets in the system, then do not drop the
packet immediately as there can be other L2 network handlers
like gPTP in the system. This will also allow ICMP messages
to pass to local handler.

Fixes #34865
commit 46efe3ed (2021-05-25): net: packet: Do not drop net_pkt immediately

If there are no sockets in the system, then do not drop the
packet immediately as there can be other L2 network handlers
like gPTP in the system. This will also allow ICMP messages
to pass to local handler.

Fixes #34865