zephyrproject-rtos/zephyr #34872: Question about adding CANFD support to 6LoCAN

ehunck, 2021-05-05:
This is really a question related to an enhancement that I am working on which is adding CANFD support to 6LoCAN.

**Is your enhancement proposal related to a problem? Please describe.**
My question is specifically related to updating [6locan.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/canbus/6locan.c) to support 64 byte frames for a CANFD enabled driver. My test setup uses two targets running the echo_server and echo_client examples with the 6LoCAN overlays and the UDP and TCP communications turned off.  I am using the UART shell to ping the one device from the other.  I have run into a few minor things that needed updating (specifically related to the expanded meaning of the DLC bits in CANFD vs CAN).  But, I have recently run into an issue in the `canbus_finish_pkt()` function.

Below is this function:
```c
static enum net_verdict canbus_finish_pkt(struct net_pkt *pkt)
{
	/* Pull the ll addresses to ignore them in upper layers */
	net_buf_pull(pkt->buffer, net_pkt_lladdr_dst(pkt)->len +
		     net_pkt_lladdr_src(pkt)->len);

	if (IS_ENABLED(CONFIG_NET_L2_CANBUS_ETH_TRANSLATOR) &&
	    canbus_is_for_translator(pkt)) {
		/* Pull room for address extension */
		net_buf_pull(pkt->buffer, sizeof(struct net_eth_addr) -
			     net_pkt_lladdr_src(pkt)->len);
		/* Set the destination address to the inline MAC and pull it */
		net_pkt_cursor_init(pkt);
		net_pkt_lladdr_dst(pkt)->addr = net_pkt_cursor_get_pos(pkt);
		net_pkt_lladdr_dst(pkt)->type = NET_LINK_ETHERNET;
		net_pkt_lladdr_dst(pkt)->len = sizeof(struct net_eth_addr);
		net_buf_pull(pkt->buffer, sizeof(struct net_eth_addr));
	}

	net_pkt_cursor_init(pkt);
	if (!net_6lo_uncompress(pkt)) {
		NET_ERR("6lo uncompression failed");
		return NET_DROP;
	}

	net_pkt_cursor_init(pkt);

	return NET_CONTINUE;
}
```

When using a 64 byte single frame packet, I find that the lines:
```c
/* Pull the ll addresses to ignore them in upper layers */
net_buf_pull(pkt->buffer, net_pkt_lladdr_dst(pkt)->len +
		     net_pkt_lladdr_src(pkt)->len);
```
end up pulling the compressed header off of the network packet.  This happens immediately before we try to `net_6lo_uncompress(pkt)`.  This causes that function to not recognize either of the compression headers: IPHC or IPV6.  Then the uncompress function just returns true assuming that the packet is not compressed `NET_DBG("pkt %p is not compressed", pkt);`. 

I have removed those lines in the version that I am testing and the uncompression works fine and the ping is successful with data frames up to 64 bytes.  **What is the original purpose of the line `net_buf_pull(pkt->buffer, net_pkt_lladdr_dst(pkt)->len +net_pkt_lladdr_src(pkt)->len);`?**  What bytes in the stack are they intended to discard?

**Describe the solution you'd like**
Some context for what these lines are intended to do.

**Describe alternatives you've considered**
The alternative solution that is working for me is removing the lines.  I would like to know why those lines are originally present in order to know if removing these is the right solution or if there is still some purpose for these lines that I am missing.

**Additional context**
None.

alexanderwachter, 2021-05-11:
I think I got the problem. In case it is not a SF, we need to pull the address, because the address was put there from the canbus_cpy_lladdr. In case of a SF, the address is already pulled.
Maybe we can pull the address directly after the copy. Need to check that.

henrikbrixandersen, 2021-12-06:
@ehunck @alexanderwachter Can this question be closed?

alexanderwachter, 2021-12-06:
Think so.

closed 2021-12-06 by alexanderwachter