zephyrproject-rtos/zephyr #34879: mec15xxevb_assy6853: tests/boards/mec15xxevb_assy6853/qspi/ multiple test failures

yerabolu, 2021-05-05:
**Describe the bug**
Multiple tests are failing on mec15xxevb_assy6853 board under tests/boards/mec15xxevb_assy6853/qspi/

**To Reproduce**
Steps to reproduce the behavior:
1. twister --hardware-map ~/mec1.file --device-testing -T tests/boards/mec15xxevb_assy6853/qspi/ 
2. See error

**Expected behavior**
All the tests should pass

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.5.0-3177-gfaf68fb213bc  ***
Running test suite test_spi
===================================================================
START - test_spi_device
I: CS control inhibited (no GPIO device)
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:94: test_spi_device: (jedec_id == W25Q128_JEDEC_ID is false)
JEDEC ID doesn't match
FAIL - test_spi_device in 0.20 seconds
===================================================================
START - test_spi_sector_erase
PASS - test_spi_sector_erase in 0.1 seconds
===================================================================
START - test_spi_single_write
PASS - test_spi_single_write in 0.1 seconds
===================================================================
START - test_spi_single_read
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:273: test_spi_single_read: (memcmp(buffer_tx, safbuf2, BUF_SIZE) == 0 is false)
Buffer read data is different to write data
FAIL - test_spi_single_read in 0.20 seconds
===================================================================
START - test_spi_dual_read
I: CS control inhibited (no GPIO device)
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:341: test_spi_dual_read: (memcmp(buffer_tx, safbuf2, BUF_SIZE) == 0 is false)
Buffer read data is different to write data
FAIL - test_spi_dual_read in 0.23 seconds
===================================================================
START - test_spi_quad_write
I: CS control inhibited (no GPIO device)
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:449: test_spi_quad_write: ((spi_status2 & SPI_STATUS2_QE) == SPI_STATUS2_QE is false)
Enable QSPI mode failure
FAIL - test_spi_quad_write in 0.22 seconds
===================================================================
START - test_spi_quad_read
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:579: test_spi_quad_read: (ret == 0 is false)
Receive fast read data spi_transceive failure: error -35
FAIL - test_spi_quad_read in 0.18 seconds
===================================================================
START - test_spi_octal_read
Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/qspi/src/main.c:645: test_spi_octal_read: (ret == 0 is false)
Send quad read address spi_transceive failure: error -35
FAIL - test_spi_octal_read in 0.30 seconds
===================================================================
Test suite test_spi failed.
===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Fedora 29
 - Toolchain: Zephyr SDK 12.4
 - Commit SHA: faf68fb213bc

**Additional context**
Tests were passing before, after bisecting tests started failing after this commit: 5e1502e816d7e33cf104ac526f6567f966b169f8

yerabolu, 2021-05-19:
@scottwcpg: Any update on this?

scottwcpg, 2021-05-19:
SPI test passed on my board. Please check the SPI flash part is a Winbond W25Q128 with JEDEC ID that matches the JEDEC ID hard coded in the test: 0x001840ef.  If you are loading the Zephyr test app from the same SPI flash as the test then it should be able to read JEDEC ID.  For quad you must make sure the jumpers for SHD_IO2 and SHD_IO3 are correct:
SHD_IO2 JP98 connect 8-9
SHD_IO3 JP98 connect 11-12

I2C_API test requires additional jumper settings: 
JP54 1-2 connected supplies +3.3V to PCA9555
Test uses I2C address 0x26 for PCA9555 requiring:
JP53 1-2 connected sets PCA9555 I2C address A0=0
JP53 3-4 unconnected sets PCA9555 I2C address A1=1
JP53 5-6 unconnected sets PCA9555 I2C address A2=1

Have not looked at GPIO test

scottwcpg, 2021-05-19:
@yerabolu Please review the above debug results

yerabolu, 2021-05-24:
@scottwcpg: will look into it and get back to you.

yerabolu, 2021-06-01:
@scottwcpg: Verified the jumper settings for SPI and I2C_API. All the settings are as you mentioned above and tests are passing now on my setup as well. 

But the GPIO tests are still failing. If the hack causes a race condition, should we look at the test to see if there is another way to test it?

scottwcpg, 2021-06-01:
@yerabolu All I can think of is to add delay in the test after enabling the GPIO interrupt, check the globals the ISR uses, if they are set clear them and proceed to the loop.  If the delay is several hundred milliseconds that should be longer than the RC time constant of the pin load.  But are there cases with other chips if the ISR fires before the test loop a real error should be reported?

yerabolu, 2021-06-04:
> @yerabolu All I can think of is to add delay in the test after enabling the GPIO interrupt, check the globals the ISR uses, if they are set clear them and proceed to the loop. If the delay is [...]

@scottwcpg: Not sure if there are any other cases. Trying to investigate further.

albertofloyd, 2021-07-07:
@yerabolu could you clarify if this affects Zephyr 2.5 onwards or only the main branch?
I'm asking since traces indicate  *** Booting Zephyr OS build zephyr-v2.5.0-2839-gb504f8b6a27b  ***

Do you have any bisection data? i.e. do we know if anything pointing to any recent changes to gpio/qspi driver or related tests?

yerabolu, 2021-07-07:
> @yerabolu could you clarify if this affects Zephyr 2.5 onwards or only the main branch?
> I'm asking since traces indicate *** Booting Zephyr OS build zephyr-v2.5.0-2839-gb504f8b6a27b ***
>
> Do you have [...]

@albertofloyd : It is observed on the main branch as well. Test started failing after this commit 5e1502e.

albertofloyd, 2021-07-07:
Thanks @yerabolu, seems this doesn't affect Zephyr 2.5 release but only Zephyr 2.6 and onwards
@scottwcpg I recall you mention latest a HAL change required a pinmux driver update, is this the HAL update you were referring? 

If so is the pinmux driver update in place?

yerabolu, 2021-07-07:
> @scottwcpg: Not sure if there are any other cases. Trying to investigate further.

@scottwcpg: Tried investigating and not sure if this helps - only mps2_an385, mps2_an521 platforms are excluded for this test. And test passes on other platforms. It is failing on Microchip.

maksimmasalski, 2021-07-22:
@scottwcpg Any updates about that problem?

scottwcpg, 2021-07-22:
@maksimmasalski  I added a fix to the GPIO driver in PR 37138 and added you as a reviewer.  The fix worked on my MEC152x EVB.

maksimmasalski, 2021-07-23:
> @maksimmasalski I added a fix to the GPIO driver in PR 37138 and added you as a reviewer. The fix worked on my MEC152x EVB.

Thank you, Scott. Will review it.

closed 2021-08-05 by carlescufi

albertofloyd, 2021-08-18:
@nashif  how do we get this backported to v2.6? 

@scottwcpg could you confirm what are the actual dependencies for this fix? 
https://github.com/zephyrproject-rtos/zephyr/pull/37479 mentions PR's 37138 and 37139.

scottwcpg, 2021-08-18:
@albertofloyd PR #37479 is the final fix for both MEC152x and MEC172x.  Previous PR's using DMB() were not a long term solution.

maksimmasalski, 2021-09-02:
@nashif seems like it was not backported to the 2.6. @scottwcpg could you backport it to 2.6? If not, I will backport.

PR #37138 (2021-08-05): drivers: GPIO: Microchip: Fix GPIO interrupt enable spurious interrupt
Microchip MEC GPIO hardware can trigger a spurious interrupt when
interrupt detection is set to edge mode especially falling edge mode.
Clearing the status immediately after enabling interrupt detection does
not work because the hardware takes a small number of AHB clocks to
set the status. After interrupt detection enable we spin for a small
number of cycles reading a hardware register to generate delay. By this
time the spurious status is present and cleared before enabling the
interrupt in the GIRQ.

Fixes #34879.