zephyrproject-rtos/zephyr #34880: Convert SoF Module to new kwork API

galak, 2021-05-05:
Convert modules/audio/sof/src/schedule/timer_domain.c to use new kwork API.

See the following issue for details:

#33104

galak, 2021-05-05:
When setting the old kwork api's as deprecated we get:
```

ccache /opt/toolchains/zephyr-sdk-0.12.4/xtensa/intel_s1000/xtensa-zephyr-elf/bin/xtensa-zephyr-elf-gcc -DBUILD_VERSION=zephyr-v2.5.0-3303-g1cbb6fb50351 -DCC_OPTIMIZE_FLAGS=\"-Os\" -DKERNEL -DSOF_GIT_TAG=\"\" -DSOF_MAJOR=1 -DSOF_MICRO=0 -DSOF_MINOR=5 -DSOF_SRC_HASH=0x0 -DSOF_TAG=\"2f8ef\" -DXCC_TOOLS_VERSION=\"zephyr\" -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -DRELATIVE_FILE=\"../../var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/src/schedule/timer_domain.c\" -I/workdir/zephyr/include -Izephyr/include/generated -I/workdir/zephyr/soc/xtensa/intel_adsp/cavs_v25 -I/workdir/zephyr/soc/xtensa/intel_adsp/cavs_v25/include -I/workdir/zephyr/soc/xtensa/intel_adsp/common/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../src/platform/intel/cavs/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../../../../zephyr/soc/xtensa/intel_adsp/common/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../src/platform/tigerlake/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/xtensa/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/hal/xtensa/zephyr/soc/intel_cavs_25 -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/ext/rimage/src/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../zephyr/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../src/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../src/arch/xtensa/include -I/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/zephyr/../zephyr/include/cavs25 -isystem /workdir/zephyr/lib/libc/minimal/include -isystem /opt/toolchains/zephyr-sdk-0.12.4/xtensa/intel_s1000/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.2.0/include -isystem /opt/toolchains/zephyr-sdk-0.12.4/xtensa/intel_s1000/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/10.2.0/include-fixed -Os -imacros /workdir/zephyr/twister-out/intel_adsp_cavs25/samples/subsys/audio/sof/sample.audio.sof/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fdiagnostics-color=always -imacros /workdir/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/workdir/zephyr/samples/subsys/audio/sof=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -nostdinc -std=gnu99 -fno-inline-functions -MD -MT modules/sof/CMakeFiles/modules_sof.dir/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/src/schedule/timer_domain.c.obj -MF modules/sof/CMakeFiles/modules_sof.dir/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/src/schedule/timer_domain.c.obj.d -o modules/sof/CMakeFiles/modules_sof.dir/var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/src/schedule/timer_domain.c.obj -c /var/lib/buildkite-agent/zephyr-module-cache/modules/audio/sof/src/schedule/timer_domain.c
--
 ```

lyakh, 2021-05-07:
fix in https://github.com/thesofproject/sof/pull/4149

PR #34990 (2021-05-07): sof: update SOF tree
Update tree to and address work queue deprecation.

Fixes #34948
Fixes #34880
commit 168e9ed6 (2021-05-07): sof: update SOF tree

Update tree to and address work queue deprecation.

Fixes #34948
Fixes #34880
closed 2021-05-07 by galak