zephyrproject-rtos/zephyr #34891: mcumgr timeout due to smp_shell_process stalling

sw, 2021-05-06:
`smp_shell_process` only tries to get one net_buf from the FIFO:
https://github.com/zephyrproject-rtos/zephyr/blob/af10c06885c92321ff947ab83c3abcf8f0281cf3/subsys/mgmt/mcumgr/smp_shell.c#L140

If there have been more than one fragment received, these will only be processed on the next call to `smp_shell_process`.

This results in timeouts when using firmware image upload using mcumgr, as the system will fail to process the frame, until the mcumgr CLI tool runs into a timeout (default 3s) and resends the frame.

There should probably be a loop in this function, or alternatively a return value that will tell the caller that the function should be called again.

Note: this can only be reproduced with a short delay between fragments sent by the mcumgr CLI tool.
See https://github.com/apache/mynewt-newtmgr/issues/180. A delay <5ms with a baud rate of 460k shows the problem here, but of course YMMV depending on system clock etc.

mniestroj, 2021-05-06:
Thanks @sw! You are certainly right, packets should be processed in a loop. Could you check if #34897 solves the issue?

sw, 2021-05-06:
That was quick. Thank you @mniestroj , that does indeed solve the issue.

PR #34897 (2021-05-06): smp: shell: consume all pending buffers in smp_shell_process()
smp_shell_process() is called only once after receiving new bytes over
shell. If multiple MCUMGR frames were received over UART one after the
other, then calling smp_shell_process() resulted in consuming only the
first one. All subsequent frames were not processed unless there was
some more RX traffic.

Process received frames in smp_shell_process() in a loop until there is
no frame left. This will make sure that received packets are not stalled
waiting for more RX traffic to trigger processing again.

Fixes: #34891
closed 2021-05-06 by carlescufi