zephyrproject-rtos/zephyr #34904: uart_mcux_lpuart: driver use k_sleep which makes it unworkable when disable CONFIG_MULTITHREADING

hakehuang, 2021-05-06:
**Describe the bug**
k_sleep is not defined when disable CONFIG_MULTITHREADING

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k82f
3. make
4. See error

**Expected behavior**
do not use k_sleep in driver

**Impact**
driver usage in no multithread mode for all NXP lpuart enalbed platform

**Logs and console output**

```
ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading$ west build -b frdm_k82f
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/shared/disk/zephyr_project/zephyr_test/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading
-- Zephyr version: 2.5.99 (/home/shared/disk/zephyr_project/zephyr_test/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.0") found components: Interpreter
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: frdm_k82f
-- Cache files will be written to: /home/ubuntu/.cache/zephyr
-- Using toolchain: zephyr 0.11.3 (/home/ubuntu/zephyr-sdk)
-- Found dtc: /home/ubuntu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/shared/disk/zephyr_project/zephyr_test/zephyr/boards/arm/frdm_k82f/frdm_k82f.dts
-- Generated zephyr.dts: /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/include/generated/device_extern.h
Parsing /home/shared/disk/zephyr_project/zephyr_test/zephyr/Kconfig
Loaded configuration '/home/shared/disk/zephyr_project/zephyr_test/zephyr/boards/arm/frdm_k82f/frdm_k82f_defconfig'
Merged configuration '/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/prj.conf'
Configuration saved to '/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/.config'
Kconfig header saved to '/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Deprecation Warning at /home/shared/disk/zephyr_project/zephyr_test/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build
-- west build: building application
[1/127] Preparing syscall dependency handling
[120/127] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
: && ccache /home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/boards/arm/frdm_k82f/libboards__arm__frdm_k82f.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  modules/nxp/lib..__modules__hal__nxp.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"/home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/thumb/v7e-m+fp/hard"  -L/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfpu=fpv4-sp-d16  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr && /home/shared/disk/zephyr_project/cmake-3.19.4-Linux-x86_64/bin/cmake -E echo
/home/ubuntu/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: zephyr/drivers/serial/libdrivers__serial.a(uart_mcux_lpuart.c.obj): in function `k_sleep':
/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/arch/arm/arm_no_multithreading/build/zephyr/include/generated/syscalls/kernel.h:86: undefined reference to `z_impl_k_sleep'

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.5.0-3397-gaf10c06885

hakehuang, 2021-05-06:
@MaureenHelm , @dleach02 , @mmahadevan108

galak, 2021-05-06:
If there's a desire to support frdm-k82f with no-multithreading, once that works it should get added to tests/arch/arm/arm_no_multithreading

hakehuang, 2021-05-07:
> If there's a desire to support frdm-k82f with no-multithreading, once that works it should get added to tests/arch/arm/arm_no_multithreading

just to add more information, all lpuart platfrom from NXP will be impacted.

MaureenHelm, 2021-05-13:
There aren't any boards with LPUART enabled for this test.

@hakehuang Can you check your test filter to see that it matches the twister filter?

Converting this issue to an enhancement.

hakehuang, 2021-05-14:
@MaureenHelm , this has been fixed in recent commit 
#d34a289503b3cbb258223ffc6cef19b47ffdc2ef
tests: arch: arm: no_multithreading: Limit platforms allowed

When multithreading is off, kernel source files like sem.c (samphore
implementation) are not present in the build. Some platforms by default
fetch modules or drivers that are using multithreading primitives and
because of that fails to compile when multithreading is off.

Limit the test to only qemu platforms since test is arch specific.

closed 2021-05-14 by hakehuang