zephyrproject-rtos/zephyr #34909: dma_loopback:lpcxpresso55s28_ns driver test failure

hakehuang, 2021-05-06:
**Describe the bug**
dma_loopback:lpcxpresso55s28_ns driver test failure
        Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/dma/loop_transfer/src/test_dma_loop.c:175: test_dma_m2m_loop: (test_loop() == TC_PASS) is false

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s28_ns
3. make flash
4. See error

**Expected behavior**
test pass

**Impact**
LPC DMA fuctions

**Logs and console output**
```
    *** Booting Zephyr OS build zephyr-v2.5.0-3077-g3a7e0f875cf1  ***

    Running test suite dma_m2m_loop_test

    ===================================================================

    START - test_dma_m2m_loop

    DMA memory to memory transfer started on DMA_0

    Preparing DMA Controller

    Starting the transfer and waiting for 1 second

    E: invalid DMA channel number -22

    ERROR: transfer config (-22)

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/dma/loop_transfer/src/test_dma_loop.c:175: test_dma_m2m_loop: (test_loop() == TC_PASS) is false

    

     FAIL - test_dma_m2m_loop in 0.30 seconds

    ===================================================================

    Test suite dma_m2m_loop_test failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-3077-g3a7e0f875cf1

hakehuang, 2021-05-06:
@MaureenHelm , @dleach02 , @mmahadevan108

galak, 2021-05-20:
@MaureenHelm / @hakehuang seems like the issue here is with CONFIG_DMA_LOOP_TRANSFER_CHANNEL_NR.  Not sure why the default value of 0 would cause the issue.

hakehuang, 2021-05-21:
> @MaureenHelm / @hakehuang seems like the issue here is with CONFIG_DMA_LOOP_TRANSFER_CHANNEL_NR. Not sure why the default value of 0 would cause the issue.

Thanks @galak , looks like LPC DMA can not use channel 0 as memory to memory dma. see this
[image]

[image]
[image]

we should use channel 32/21 instead, let me create a PR to fix this test

chrta, 2021-05-21:
Is this issue still relevant or can this be closed?
- the board lpcxpresso55s28_ns was removed, so this test cannot be built for this board any more
- i can run this test successfully on lpcxpresso55s28
```
$ west build -b lpcxpresso55s28
$ west flash
...
*** Booting Zephyr OS build v2.6.0-rc1-292-ga46a36a4cfeb  ***
Running test suite dma_m2m_loop_test
===================================================================
START - test_dma_m2m_loop
DMA memory to memory transfer started on DMA_0
Preparing DMA Controller
this platform do not support the dma channel
Starting the transfer on channel 0 and waiting for 1 second
Each RX buffer should contain the full TX buffer string.
RX data Loop 0: The quick brown fox jumps over the lazy dog
RX data Loop 1: The quick brown fox jumps over the lazy dog
RX data Loop 2: The quick brown fox jumps over the lazy dog
RX data Loop 3: The quick brown fox jumps over the lazy dog
RX data Loop 4: The quick brown fox jumps over the lazy dog
Finished: DMA
 PASS - test_dma_m2m_loop in 1.49 seconds
===================================================================
Test suite dma_m2m_loop_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

galak, 2021-05-21:
Going to close this based on @chrta report.

closed 2021-05-21 by galak