zephyrproject-rtos/zephyr #34911: tests/kernel/mem_protect/mem_protect: frdm_k82f unexpected fatal error

hakehuang, 2021-05-06:
**Describe the bug**
in memory protection test meet belo werror in test_kobject_perm_error
fatal error was unexpected, aborting

**To Reproduce**
Steps to reproduce the behavior:
tests/kernel/mem_protect/mem_protect
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k82f
3. make flash
4. See error

**Expected behavior**
all test PASS

**Impact**
memory protection featues

**Logs and console output**
see attached file

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-2637-ga3469a049785

hakehuang, 2021-05-06:
@MaureenHelm , @dleach02 , @mmahadevan108

galak, 2021-05-11:
Does this reproduce on K64F?

galak, 2021-05-11:
Tested this out on K64F and seeing similar issue there.

dleach02, 2021-05-12:
I can see the problem. 

It was introduced with e53d549. What I'm seeing is that the test case added with the reference commit is failing on NXP boards. There are some assumptions being made on the number of partitions available with test_mem_part_assert_add_overmax() test where the arch_mem_domain_max_partitions_get() function is returning 4 maximum partitions but the test_domain structure has 12 partitions in it for the function k_mem_domain_add_partition() to allocate from so it doesn't fail as the test case is expecting. 

I'm not sure if these functions are returning different values for non-NXP parts or not but if I take out the test_mem_part_assert_add_overmax() the rest of the test suite works fine.

@maksimmasalski, @enjiamai, @ioannisg

enjiamai, 2021-05-12:
Thanks for the message of debugging. @dleach02 , yes, test_mem_part_assert_add_overmax() is expected exception but nothing happened. Did you reproduce it on frdm_k64f? I check the fdrm_k64f it passed the test case.

dleach02, 2021-05-12:
@enjiamai yes I reproduced on frdm_k64f. In thinking about this, I think there is a mixing of kernel and driver calls here that is causing problems. The NXP parts may have a number of MPU slots but they are used differently then other ARM based parts. I haven't fully grepped the interplay of the test_domain structure and the normal memory protection structures used.

enjiamai, 2021-05-12:
Hi @dleach02, I did not run totally follow the steps that @hakehuang provided, (because today I use them frdm_k64f remotely, I can't reset it after west flash. I can try it tomorrow.) 

For the commit 76dee39 or 8e69dafccf25bc5b4015a607cdba114216c4ea0e It shows pass when I run it by this command:
```
twister -p frdm_k64f --device-testing --device-serial="/dev/ttyACM0" -T tests/kernel/mem_protect/mem_protect -vv -DCONFIG_USERSPACE=y
```
or
```
twister -p frdm_k64f --device-testing --device-serial="/dev/ttyACM0" -T tests/kernel/mem_protect/mem_protect -vv -xCONFIG_USERSPACE=y
```
So @dleach02 @hakehuang , could you please help to try it this way in your frdm_k64f  and frdm_k82f, thank you for the help!

hakehuang, 2021-05-13:
adding CONFIG_USERSPACE=y does not work for me. BTW, what is your console log? @enjiamai

enjiamai, 2021-05-13:
@hakehuang here is the log for your reference:
[memprotec.frdm_k64f.output.txt](https://github.com/zephyrproject-rtos/zephyr/files/6470207/memprotec.frdm_k64f.output.txt)

hakehuang, 2021-05-13:
> @hakehuang here is the log for your reference:
> [memprotec.frdm_k64f.output.txt](https://github.com/zephyrproject-rtos/zephyr/files/6470207/memprotec.frdm_k64f.output.txt)

sorry enjiamai, adding CONFIG_USERSPACE=y works for me, but the log show some extra characters.

dleach02, 2021-05-13:
I don't quite understand how this test_mem_part_assert_add_overmax() test case is supposed to work. It is using a mix of MPU driver code and kernel code to fiddle with partitions. When it makes a call to arch_mem_domain_max_partitions_get() to get the maximum number of partitions this function is for the platform and in the case of the NXP core it returns a value that also removes the static partitions from the real memory map (including reserving parts for stack guards for supervisor and thread stacks). So by the time this test case is run, the partitions_get() returns a value of 4. 

But the test case is using a locally defined  struct k_mem_domain test_domain structure to play with partitions. This structure has an array for the number of partitions set to CONFIG_MAX_DOMAIN_PARTITIONS which is 16. So the call to k_mem_domain_add_partition() will return true because only 4 of the entries are used and as such the particular test case fails. 

I suspect these tests were originally written for x86 where ARCH_MEM_DOMAIN_SYNCHRONOUS_API is true which would force a synchronous update of the added partition and thus give an immediate error to k_mem_domain_add_partition(). The NXP cores don't use this setting and because the kernel function doesn't have any insight on what is really available from the SOC MPU it will happily add the partition.

I don't know how this behaves on non-NXP ARM cores but I'm thinking it may be the same thing.

dleach02, 2021-05-13:
As for the original reason for this bug ticket, test_kobject_perm_error fails because the previous failed memory test, test_mem_part_assert_add_overmax(), did not fault as expected so the variable need_recover_spinlock used to release the z_mem_domain_lock spinlock is still set. So when the first test case in kobject_perm faults, the after fault handler thinks it needs to reset the z_mem_domain_lock spinlock and gets a kernel assert because it doesn't own the lock.

I can add a reset of the variable but this leaves the earlier problem with test_mem_part_assert_add_overmax() still not working as expected on these boards.

dleach02, 2021-05-13:
To get the NXP style MPU to detect a overmax() condition, you have to have the thread owning the memory domain to be running. Then when the swap tries to load up the partitions it will fail.

dleach02, 2021-05-14:
Now that I have a PR out it seems like this PR #34086 by @mmahadevan108 either fixed it or adjusted the timings. I'm trying to think through how I originally bisected this and how others reproduced this problem.

enjiamai, 2021-05-14:
Sorry for the late reply. Thanks for the great analysis, @dleach02 . Indeed, the assumption of test_mem_part_assert_add_overmax() is all of the previous testcases of operating the mem domain work well as expected.  And it aims to verify the assertion in kernel/mem_domain.c line 280 can detect this error.

enjiamai, 2021-05-14:
Excuse me, @dleach02 @hakehuang, could it still reproduce both in fdrm_k64f and fdrm_k82f?

I also have tried by below command:  Log [Uploading memprotec.frdm_k64f.output.west.2.txt…]()
```
west build tests/kernel/mem_protect/mem_protect/ -b frdm_k64f -p always
west flash
```

hakehuang, 2021-05-14:
> Excuse me, @dleach02 @hakehuang, could it still reproduce both in fdrm_k64f and fdrm_k82f?
> 
> I also have tried by below command: Log
> [memprotec.frdm_k64f.output.west.2.txt](https://github.com/zephyrproject-rtos/zephyr/files/6475953/memprotec.frdm_k64f.output.west.2.txt)
> 
> ```
> west build tests/kernel/mem_protect/mem_protect/ -b frdm_k64f -p always
> west flash
> ```

you mean with least main branch code?

enjiamai, 2021-05-14:
> you mean with least main branch code?

Yes, @hakehuang ,  I think we could use the latest code to validate this, please. Thank you so much!

enjiamai, 2021-05-14:
I verified what @dleach02 provided, this issue appears to be fixed after #34086 , in frdm_k64f.  So, @hakehuang , could this issue be closed if it is also passed in frdm_k82f?

hakehuang, 2021-05-14:
> I verified what @dleach02 provided, this issue appears to be fixed after #34086 , in frdm_k64f. So, @hakehuang , could this issue be closed if it is also passed in frdm_k82f?

yes, it is ok now

dleach02, 2021-05-17:
Yes. The underlying problem was fixed as a side affect of the #34086. I'm going to close this and my PR.

closed 2021-05-17 by dleach02