zephyrproject-rtos/zephyr #34915: arch.interrupt.gen_isr_table.arm_mainline:lpcxpresso55s16_ns: interrupt 57 does not work

hakehuang, 2021-05-06:
**Describe the bug**
    test_build_time_interrupt
        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/gen_isr_table/src/main.c:278: test_build_time_interrupt: (check_sw_isr(isr3, ISR3_ARG, ISR3_OFFSET) is non-zero)

    check interrupt isr3 failed

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s16_ns
3. make flash
4. See error

**Expected behavior**
test pass

**Impact**
interrupt function

**Logs and console output**
```
    *** Booting Zephyr OS build zephyr-v2.5.0-3077-g3a7e0f875cf1  ***

    START - Test gen_isr_tables

    IRQ configuration (total lines 60):

    Running test suite context

    ===================================================================

    START - test_build_time_direct_interrupt

    isr1 isr=0x4ed irq=59

    isr2 isr=0x55d irq=58

    Checking _irq_vector_table entry 59 for irq 59

    triggering irq 59

    Triggering irq : 59

    isr1 ran

    Checking _irq_vector_table entry 58 for irq 58

    triggering irq 58

    Triggering irq : 58

    isr2 ran

     PASS - test_build_time_direct_interrupt in 0.21 seconds

    ===================================================================

    START - test_build_time_interrupt

    _sw_isr_table at location 0x2000006c

    isr3 isr=0x45d irq=57 param=0xb01dface

    Checking _sw_isr_table entry 57 for irq 57

    triggering irq 57

    Triggering irq : 57

    interrupt 57 didn't run once, ran 0 times

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/gen_isr_table/src/main.c:278: test_build_time_interrupt: (check_sw_isr(isr3, ISR3_ARG, ISR3_OFFSET) is non-zero)

    check interrupt isr3 failed

     FAIL - test_build_time_interrupt in 0.36 seconds

    ===================================================================

    START - test_run_time_interrupt

    isr5 isr=0x4a5 irq=55 param=0xf0ccac1a

    Checking _sw_isr_table entry 55 for irq 55

    triggering irq 55

    Triggering irq : 55

    isr5 ran with parameter 0xf0ccac1a

    isr6 isr=0x4c9 irq=54 param=0xba5eba11

    Checking _sw_isr_table entry 54 for irq 54

    triggering irq 54

    Triggering irq : 54

    isr6 ran with parameter 0xba5eba11

     PASS - test_run_time_interrupt in 0.28 seconds

    ===================================================================

    Test suite context failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  zephyr-v2.5.0-3077-g3a7e0f875cf1

hakehuang, 2021-05-06:
@MaureenHelm , @dleach02 , @mmahadevan108

galak, 2021-05-11:
Is this config one we support w/twister -- if not we probably don't consider a valid config to try to test.

hakehuang, 2021-05-12:
> Is this config one we support w/twister -- if not we probably don't consider a valid config to try to test.

Hi @galak 
```
  arch.interrupt.gen_isr_table.arm_mainline:
    arch_allow: arm
    platform_exclude: stmf103_mini nucleo_f103rb olimexino_stm32 stm32_min_dev_black
        stm32_min_dev_blue usb_kw24d512 v2m_beetle cc1352r1_launchxl
        cc26x2r1_launchxl olimex_stm32_h103 cc1352r_sensortag
    filter: CONFIG_GEN_ISR_TABLES and CONFIG_ARMV7_M_ARMV8_M_MAINLINE
```

form the west test config, this platform shall support this case, but maybe we need exclude lpc platforms, looks like the case is not well designed

henrikbrixandersen, 2021-05-19:
@hakehuang Are you sure this is failing on the lpcxpresso55s28 as well? The lpcxpresso55s16 has IRQ 57 reserved, but according to the user manual for the lpcxpresso55s28, it does not reserve IRQ 57.

I have tested the fix in #35472 on the lpcxpresso55s16 but I do not have access to a  lpcxpresso55s28 EVK.

PR #35472 (2021-05-19): tests: kernel: gen_isr_table: do not use IRQ 57 on NXP LPC55S16
IRQ 57 is reserved in the NXP LPC55S16 SoC. Thus, limit the number of interrupts reported to the test, so that it does not try to use it.

Fixes #34915
commit aa78a647 (2021-05-19): tests: kernel: gen_isr_table: do not use IRQ 57 on NXP LPC55S16

IRQ 57 is reserved in the NXP LPC55S16 SoC. Thus, limit the number of
interrupts reported to the test, so that it does not try to use it.

Fixes #34915
closed 2021-05-19 by galak