zephyrproject-rtos/zephyr #34917: arch.interrupt.arm| arch.interrupt.extra_exception_info: lpcxpresso55s28 series: test failure

hakehuang, 2021-05-06:
**Describe the bug**
interrupt testing failure, but is disable optimization and enlarge the stack size test can pass 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b lpcxpresso55s28 -- -DCONFIG_EXTRA_EXCEPTION_INFO=y
   or
    west build -b lpcxpresso55s28
3. make flash
4. See error

test pass with below build options

west build -b lpcxpresso55s28 --  -DCONFIG_NO_OPTIMIZATIONS=y -DCONFIG_IDLE_STACK_SIZE=512 -DCONFIG_MAIN_STACK_SIZE=1024

**Expected behavior**
Caught system error -- reason 4

ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993

	ESF could not be retrieved successfully. Shall never occur.

ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993

	ESF could not be retrieved successfully. Shall never occur.

**Impact**
interrupt in lpc55 series

**Logs and console output**

```

*** Booting Zephyr OS build zephyr-v2.5.0-3077-g3a7e0f875cf1  ***

Running test suite arm_interrupt

===================================================================

START - test_arm_null_pointer_exception

Skipped

 PASS - test_arm_null_pointer_exception in 0.1 seconds

===================================================================

START - test_arm_interrupt

Available IRQ line: 59

E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0

E: Current thread: 0x30000138 (unknown)

Caught system error -- reason 1

E: r0/a1:  0x00000003  r1/a2:  0x30002de0  r2/a3:  0x00000003

E: r3/a4:  0x30002bc0 r12/ip:  0xa0000000 r14/lr:  0x100025e7

E:  xpsr:  0x6100004b

E: EXC_RETURN: 0x0

E: Faulting instruction address (r15/pc): 0x1000070a

E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0

E: Fault during interrupt handling

E: Current thread: 0x30000138 (unknown)

Caught system error -- reason 3

E: r0/a1:  0x00000004  r1/a2:  0x30002de0  r2/a3:  0x00000004

E: r3/a4:  0x30002bc0 r12/ip:  0x00000000 r14/lr:  0x100025e7

E:  xpsr:  0x6100004b

E: EXC_RETURN: 0x0

E: Faulting instruction address (r15/pc): 0x10000728

E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0

E: Fault during interrupt handling

E: Current thread: 0x30000138 (unknown)

Caught system error -- reason 4

ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:216

	Intentional assert

E: r0/a1:  0x00000004  r1/a2:  0x000000d8  r2/a3:  0x80000000

E: r3/a4:  0x0000004b r12/ip:  0xa0000000 r14/lr:  0x100025e7

E:  xpsr:  0x4100004b

E: EXC_RETURN: 0x0

E: Faulting instruction address (r15/pc): 0x1000b7b0

E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0

E: Fault during interrupt handling

E: Current thread: 0x30000138 (unknown)

Caught system error -- reason 4

ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993

	ESF could not be retrieved successfully. Shall never occur.

ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993

	ESF could not be retrieved successfully. Shall never occur.
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-3077-g3a7e0f875cf1

hakehuang, 2021-05-06:
@MaureenHelm , @dleach02 , @mmahadevan108

galak, 2021-05-11:
@hakehuang are you seeing this on 55s69?

hakehuang, 2021-05-12:
> @hakehuang are you seeing this on 55s69?

yes, there are same issue on 55s69

ioannisg, 2021-05-20:
@hakehuang what is the commit you were testing? I'd like to try to reproduce this, if possible on other platforms?

hakehuang, 2021-05-21:
> @hakehuang what is the commit SHA you were testing? I'd like to try to reproduce this, if possible on other platforms?

it is on zephyr-v2.5.0-3077-g3a7e0f875cf1 that I meet this failure. @ioannisg

chrta, 2021-05-21:
I see the error with the current master v2.6.0-rc1-292-ga46a36a4cf using zephyr-sdk-0.12.3 in all three mentioned above configuration on the board lpcxpresso55s28.

ioannisg, 2021-05-28:
I can't reproduce any of the tests failures on v2.6.0-rc2, on lpcxpresso55s69. 
@hakehuang if this is the same for 55s28 I suggest you close this ticket.

hakehuang, 2021-06-09:
@ioannisg, when useing sdk-0.12.3 and it still fails

```
*** Booting Zephyr OS build zephyr-v2.6.0-90-gefccfcbfc915  ***
    Running test suite arm_interrupt
    ===================================================================
    START - test_arm_null_pointer_exception
    Skipped
     PASS - test_arm_null_pointer_exception in 0.1 seconds
    ===================================================================
    START - test_arm_interrupt
    Available IRQ line: 59
    E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
    E: Current thread: 0x30000138 (test_arm_interrupt)
    Caught system error -- reason 1
    E: r0/a1:  0x00000003  r1/a2:  0x300031e0  r2/a3:  0x00000003
    E: r3/a4:  0x30002fc0 r12/ip:  0xa0000000 r14/lr:  0x100025f7
    E:  xpsr:  0x6100004b
    E: EXC_RETURN: 0x0
    E: Faulting instruction address (r15/pc): 0x1000070a
    E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
    E: Fault during interrupt handling
    
    E: Current thread: 0x30000138 (test_arm_interrupt)
    Caught system error -- reason 3
    E: r0/a1:  0x00000004  r1/a2:  0x300031e0  r2/a3:  0x00000004
    E: r3/a4:  0x30002fc0 r12/ip:  0x00000000 r14/lr:  0x100025f7
    E:  xpsr:  0x6100004b
    E: EXC_RETURN: 0x0
    E: Faulting instruction address (r15/pc): 0x10000728
    E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
    E: Fault during interrupt handling
    
    E: Current thread: 0x30000138 (test_arm_interrupt)
    Caught system error -- reason 4
    ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:216
    	Intentional assert
    
    E: r0/a1:  0x00000004  r1/a2:  0x000000d8  r2/a3:  0x80000000
    E: r3/a4:  0x0000004b r12/ip:  0xa0000000 r14/lr:  0x100025f7
    E:  xpsr:  0x4100004b
    E: EXC_RETURN: 0x0
    E: Faulting instruction address (r15/pc): 0x1000b9a4
    E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
    E: Fault during interrupt handling
    
    E: Current thread: 0x30000138 (test_arm_interrupt)
    Caught system error -- reason 4
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
    	ESF could not be retrieved successfully. Shall never occur.
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
    	ESF could not be retrieved successfully. Shall never occur.
```

PR #36225 (2021-06-24): tests: arm_interrupt: remove redundant test case part
The test case part in test_arm_interrupt, which is
triggering a SysTick IRQ (when SysTick is not used
for system timer implementation), to catch a spurious
interrupt is not really testing what it is designed
for. The reason is that the z_arm_exc_spurious has
be re-designed to be invoked only if the kernel is
built without support for system clock, i.e. with
CONFIG_SYS_CLOCK_EXISTS=n, but this configuration is
not compatible with ZTESTs and is, therefore, never
tested in this test, in any configuration. When the
SysTick is not implemented in [... cut]
closed 2021-06-24 by ioannisg