zephyrproject-rtos/zephyr #34923: net.socket.get_addr_info: frdm_k64f test fails

hakehuang, 2021-05-06:
**Describe the bug**
below two cases fails, looks like it is not stable enough to all platform
test_getaddrinfo_ok
test_getaddrinfo_cancelled 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f
3. make falsh
4. See error

**Expected behavior**
all test pass

**Impact**
addressinfo api call

**Logs and console output**

```
    start application at 0x70008729 !!!

    start stack at 0x80008800 !!!

    *** Booting Zephyr OS build zephyr-v2.5.0-3077-g3a7e0f875cf1  ***

    Running test suite socket_getaddrinfo

    ===================================================================

    START - test_getaddrinfo_setup

     PASS - test_getaddrinfo_setup in 0.90 seconds

    ===================================================================

    START - test_getaddrinfo_ok

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/getaddrinfo/src/main.c:232: test_getaddrinfo_ok: (false is false)

    Did not receive all queries

     FAIL - test_getaddrinfo_ok in 2.155 seconds

    ===================================================================

    START - test_getaddrinfo_cancelled

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/getaddrinfo/src/main.c:252: test_getaddrinfo_cancelled: (false is false)

    Did not receive all queries

     FAIL - test_getaddrinfo_cancelled in 0.15 seconds

    ===================================================================

    START - test_getaddrinfo_no_host

     PASS - test_getaddrinfo_no_host in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_num_ipv4

     PASS - test_getaddrinfo_num_ipv4 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_num_ipv6

     PASS - test_getaddrinfo_num_ipv6 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_flags_numerichost

     PASS - test_getaddrinfo_flags_numerichost in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_ipv4_hints_ipv6

     PASS - test_getaddrinfo_ipv4_hints_ipv6 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_ipv6_hints_ipv4

     PASS - test_getaddrinfo_ipv6_hints_ipv4 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_port_invalid

     PASS - test_getaddrinfo_port_invalid in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_null_host

     PASS - test_getaddrinfo_null_host in 0.1 seconds

    ===================================================================

    Test suite socket_getaddrinfo failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-3077-g3a7e0f875cf1

hakehuang, 2021-05-06:
@MaureenHelm , @dleach02 , looks like the test is not reliable, but need owner to confirm

dleach02, 2021-05-10:
Hake, I'm not able to reproduce this problem. 

*** Booting Zephyr OS build zephyr-v2.5.0-3651-g793e2c1891b4  ***

jukkar, 2021-05-10:
I tried this with 2.6.0-rc1
```
** Booting Zephyr OS build zephyr-v2.5.0-3767-g5e9d6d638c62  **
```
and did not see any issues.

hakehuang, 2021-05-10:
@jukkar @dleach02 
I still have problem
```
    *** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***

    Running test suite socket_getaddrinfo

    ===================================================================

    START - test_getaddrinfo_setup

     PASS - test_getaddrinfo_setup in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_ok

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/getaddrinfo/src/main.c:232: test_getaddrinfo_ok: (false is false)

    Did not receive all queries

     FAIL - test_getaddrinfo_ok in 2.115 seconds

    ===================================================================

    START - test_getaddrinfo_cancelled

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/getaddrinfo/src/main.c:252: test_getaddrinfo_cancelled: (false is false)

    Did not receive all queries

     FAIL - test_getaddrinfo_cancelled in 0.15 seconds

    ===================================================================

    START - test_getaddrinfo_no_host

     PASS - test_getaddrinfo_no_host in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_num_ipv4

     PASS - test_getaddrinfo_num_ipv4 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_num_ipv6

     PASS - test_getaddrinfo_num_ipv6 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_flags_numerichost

     PASS - test_getaddrinfo_flags_numerichost in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_ipv4_hints_ipv6

     PASS - test_getaddrinfo_ipv4_hints_ipv6 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_ipv6_hints_ipv4

     PASS - test_getaddrinfo_ipv6_hints_ipv4 in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_port_invalid

     PASS - test_getaddrinfo_port_invalid in 0.1 seconds

    ===================================================================

    START - test_getaddrinfo_null_host

     PASS - test_getaddrinfo_null_host in 0.1 seconds

    ===================================================================

    Test suite socket_getaddrinfo failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

dleach02, 2021-05-10:
Okay. If you connect the K64F to a network then I see the failures Hake is seeing

hakehuang, 2021-05-10:
if connect with ethernet cable you will see more tests is runable
```
*** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***
Running test suite socket_getaddrinfo
===================================================================
START - test_getaddrinfo_setup
 PASS - test_getaddrinfo_setup in 0.1 seconds
===================================================================
START - test_getaddrinfo_ok

    Assertion failed at ../src/main.c:232: test_getaddrinfo_ok: (false is false)
Did not receive all queries
 FAIL - test_getaddrinfo_ok in 2.111 seconds
===================================================================
START - test_getaddrinfo_cancelled

    Assertion failed at ../src/main.c:252: test_getaddrinfo_cancelled: (false is false)
Did not receive all queries
 FAIL - test_getaddrinfo_cancelled in 0.11 seconds
===================================================================
START - test_getaddrinfo_no_host
 PASS - test_getaddrinfo_no_host in 0.1 seconds
===================================================================
START - test_getaddrinfo_num_ipv4
 PASS - test_getaddrinfo_num_ipv4 in 0.1 seconds
===================================================================
START - test_getaddrinfo_num_ipv6
 PASS - test_getaddrinfo_num_ipv6 in 0.1 seconds
===================================================================
START - test_getaddrinfo_flags_numerichost
 PASS - test_getaddrinfo_flags_numerichost in 0.1 seconds
===================================================================
START - test_getaddrinfo_ipv4_hints_ipv6
 PASS - test_getaddrinfo_ipv4_hints_ipv6 in 0.1 seconds
===================================================================
START - test_getaddrinfo_ipv6_hints_ipv4
 PASS - test_getaddrinfo_ipv6_hints_ipv4 in 0.1 seconds
===================================================================
START - test_getaddrinfo_port_invalid
 PASS - test_getaddrinfo_port_invalid in 0.1 seconds
===================================================================
START - test_getaddrinfo_null_host
 PASS - test_getaddrinfo_null_host in 0.1 seconds
===================================================================
Test suite socket_getaddrinfo failed.
===================================================================
PROJECT EXECUTION FAILED
I: ETH_0 enabled 100M full-duplex mode.
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0x01f3d900  r1/a2:  0x000027f3  r2/a3:  0x200020b0
E: r3/a4:  0x20001338 r12/ip:  0x400c0000 r14/lr:  0x00011ef5
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00011eaa
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200014c0 (sysworkq)
E: Halting system
```

but if no ethernet cable, then
```
*** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***
Running test suite socket_getaddrinfo
===================================================================
START - test_getaddrinfo_setup
 PASS - test_getaddrinfo_setup in 0.1 seconds
===================================================================
START - test_getaddrinfo_ok
 PASS - test_getaddrinfo_ok in 4.1 seconds
===================================================================
START - test_getaddrinfo_cancelled
 PASS - test_getaddrinfo_cancelled in 4.2 seconds
===================================================================
START - test_getaddrinfo_no_host
 PASS - test_getaddrinfo_no_host in 0.1 seconds
===================================================================
START - test_getaddrinfo_num_ipv4
 PASS - test_getaddrinfo_num_ipv4 in 0.1 seconds
===================================================================
START - test_getaddrinfo_num_ipv6
 PASS - test_getaddrinfo_num_ipv6 in 0.1 seconds
===================================================================
START - test_getaddrinfo_flags_numerichost
 PASS - test_getaddrinfo_flags_numerichost in 0.1 seconds
===================================================================
START - test_getaddrinfo_ipv4_hints_ipv6
 PASS - test_getaddrinfo_ipv4_hints_ipv6 in 0.1 seconds
===================================================================
START - test_getaddrinfo_ipv6_hints_ipv4
 PASS - test_getaddrinfo_ipv6_hints_ipv4 in 0.1 seconds
===================================================================
START - test_getaddrinfo_port_invalid
 PASS - test_getaddrinfo_port_invalid in 0.1 seconds
===================================================================
START - test_getaddrinfo_null_host
 PASS - test_getaddrinfo_null_host in 0.1 seconds
===================================================================
Test suite socket_getaddrinfo succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

jukkar, 2021-05-10:
The test used loopback interface and as we do not really need Ethernet, then disabling Ethernet will fix the issue.

PR #35066 (2021-05-14): tests: net: getaddrinfo: Disable onboard Ethernet controller
If the tests/net/socket/getaddrinfo is run in a board that has
Ethernet controller, then it is possible that it will interfere
the test if the Ethernet cable is connected. As the test only
needs loopback support to run, disable the Ethernet as it is
not needed by the test.

Fixes #34923
commit dab391f2 (2021-05-14): tests: net: getaddrinfo: Disable onboard Ethernet controller

If the tests/net/socket/getaddrinfo is run in a board that has
Ethernet controller, then it is possible that it will interfere
the test if the Ethernet cable is connected. As the test only
needs loopback support to run, disable the Ethernet as it is
not needed by the test.

Fixes #34923
closed 2021-05-14 by nashif