zephyrproject-rtos/zephyr #34925: tests/lib/cbprintf_package fails to build

galak, 2021-05-06:
`./scripts/twister -l -T tests/lib/cbprintf_package/`

Will get:

```
ERROR   - qemu_x86_64               tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/build.log
INFO    - Total complete:    1/  83   1%  skipped: 2430, failed:    1
ERROR   - qemu_x86_64               tests/lib/cbprintf_package/libraries.cbprintf_package_long_double FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double/build.log
INFO    - Total complete:    2/  83   2%  skipped: 2430, failed:    2
ERROR   - qemu_x86_64               tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset/build.log
INFO    - Total complete:    3/  83   3%  skipped: 2430, failed:    3
ERROR   - qemu_riscv32              tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv32/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/build.log
INFO    - Total complete:    4/  83   4%  skipped: 2430, failed:    4
ERROR   - qemu_riscv64              tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv64/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/build.log
INFO    - Total complete:    5/  83   6%  skipped: 2430, failed:    5
ERROR   - qemu_riscv64              tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv64/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset/build.log
INFO    - Total complete:    6/  83   7%  skipped: 2430, failed:    6
ERROR   - qemu_riscv64              tests/lib/cbprintf_package/libraries.cbprintf_package_long_double FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv64/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double/build.log
INFO    - Total complete:    7/  83   8%  skipped: 2430, failed:    7
ERROR   - qemu_cortex_a53           tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_a53/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/build.log
INFO    - Total complete:    8/  83   9%  skipped: 2430, failed:    8
ERROR   - qemu_cortex_a53           tests/lib/cbprintf_package/libraries.cbprintf_package_long_double FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_a53/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double/build.log
INFO    - Total complete:    9/  83  10%  skipped: 2430, failed:    9
ERROR   - qemu_cortex_a53           tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_a53/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset/build.log
INFO    - Total complete:   11/  83  13%  skipped: 2430, failed:   10
ERROR   - qemu_cortex_m3            tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_m3/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/handler.log
INFO    - Total complete:   13/  83  15%  skipped: 2430, failed:   11
ERROR   - qemu_x86_64               tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset/build.log
INFO    - Total complete:   14/  83  16%  skipped: 2430, failed:   12
ERROR   - qemu_riscv32              tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv32/tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset/build.log
INFO    - Total complete:   15/  83  18%  skipped: 2430, failed:   13
ERROR   - qemu_riscv64              tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv64/tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset/build.log
INFO    - Total complete:   16/  83  19%  skipped: 2430, failed:   14
ERROR   - qemu_cortex_a53           tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_a53/tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset/build.log
INFO    - Total complete:   17/  83  20%  skipped: 2430, failed:   15
ERROR   - qemu_riscv64              tests/lib/cbprintf_package/libraries.cbprintf_package_fp FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv64/tests/lib/cbprintf_package/libraries.cbprintf_package_fp/build.log
INFO    - Total complete:   18/  83  21%  skipped: 2430, failed:   16
ERROR   - qemu_riscv32              tests/lib/cbprintf_package/libraries.cbprintf_package_fp FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_riscv32/tests/lib/cbprintf_package/libraries.cbprintf_package_fp/build.log
INFO    - Total complete:   19/  83  22%  skipped: 2430, failed:   17
ERROR   - qemu_cortex_a53           tests/lib/cbprintf_package/libraries.cbprintf_package_fp FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_a53/tests/lib/cbprintf_package/libraries.cbprintf_package_fp/build.log
INFO    - Total complete:   20/  83  24%  skipped: 2430, failed:   18
ERROR   - qemu_x86_64               tests/lib/cbprintf_package/libraries.cbprintf_package_fp FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86_64/tests/lib/cbprintf_package/libraries.cbprintf_package_fp/build.log
INFO    - Total complete:   22/  83  26%  skipped: 2430, failed:   19
ERROR   - qemu_leon3                tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Exited with 0
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_leon3/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/handler.log
INFO    - Total complete:   23/  83  27%  skipped: 2430, failed:   20
ERROR   - qemu_x86                  tests/lib/cbprintf_package/libraries.cbprintf_package_nano FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_x86/tests/lib/cbprintf_package/libraries.cbprintf_package_nano/handler.log
INFO    - Total complete:   24/  83  28%  skipped: 2430, failed:   21
ERROR   - qemu_leon3                tests/lib/cbprintf_package/libraries.cbprintf_package_long_double FAILED: Exited with 0
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_leon3/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double/handler.log
[... 38 lines cut ...]
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_xtensa/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double/handler.log
INFO    - Total complete:   62/  83  74%  skipped: 2430, failed:   35
ERROR   - qemu_cortex_m0            tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_m0/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double_align_offset/handler.log
INFO    - Total complete:   63/  83  75%  skipped: 2430, failed:   36
ERROR   - qemu_cortex_m0            tests/lib/cbprintf_package/libraries.cbprintf_package_long_double FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_m0/tests/lib/cbprintf_package/libraries.cbprintf_package_long_double/handler.log
INFO    - Total complete:   64/  83  77%  skipped: 2430, failed:   37
ERROR   - qemu_cortex_m0            tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_m0/tests/lib/cbprintf_package/libraries.cbprintf_package_fp_align_offset/handler.log
[... cut ...]
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_xtensa/tests/lib/cbprintf_package/libraries.cbprintf_package/handler.log
INFO    - Total complete:   68/  83  81%  skipped: 2430, failed:   41
ERROR   - qemu_cortex_m0            tests/lib/cbprintf_package/libraries.cbprintf_package FAILED: Timeout
ERROR   - see: /home/galak/git/zephyr/twister-out/qemu_cortex_m0/tests/lib/cbprintf_package/libraries.cbprintf_package/handler.log
INFO    - Total complete:   69/  83  83%  skipped: 2430, failed:   42^CINFO    - Execution interrupted

```

galak, 2021-05-06:
@nordic-krch in testcase.yaml we have an issue with integration_platforms & platform_allow creating an empty set of platforms to test on.  ie, `native_posix` isn't in the platform_allow list.  Additionally, it seems since C++ support varies a bit arch to arch so maybe best to just drop `integration_platforms` and leave it to the platform_allow list.

```
tests:
  libraries.cbprintf_package:
    tags: cbprintf
    integration_platforms:
      - native_posix
    platform_allow: >
      qemu_arc_em qemu_arc_hs qemu_cortex_a53 qemu_cortex_m0 qemu_cortex_m3
      qemu_cortex_r5 qemu_leon3 qemu_nios2 qemu_riscv32 qemu_riscv64 qemu_x86
      qemu_x86_64 qemu_xtensa
    extra_configs:
      - CONFIG_CBPRINTF_COMPLETE=y
```

nordic-krch, 2021-05-06:
Will be looking at this. Seems that C++ in fact behavior is platform specific.

nashif, 2021-05-06:
#34946 fixes the filtering and shows what is failing as well.

PR #34951 (2021-05-07): lib: os: cbprintf: Fixes for c++ and coverity
Fixes for C++ includes:
    - avoid calling static inline template function in static
    assert on certain platforms which consider it non const expression
    - Add 0 to variable before calling C++ argument storing function
    to promote a variable.
    
Fix for coverity:
    - avoid using sizeof(n + 0) by using local variable (gcc extension)

Additionally removed intergration platform from `tests/lib/cbprintf_package` since it is platform specific test.

Fixes #34925.
Fixes #34838.

It also suppose to fix 14K coverity issues.
closed 2021-05-07 by galak