zephyrproject-rtos/zephyr #34948: SoF module is not pointing at Zehpyr repo

galak, 2021-05-07:
West.yml has:

```
    - name: sof
      url: https://github.com/nordic-krch/sof
      revision: 2f8ef204454d278f0c189e8c7cc70e84f745561a
      path: modules/audio/sof
```

This is not acceptable for release and needs to be resolved to use the Zephyr SoF fork.

nordic-krch, 2021-05-07:
@nashif Fix in sof has been merged but I am not sure how that gets to zephyr fork.

nashif, 2021-05-07:
huh,  and I was wondering how the MAX issue was worked around, @carlescufi  how can we block such changes from being merged, anyway we can do a check on the manifest and block PRs that have such entries?

closed 2021-05-07 by galak

marc-hb, 2021-05-07:
> huh, and I was wondering how the MAX issue was worked around,

https://github.com/zephyrproject-rtos/zephyr/issues/33567#issuecomment-829868678

PR #34990 (2021-05-07): sof: update SOF tree
Update tree to and address work queue deprecation.

Fixes #34948
Fixes #34880
commit 168e9ed6 (2021-05-07): sof: update SOF tree

Update tree to and address work queue deprecation.

Fixes #34948
Fixes #34880