zephyrproject-rtos/zephyr #34949: BLE UART CONSOLE

cederom, 2021-05-07:
**Is your feature request related to a problem? Please describe.**
There is a CONSOLE module in Zephyr that helps in device control and development. It would be really nice to have also BLE UART module that would allow remote testing and control of devices with no physical usb/uart over the wire. I know that Nordic provides custom service for BLE UART [1].

**Describe the solution you'd like**
BLE UART module that would implement wireless CONSOLE for wireless testing, just as it is done for over-the-wire connection.

**Describe alternatives you've considered**
There is no alternative at this moment to use CONSOLE over the BLE connection. Creating custom BLE service for variable amount of control commands and debug log is not an option. Simple CONSOLE reroute to BLE UART seems simplest and most sensible solution.

**Additional context**
[1] https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.0.0%2Fble_sdk_app_nus_eval.html.

cederom, 2021-05-08:
This could implement underlying `SHELL_BT_NUS`: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/include/shell/shell_bt_nus.html :-)

jori-nordic, 2023-02-16:
closing as duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/12714 .
When we have a UART driver that goes over BLE, console output will just work.

closed 2023-02-16 by jori-nordic

vChavezB, 2023-03-28:
I am interested in this feature as I have IoT devices that I need to check over the Zephyr Logging Module. It would be nice to change the Logging module to point to a BLE Service that sends over a virtual UART driver. Are there any starting points where I could look to check this out?

Would the [HCI UART](https://github.com/intel/zephyr/tree/main/samples/bluetooth/hci_uart) and NRF BT Nus Shell be a good starting point?

jori-nordic, 2023-03-28:
You could write a re-implementation of the NUS. It's not a super-complicated service.
You can't just copy the code though, even if it's open source as it's under a different license.

hci_uart has nothing to do with what you want, it exposes the Bluetooth controller over UART.

What you want is a module that moves data back and forth over the nordic uart service or a similar one. NUS is nice because you have built-in support in the nRF connect phone and desktop apps.

You can also try to implement it using l2cap dynamic channels, that could also work, it's a lower level in the stack.

vChavezB, 2023-03-28:
Thanks for the info. I am not well familiar with the Bluetooth stack and have used mostly the high-levle zephyr API. 
I am already using as a base the uart driver api to build a virtual driver that connects to a UART service. Perhaps I will use the same NUS UUID so I can use the app from NRF or the standard UART UUID service for bluetooth.

A question that I still have is how do you think it would be best to deisgn this driver as its not a hardware component. The main objectives of the driver are

- Obtain the data recieved from the `poll_out ` uart api. 
- After a certain timeout or full buffer send the data through a BLE notification.

My questions:

- Would there be only one BLE Service for UART or multiple instances (is there a practical  use case for multiple instances?)
- Would it be the job of the end-user to set the bluetooth connection, advertisement parameters,etc ? We have to remember that the BLE Virtual UART service is mainly used for debugging or mantainance. If I have a BLE App I wouldnt like that this driver overrides my BLE Connection callbacks and configurations.
- Under which folder should I develop the driver? Its not a bluetooth stack but rather a virtual uart driver that is implemented over a BLE Service. Would it be better to be under the "serial" folder?

jori-nordic, 2023-03-28:
If you only care about logging, you could start with just a logging backend and do a console driver only if necessary. That way you avoid messing with devicetree and drivers in general.

The new logging backend would be placed e.g. in `zephyr/subsys/logging/backends/log_backend_ble.c`.

If I were you I would look at the `_net` implementation, namely `zephyr/subsys/logging/backends/log_backend_net.c` and `zephyr/samples/net/syslog_net/src/main.c`. It has the same problems as Bluetooth, where you need to establish the network first.

I think it's fair to leave it to the app to setup the connection to log on, and select which one to use (if there are concurrent conns).

vChavezB, 2023-03-28:
Thanks for the info. I already made a pull-request in case its of interest to add a ble backend logger to zephyr :)