zephyrproject-rtos/zephyr #34950: xtensa arch ：The source code version is too old

hongshui3000, 2021-05-07:
**Describe the bug**
I tried the latest xtensa compilation tool to compile zephyr, and found that the ADDI.A instruction could not be recognized (because my dsp does not support this instruction but there is this instruction in my tie-asm.h).

The latest XTOS has the following changes:
The "addi.a" is an assembler macro for "addi" instruction, conditionally defined in <xtensa/coreasm.h>.
#ifndef XT_ADDI_A
     .macro addi.a a,b,c; addi \a, \b, \c; .endm
#endif

Referring to header file, <xtensa/config/tie-asm.h>, the addi.a macro gets used in assembler macros xchal_cp1_store() and xchal_cp1_load().

And zephyr is still using the old xtensa/coreasm.h

There will be such problems in 2020 and 2021 xtensa compilation tools

andyross, 2021-05-07:
> And zephyr is still using the old xtensa/coreasm.h

Um... where?  I don't see that syntax anywhere in the tree.  In fact Zephyr is almost entirely XTOS/HAL-free at this point, with just a few spots in the startup code and the esp32-contributed stack walker.

hongshui3000, 2021-05-08:
@andyross 
The files in the following directories have the syntax I’m talking about
C:\usr\xtensa\XtDevTools\install\tools\RI-2021.6-win32\XtensaTools\xtensa-elf\include\xtensa\coreasm.h

The reasons for this usage are:
The software configuration (hifi4) generated by the new configuration tool of candenc will “tie-asm.h”contain the instruction of addi.a (this instruction does not exist in hifi4), which requires macro replacement

The tie-asm.h header file is used in the zephyr xtensa assembly. At present, an error occurs when the macro is expanded

tie-asm.h：modules\hal\xtensa\zephyr\soc\mysoc\xtensa\config\tie-asm.h（My own hifi4 configuration）

The following is cadence's response to this question:
[image]

The porting of the new xtensa soc affected by this problem

dcpleung, 2021-05-08:
That `coreasm.h` comes from Cadence's toolchain and is not distributed by Zephyr.

hongshui3000, 2021-05-08:
@dcpleung 
These porting used by zephyrrtos may be a bit outdatedf
modules\hal\xtensa\include\xtensa\coreasm.h

andyross, 2021-05-10:
Those files are copied directly from the toolchain associated with those platforms.  They are stored separately because we have (somewhat clumsily) tried to unify our toolchains in the core platforms so the Zephyr SDK doesn't have to include a dozen separate xtensa compilers.  When the toolchain changes, they will change.

If you're building a platform for Xtensa IP not included in the Zephyr core platforms, you will no doubt be using your own toolchain (or gcc overlay) that you got from Cadence.  You won't (can't) be using these HAL files, unless there's a bug in the include path somewhere.

closed 2021-05-10 by andyross