zephyrproject-rtos/zephyr #34962: tfm: cmake: Toolchain not being passed into psa-arch-tests

microbuilder, 2021-05-07:
**Describe the bug**

When running unit tests for `psa_arch_tests` with TF-M, cmake falls back to the default C compiler, rather than forwarding the correct ARM cross-compiler, resulting in build failure.

The use of the host compiler is caused by this: https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/master/psa-arch-tests/api-tests/CMakeLists.txt#L415-L436

Where the compiler is not passed on, but a C compiler is being looked up here:
https://github.com/zephyrproject-rtos/trusted-firmware-m/blob/51cdecd6f9e6b0aa66da45db22f4b478183d472f/psa-arch-tests/api-tests/tools/scripts/target_cfg/CMakeLists.txt#L29

```cmake
project(TargetConfigGen LANGUAGES C)
```

**To Reproduce**

```bash
$ ./scripts/twister -p mps2_an521_nonsecure -N --inline-logs -s samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test
```

This will result in trying to run the system C compiler (`/Library/Developer/CommandLineTools/usr/bin/cc` here, on OS X):

```
[2/5] Building C object CMakeFiles/TargetConfigGen.dir/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c.o
FAILED: CMakeFiles/TargetConfigGen.dir/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c.o 
/Library/Developer/CommandLineTools/usr/bin/cc  -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/val/nspe -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/val/common -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/common/nspe -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/common/nspe/crypto -I/Users/kevin/Linaro/zephyr/modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/tgt_dev_apis_tfm_an521/nspe -arch arm64 -o CMakeFiles/TargetConfigGen.dir/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c.o -c /Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c
/Users/kevin/Linaro/zephyr/zephyr/twister-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test/tfm/app/psa_api_tests/targetConfigGen.c:1:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
```

**Expected behavior**

The toolchain should be passed on in the cmake function when running the PSA tests.

**Additional context**

This fix should happen upstream in TF-M, but the issue is being filed here for tracking purposes until it is resolved either upstream and/or downstram in Zephyr's TF-M module.

oyvindronningstad, 2021-05-07:
This is psa-arch-tests. An issue can be opened here: https://github.com/ARM-software/psa-arch-tests/issues

microbuilder, 2021-05-07:
I'll file one and put a change request together, and update the issue here with a link.

microbuilder, 2021-05-07:
https://github.com/ARM-software/psa-arch-tests/issues/265

carlescufi, 2021-05-19:
@tejlmand is not the right assignee here, added relevant ones.

carlescufi, 2021-05-19:
@galak this is an upstream TF-M issue, so you could exclude it from the release bug tally.

ioannisg, 2021-05-20:
Well, it is an issue, since we integrate the PSA arch tests in Zephyr. So I am also fine with the LOW bug here :)
 @microbuilder do you think you could send a patch to fix this? Downstream is fine (then if it is fixed upstream we can do an upmerge and remove the downstream fix)

microbuilder, 2021-05-20:
@ioannisg will do 👍

microbuilder, 2021-05-20:
@ioannisg @oyvindronningstad https://github.com/ARM-software/psa-arch-tests/issues/265 allows the API tests to be run with:

```bash
$ ./scripts/twister -p mps2_an521_nonsecure -N --inline-logs \
  -s samples/tfm_integration/tfm_psa_test/sample.tfm.psa_internal_trusted_storage_test
```

If `CROSS_COMPILE` is not defined, it should continue to fall back to the default C compiler as before.

Tested on Darwin with `arm-none-eabi-gcc`.

microbuilder, 2021-05-25:
Fixed by #35510

PR #35510 (2021-05-26): manifest: update TF-M with latest upstream
The TF-M update brings in the latest PSA crypto headers,
after TF-M switched formally to MBedTLS 2.26.0. It also
brings in some important fixes for cache enabling in nRF
platforms.

Fixes #33423.
Fixes #33667.
Fixes #34962 

Supersedes #35455
closed 2021-05-26 by galak