zephyrproject-rtos/zephyr #34997: Zephyr docs inconsistently report module Kconfig vars

gregshue, 2021-05-08:
**Describe the bug**
When building Zephyr documentation from a workspace including a module with board and driver Kconfig settings, the documentation includes the Kconfigs of the driver in the module but not the Kconfigs of the board in the module.

**To Reproduce**
Steps to reproduce the behavior:
1. west init -m https://github.com/zephyrproject-rtos/example-application --mr main my-workspace
2. cd my-workspace
3. west update
4. cd zephyr
5. source zephyr-env.sh
6. cd doc
7. make

**Expected behavior**
I expect the Kconfigs of the board in the module to appear in the Zephyr documentation (since Zephyr allows me to build hello_world with the board of the module).  Similarly, I expect that board's eventual .rst documentation to appear in the supported boards list of Zephyr.

**Impact**
An annoyance for now, but this can block delivering complete documentation for contracted solutions built on Zephyr LTS.

**Logs and console output**
I have no idea which things would be useful.  Since this is independent of any target system I will skip attaching any generated output.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used
    - https://github.com/zephyrproject-rtos/example-application@992b90172b31c120e01d98359f6329bd8602b2a9
       NOTE: example-application/west.yml imports the latest zephyr, which checked out:
       - https://github.com/zephyrproject-rtos/zephyr@03f097b17897b4005243ab7e4a0f4c32bf7b69b7

**Additional context**
When I symbolically linked example-application/boards/arm/custom-plank to appear as zephyr/boards/arm/custom-plank and rebuilt, the board's Kconfig settings appeared in the re-generated documentation.

gmarull, 2021-05-10:
As of today only Kconfig options in https://github.com/zephyrproject-rtos/zephyr/tree/master/modules are part of the Kconfig, not external modules.

gregshue, 2021-05-10:
@gmarull, The current implementation _is_ generating some content from external modules (e.g. example-application), so I'm confused by your statement. I expect the broad impact of adding features to module.yml has not been assessed yet. Are you trying to express the current  the design intent, projected requirements, a decision after an impact assessment, or something else?

gregshue, 2021-05-10:
@gmarull I verified in my local environment that a full, clean doc `make` does generate an entry on the All Configuration Options page for CONFIG_EXAMPLESENSOR (from the external example-application module).  Since I didn't know whether the bug was that this content appears or that the board content doesn't, I reported it as "inconsistent".

> As of today only Kconfig options in https://github.com/zephyrproject-rtos/zephyr/tree/master/modules are part of the Kconfig docs, not external modules.

This happens to be true of the published docs, but does not match the **behavior** of the current tool.  As a user/integrator, I need to know how the tools are supposed to behave as I add modules to my build. I expect that the impact of `module.yml` on documentation generation and structure has not been assessed.  If they have, then please point me toward the conclusions.

gmarull, 2021-05-20:
Removing the `bug` label, since this is actually a feature request. https://github.com/nrfconnect/sdk-nrf/ can be taken as an example of docs for a complex module.

gregshue, 2021-05-21:
@gmarull As clarified by @galak during a recent triage meeting, because the Zephyr document generation behaves *inconsistently* this is considered a bug (even if resolving the bug involves an enhancement). Please restore the `bug` label.

We each have examples for docs for a complex module, but none are under the `zephyrproject-rtos` organization, so none are suitable for verifying the Zephyr document generation system.  This module is the first one in `zephyrproject-rtos` scoped to verify the extensibility available through `zephyr/module.yml`, so will be a test input to Zephyr.  Now we just have to figure out what Zephyr doc generation *should* do with it.

gmarull, 2021-05-21:
If I'm not wrong Zephyr docs do not claim to offer the described feature, this is why I think it can't be cobsidered a bug. Feel free to add the label back though.

I'll remove my assignment since I think Zephyr docs are and should be designed for the project itself. How external projects use it may vary, there are many ways to do that (eg NCS approach, simple usage of intersphinx, etc)

gregshue, 2021-05-21:
@gmarull, I do not have permission to add tags.  Please restore the `bug` tag that you remove.  Here's my rationale:

> If I'm not wrong Zephyr docs do not claim to offer the described feature,

From [Documentation Overview](https://docs.zephyrproject.org/latest/README.html#documentation-overview):
> Script-generated material for kernel configuration options based on Kconfig files found in the source code tree, ...

From [Developer-mode Document Building](https://docs.zephyrproject.org/latest/README.html#developer-mode-document-building):
> Building the documentation for all the Kconfig options ...

To me those sound like pretty overt claims to support this feature. 

Also consider the following evidence that the documentation system is expected to be used beyond the project website and is implicitly expected to describe the entire workspace:
-  [Documentation Overview](https://docs.zephyrproject.org/latest/README.html#documentation-overview):
   - > This same .rst content is also fed into the Zephyr Project’s public website documentation area (with a different theme applied).
   - > Doxygen-generated material used to create all API-specific documents
   - > Most of the reStructuredText sources are found in the /doc directory, but others are stored within the code source tree near their specific component (such as /samples and /boards)
- The existence of the API Reference page
- The existence of the External Module Samples page

gregshue, 2021-05-21:
Even if you interpret the the feature to be limited to the Zephyr repository or modules included in the default manifest, the current behavior goes beyond that and so is a bug.

gmarull, 2021-05-21:
The Zephyr documentation is designed to work for the Zephyr repository, how modules/applications use the Zephyr docs is a separate topic. Applications/modules are free to use whatever they consider best for their documentation, Sphinx is not the only available option. If they choose Sphinx, they can refer to Zephyr docs using `intersphinx` and can re-use dt/kconfig scripts to generate docs for their own Kconfig (including Zephyr ones if they prefer).

> Even if you interpret the the feature to be limited to the Zephyr repository or modules included in the default manifest, the current behavior goes beyond that and so is a bug.

As far as I understand Zephyr supports building its docs. Beyond that point, what works and what doesn't should be considered *undefined behavior*.

If you consider this a critical feature I'd suggest sending an RFC with a proof of concept of how you think docs should work.

gmarull, 2021-05-21:
An example of what I consider a well-designed module with a custom documentation that has no relation with Zephyr nor Sphinx:
https://github.com/zscilib/zscilib

gregshue, 2021-05-21:
> Beyond that point, what works and what doesn't should be considered undefined behavior.

Yes, the behavior is undefined, but **the triage team assessed the current _inconsistent_ behavior as a "bug"**.  It seems it is up to you to justify to them that the "bug" label should be removed.

gregshue, 2021-05-21:
The module you reference states "the main development target for the library is the Zephyr Project," and has:
- .rst files in the samples with Zephyr-style documentation
- Doxygen comments in the headers
- `zephyr_*` functions in the CMakeLists.txt
- A Kconfig file that is seems only to be referenced from `zephyr/module.yml`

Though it may not be sponsored or managed by the Zephyr Project, it is explicitly related to Zephyr and implicitly is intended to have its content included when Zephyr documentation is generated with this in the codebase.  If this included a board definition it also would have shown the current behavior assessed as a bug.

gregshue, 2021-05-28:
@galak, the bug label was removed from this issue since the required behavior is unspecified.  During the triage meeting a few weeks ago we discussed the criteria for "bug" vs "enhancement", and concluded the inconsistent behavior should be a "bug".  If you still come to the same conclusion, please restore the "bug" label, as I do not have permission to do so.

kartben, 2024-02-08:
Closing as won't fix. As mentioned by @gmarull on multiple occasions in this thread, I don't think it's practical nor desirable at this point to extend the scope of or documentation tooling to things that are not Zephyr itself.

closed 2024-02-08 by kartben