zephyrproject-rtos/zephyr #35027: frdm_k64f: failed to run testcase tests/drivers/adc/adc_emul/

Zhaoningx, 2021-05-10:
To Reproduce

Failed to run testcase tests/drivers/adc/adc_emul/

Steps to reproduce the behavior:

export ZEPHYR_SDK_INSTALL_DIR=~/zephyr-sdk-0.12.4

twister -p frdm_k64f --device-testing --device-serial /dev/ttyACM0 -T tests/drivers/adc/adc_emul/

see error

@I: adc init done
I: adc init done
*** Booting Zephyr OS build zephyr-v2.5.0-3742-g0622bde3bf22  ***
Running test suite adc_basic_test

===================================================================
START - test_adc_emul_single_value
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x20002de8
E:   NXP MPU error, port 3
E:     Mode: User, Data Address: 0x20002de8
E:     Type: Read, Master: 0, Regions: 0x8100
E: r0/a1:  0x20002de4  r1/a2:  0x00000000  r2/a3:  0x000005dc
E: r3/a4:  0x00000040 r12/ip:  0x0000c189 r14/lr:  0x00001319
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x000045be
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200000e0 (unknown)
E: Halting system

Environment
OS: Fedora33
Toolchain:zephyr-sdk-0.12.4
Commit ID: 0622bde

erwango, 2021-05-10:
Test is also failed on STM32 based platforms.
From what I understand it is meant to be tested on native_posix only and .yaml filter should be fixed.
Cf https://github.com/zephyrproject-rtos/zephyr/pull/34640

^^ @semihalf-michalec-tomasz

semihalf-michalec-tomasz, 2021-05-11:
I thought that ADC emulator code doesn't use posix specific functions and it compiled fine for other platforms.  Since I don't see use case for ADC emulator on different platforms than native_posix, I will prepare patch for this test .yaml.

PR #35090 (2021-05-11): adc_emul: case can only run on native_posix board
we meet hardfault when run on real board.
as the device is a virtual device, this 

I: adc init done

*** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***

Running test suite adc_basic_test

===================================================================

START - test_adc_emul_single_value

E: ***** BUS FAULT *****

E:   Precise data bus error

E:   BFAR Address: 0x20002d00

E:   NXP MPU error, port 3

E:     Mode: User, Data Address: 0x20002d00

E:     Type: Read, Master: 0, Regions: 0x8100

E: r0/a1:  0x20002cfc  r1/a2:  0x00 [... cut]
commit 53b8706f (2021-05-11): adc_emul: case can not run on native_posix board

we meet hardfault when run on real board.
as the device is a virtual device

I: adc init done

*** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***

Running test suite adc_basic_test

===================================================================

START - test_adc_emul_single_value

E: ***** BUS FAULT *****

E:   Precise data bus error

E:   BFAR Address: 0x20002d00

E:   NXP MPU error, port 3

E:     Mode: User, Data Address: 0x20002d00

E:     Type: Read, Master: 0, Regions: 0x8100

E: r0/a1:  0x20002cfc  r1/a2:  0x00000000  [... cut]