zephyrproject-rtos/zephyr #35043: NXP: Build error : ModuleNotFoundError: No module named 'elftools'

shehamb, 2021-05-10:
Hello,
I am trying to build sample application on for mimxrt1064_evk on windows 10. I seem to have installed everything, but i still get this error:
C:\Users\mbajaj\zephyrproject\zephyr>west build -p auto -b mimxrt1064_evk samples/hello_world
[1/130] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
FAILED: zephyr/include/generated/kobj-types-enum.h zephyr/include/generated/otype-to-str.h zephyr/include/generated/otype-to-size.h
cmd.exe /C "cd /D C:\Users\mbajaj\zephyrproject\zephyr\build\zephyr && C:\Python36\python.exe C:/Users/mbajaj/zephyrproject/zephyr/scripts/gen_kobject_list.py --kobj-types-output C:/Users/mbajaj/zephyrproject/zephyr/build/zephyr/include/generated/kobj-types-enum.h --kobj-otype-output C:/Users/mbajaj/zephyrproject/zephyr/build/zephyr/include/generated/otype-to-str.h --kobj-size-output C:/Users/mbajaj/zephyrproject/zephyr/build/zephyr/include/generated/otype-to-size.h --include C:/Users/mbajaj/zephyrproject/zephyr/build/zephyr/misc/generated/struct_tags.json "
Traceback (most recent call last):
File "C:/Users/mbajaj/zephyrproject/zephyr/scripts/gen_kobject_list.py", line 62, in
import elftools
ModuleNotFoundError: No module named 'elftools'
[2/130] Generating include/generated/driver-validation.h
FAILED: zephyr/include/generated/driver-validation.h
cmd.exe /C "cd /D C:\Users\mbajaj\zephyrproject\zephyr\build\zephyr && C:\Python36\python.exe C:/Users/mbajaj/zephyrproject/zephyr/scripts/gen_kobject_list.py --validation-output C:/Users/mbajaj/zephyrproject/zephyr/build/zephyr/include/generated/driver-validation.h --include C:/Users/mbajaj/zephyrproject/zephyr/build/zephyr/misc/generated/struct_tags.json "
Traceback (most recent call last):
File "C:/Users/mbajaj/zephyrproject/zephyr/scripts/gen_kobject_list.py", line 62, in
import elftools
ModuleNotFoundError: No module named 'elftools'
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\mbajaj\zephyrproject\zephyr\build'

And elftools is installed as verified by:

pip show pyelftools
Name: pyelftools
Version: 0.27
Summary: Library for analyzing ELF files and DWARF debugging information
Home-page: https://github.com/eliben/pyelftools
Author: Eli Bendersky
Author-email: eliben@gmail.com
License: Public domain
Location: c:\python39\lib\site-packages
Requires:
Required-by: pyocd

shehamb, 2021-05-10:
[https://github.com/zephyrproject-rtos/zephyr/issues/1392](url)

carlescufi, 2021-05-17:
You need to `pip3 install pyelftools` or, even better, `pip3 install zephyr/scripts/requirements.txt`

carlescufi, 2021-05-17:
Reopening since I missed some of the input

carlescufi, 2021-05-17:
I tried this locally and works fine on a Windows 10 machine. It might be that you have multiple Python 3 installations and CMake is picking one where you don't have `pyelftools` installed.

closed 2021-05-17 by carlescufi