zephyrproject-rtos/zephyr #35051: CONFIG_LOG2 fails for floating point output with warning and bad output

SwissKnife64, 2021-05-10:
The following code results in compiler warnings and wrong output.
The warning request a "double value" as argument?

==== Code:
    char strWithDouble[] = "2.718";
    double doubleValue = strtod(strWithDouble, &endPtr);

    sprintf( outputStr, "%f", doubleValue);
    LOG_INF("double value as string from sprintf %s", outputStr);

    LOG_INF("double value with format %%f %f", doubleValue);
    LOG_INF("double value with format %%g %g", doubleValue);
    LOG_INF("double value with format %%a %a", doubleValue);

==== Output:
[00:00:00.006,126] <inf> blinky_json: double value as string from sprintf 2.718000
[00:00:00.006,133] <inf> blinky_json: double value with format %f -1761758827172083000000000000000000000000000.000000
[00:00:00.006,137] <inf> blinky_json: double value with format %g -1.76176e+42
[00:00:00.006,142] <inf> blinky_json: double value with format %a -0x1.439580801013ep+140

==== CONFIG
CONFIG_LOG=y
CONFIG_LOG2_MODE_DEFERRED=y

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y

CONFIG_CBPRINTF_FP_A_SUPPORT=y

nordic-krch, 2021-05-11:
Can you specify what platform are you building and what kind of warnings you see? I tried adding your code snippet to hello world and it compiles and runs fine for qemu supporting fpu.

SwissKnife64, 2021-05-11:
No, on my nucleo_h743zi board the output is still wrong, 
only the warnings are gone.

My code:
```
    char strWithDouble[] = "2.718";
    double doubleValue = strtod(strWithDouble, &endPtr);

    sprintf( outputStr, "%f", doubleValue);
    LOG_INF("double value as string from sprintf %s", outputStr);

    LOG_INF("double value with format %%f %f", doubleValue);
    LOG_INF("double value with format %%g %g", doubleValue);
    LOG_INF("double value with format %%a %a", doubleValue);

    char strWithFloat[] = "3.1415";
    float floatValue = strtof(strWithFloat, &endPtr);

    sprintf( outputStr, "%f", floatValue);
    LOG_INF("float value as string from sprintf %s", outputStr);

    LOG_INF("float value with format %%f %f", floatValue);
    LOG_INF("float value with format %%g %g", floatValue);
    LOG_INF("float value with format %%a %a", floatValue);

```
My Config:
```
CONFIG_SHELL=y
CONFIG_LOG_MAX_LEVEL=3

CONFIG_GPIO=y
CONFIG_MAIN_STACK_SIZE=4096

CONFIG_LOG=y
CONFIG_LOG2_MODE_DEFERRED=y

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y

CONFIG_FPU=y
CONFIG_CBPRINTF_FP_SUPPORT=y
CONFIG_CBPRINTF_FP_A_SUPPORT=y

```
My CMakeList.txt
```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.13.1)

# START added for CLion
#set(ENV{BOARD} smartguardcom)
set(ENV{BOARD} nucleo_h743zi)

set(ENV{ZEPHYR_SDK_INSTALL_DIR} /usr/local/zephyr-sdk-0.12.3)
set(ENV{ZEPHYR_TOOLCHAIN_VARIANT} gnuarmemb)
# STOP added for CLion

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(blinky)

target_sources(app PRIVATE src/main.c)

```
My output
```
[00:00:00.006,032] <inf> blinky_json: intValue 42
[00:00:00.006,106] <inf> blinky_json: double value as string from sprintf 2.718000
[00:00:00.006,112] <inf> blinky_json: double value with format %f -1761758827171789000000000000000000000000000.000000
[00:00:00.006,116] <inf> blinky_json: double value with format %g -1.76176e+42
[00:00:00.006,121] <inf> blinky_json: double value with format %a -0x1.439580800fd88p+140
[00:00:00.006,160] <inf> blinky_json: float value as string from sprintf 3.141500
[00:00:00.006,165] <inf> blinky_json: float value with format %f -2.000000
[00:00:00.006,170] <inf> blinky_json: float value with format %g -2
[00:00:00.006,174] <inf> blinky_json: float value with format %a -0x1.000000800fe0fp+1

```

nordic-krch, 2021-05-17:
It is in shell log backend. Shell copies log messages to its own buffer. Log messages must be stored in the buffer which is properly aligned (by default 8 bytes unless long double support is enabled). Shell buffer was not aligned and values exceeding word size were corrupted when formatted by cbprintf.

PR #35332 (2021-05-18): shell: log_backend: Add mpsc_pbuf buffer alignment
Buffer used for storing log messages must be aligned as specified
by Z_LOG_MSG2_ALIGNMENT. Added missing alignment.

Fixes #35051.

Note, that buffer is currently fixed size. Must be changed to configurable value in the future.
closed 2021-05-18 by galak