zephyrproject-rtos/zephyr #35076: APIC TSC Deadline Timer does not support CPUs >1

jenmwms, 2021-05-10:
**Describe the bug**
The new APICTSC Deadline Timer supports SMP=y configuration, but does not yet work with more than one CPU. 

**To Reproduce**
Steps to reproduce the behavior:
1. Configure acrn_ehl_crb_defconfig with CONFIG_MP_NUM_CPUs=2
2. west build -p -b acrn_ehl_crb zephyr/tests/kernel/interrupt
3. Flash and observe console.

**Expected behavior**
Build successfully and use CONFIG_MP_NUM_CPUs >1 without issue, platform-wide tests. This is just example.

**Impact**
cannot use SMP with more than 1 CPU with APIC timer. showstopper.

**Logs and console output**
```
DEBUG   - DEVICE: ACRN:\>vm_console 0
DEBUG   - DEVICE: VM is not valid
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Error: Invalid parameters.
DEBUG   - Timed out while monitoring serial output on acrn_ehl_crb
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 2.4
 - Commit SHA 4ee61309b1203abaa4ed0a35c479e20d3dbb6155

andyross, 2021-05-10:
FWIW: it's very unlikely to be the timer driver if the system is dead or faulted or whatever state it's in by the time get get to the ACRN prompt.  Most likely it's being terminated for sending a startup IPI to a bad APIC ID or some other kind of misbehavior.  Timer setup happens later.

jenmwms, 2021-05-10:
Thanks Andy! I'll have to work on better description here, then. Sounds like moving to the APIC timer driver which supports SMP enabled us to start trying > 1 CPUs on this board?

jenmwms, 2021-05-12:
Found out that ACRN does not use the same values for local APIC IDs as the native EHL CRB. They use use virtualized APIC ID.
For a quad-core CPU: { 0x0, 0x1, 0x2, 0x3 };

andyross, 2021-05-14:
Partial fix is up at #35310.  The root cause is, again, the fact that ACRN hard-coded the entry point, which means that the same address (0x7000) needs to function as both the 32 bit grub/multiboot-style entry point for OS startup **and** the 16 bit entry point for the other SMP CPUs once they get launched form a startup IPI.  I'm... not proud of what I did to work around this, but it does work (and on other x86_64 platforms, I checked).

But it's not complete.  Now we get all the way through boot and into the second CPU startup.  But then it hangs once it sends the IPI.  And on the ACRN console I see:

```
[22960816us][cpu=0][(null)][sev=2][seq=2]:HV version 2.4-2021-04-16 16:27:49-c80508fd7 DBG (daily tag:acrn-2021w11.5-180000p) hybrid@ehl-crb-b build by mm, start time 22948191us
[22978647us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[22984587us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Genuine Intel(R) CPU 0000 @ 1.90GHz
[24001366us][cpu=3][(null)]2][seq=21]:Start VM id: 0 name: ACRN PRE-LAUNCHED VM0
[24009559us][cpu=3][vm0:vcpu0][sev=3][seq=29]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa
[24010775us][cpu=0][(null)][sev=2][seq=32]:Start VM id: 1 name: ACRN SOS VM
[24023408us][cpu=3][vm0:vcpu0][sev=3][seq=35]:vlapic: vm_apicid2vcpu_id: bad lapicid 1
[24029366us][cpu=0][vm1:vcpu0][sev=3][seq=36]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa
[24048274us][cpu=3][vm0:vcpu0][sev=3][seq=37]:vlapic: vm_apicid2vcpu_id: bad lapicid 1ACRN:\>
[24598624us][cpu=0][vm1:vcpu0][sev=3][seq=40]:vlapic: Start Secondary VCPU1 for VM[1]...
[24607574us][cpu=1][vm1:vcpu1][sev=3][seq=41]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa
[24620997us][cpu=0][vm1:vcpu0][sev=3][seq=43]:vlapic: Start Secondary VCPU2 for VM[1]...
[24629937us][cpu=2][vm1:vcpu2][sev=3][seq=44]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa
```

Note that for "vm0" (that's us, the pre-launched VM), there is onl one "vm0:vcpu0" line announcing some kind of warning about VMX-whatever not fully enabled, and that there's a "[cpu=3]" block presumably telling us that it's been assigned to the fourth physical CPU on the 4-cpu device.  But for "vm1" (Linux), you see lines about "vm1:vcpu0", "vm1:vcpu1", and "vm1:vcpu2" correponding to physical CPUs 0, 1, and 2.  Finally, note the error "bad lapicid 1" which is being tossed right as we are trying to send a startup IPI to APIC ID "1".

Basically: I strongly suspect that ACRN has been misconfigured for SMP, and that we've been assigned only one CPU.  Someone with more ACRN configuration knowledge needs to fix that.  But I'm willing to be that this is the last major missing piece.  There is no more SMP-related hardware work to do after that IPI gets handled.

andyross, 2021-05-14:
And just to be clear about my feelings on the issue: **SOMEONE NEEDS TO GET ACRN TO STOP HARDCODING JUNK FROM ZEPHYR AND MAKE THEM USE OUR IMAGE AS WE BUILD IT.  THIS HAS BEEN HAPPENING REPEATEDLY FOR YEARS AND I'M GETTING KINDA SICK OF IT**.

PR #35786 (2021-06-04): ACRN SMP fixes
Fixes needed to get Zephyr running with more than one CPU under the ACRN hypervisor.  Nothing really complicated, though the first patch is an ugly hack.

See earlier conversation in #35310 which contains a different version of the first patch.

Fixes #35076
commit 9cb8dcbf (2021-06-04): arch/x86_64: Use modern CR0 assembly

The 16 bit bootstrap code for SMP CPUs was using the 286-era "lmsw"
instruction (load machine status word) to set the protected bit in CR0
(which is the modern evolution of the same register), presumably
because this is 16 bit code and we can't move a dword into CR0.

But that's wrong, because the full instruction set *is* available in
real mode on a 386, you just have to use a operand size prefix to get
to it, which the assembler emits for you automatically when you use
the .code16 directive.

Write this conventionally and use modern (e.g. 1986-era) instr [... cut]
closed 2021-06-04 by galak