zephyrproject-rtos/zephyr #35089: stm32h7: systematic crash at each second boot with NETWORKING=y

erwango, 2021-05-11:
**Describe the bug**
On STM32H7 platform, a systematic error happens at each second boot.
git bisect leads to following PR: https://github.com/zephyrproject-rtos/zephyr/pull/34086

**To Reproduce**
Use any sample that uses `NETWORKING=y` on any h7 based board.
For instance:
`west build -b nucleo_h745zi_q_m7 samples/posix/eventfd/`

**Expected behavior**
No crash at boot

**Impact**
Blocker

**Logs and console output**
```
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
arch_system_halt (reason=reason@entry=0) at /local/mcu/zephyrproject/zephyr/kernel/fatal.c:32
32		}
(gdb) bt
#0  arch_system_halt (reason=reason@entry=0) at /local/mcu/zephyrproject/zephyr/kernel/fatal.c:32
#1  0x08012204 in k_sys_fatal_error_handler (reason=reason@entry=0, esf=esf@entry=0x24003a80 <z_interrupt_stacks+2016>) at /local/mcu/zephyrproject/zephyr/kernel/fatal.c:44
#2  0x08012228 in z_fatal_error (reason=0, esf=esf@entry=0x24003a80 <z_interrupt_stacks+2016>) at /local/mcu/zephyrproject/zephyr/kernel/fatal.c:127
#3  0x0800e842 in z_arm_fatal_error (reason=<optimized out>, esf=esf@entry=0x24003a80 <z_interrupt_stacks+2016>) at /local/mcu/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:62
#4  0x080043c2 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>)
    at /local/mcu/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:1034
#5  0x0800423c in z_arm_usage_fault () at /local/mcu/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102

```

mnkp, 2021-05-11:
This seems to be related to #35107. STM32H7 platform also requires `CONFIG_NOCACHE_MEMORY=y`.

galak, 2021-05-18:
Fixed by #35258

closed 2021-05-18 by galak