zephyrproject-rtos/zephyr #35099: benchmark.kernel.application.fp.arm: Illegal load of EXC_RETURN into PC on lpcxpresso55s16_ns and lpcxpresso55s69_ns

hakehuang, 2021-05-11:
**Describe the bug**
Illegal load of EXC_RETURN into PC when run benchmark.kernel.application.fp.arm with FP config enabled

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/benchmarks/app_kernel; mkdir build; cd build
2. west build -b lpcxpresso55s16_ns -- -DCONF_FILE=prj_fp.conf
3. make; make flash
4. See error

**Expected behavior**
test pass

**Impact**
float point handler

**Logs and console output**
```

    *** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***

    E: ***** USAGE FAULT *****

    E:   Illegal load of EXC_RETURN into PC

    E: r0/a1:  0x20003070  r1/a2:  0x000029a1  r2/a3:  0x00002901

    E: r3/a4:  0x000029cd r12/ip:  0x000029cd r14/lr:  0x000029cd

    E:  xpsr:  0x00002800

    E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000

    E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000

    E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000

    E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000

    E: fpscr:  0x00002979

    E: Faulting instruction address (r15/pc): 0x000029cd

    E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0

    E: Current thread: 0x200002f8 (RECVTASK)

    E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc1

hakehuang, 2021-05-11:
@MaureenHelm , @dleach02 , @mmahadevan108

henrikbrixandersen, 2021-05-18:
I cannot reproduce this on the lpcxpresso55s16 with the changes in #35416.

henrikbrixandersen, 2021-05-19:
```
*** Booting Zephyr OS build v2.6.0-rc1-233-gd381ca924c51  ***

|-----------------------------------------------------------------------------|
|          S I M P L E   S E R V I C E    M E A S U R E M E N T S  |  nsec    |
|-----------------------------------------------------------------------------|
|-----------------------------------------------------------------------------|
| enqueue 1 byte msg in FIFO                                       |      3501|
| dequeue 1 byte msg in FIFO                                       |      3481|
| enqueue 4 bytes msg in FIFO                                      |      4043|
| dequeue 4 bytes msg in FIFO                                      |      3960|
| enqueue 1 byte msg in FIFO to a waiting higher priority task     |     18845|
| enqueue 4 bytes in FIFO to a waiting higher priority task        |     18949|
|-----------------------------------------------------------------------------|
| signal semaphore                                                 |      2376|
| signal to waiting high pri task                                  |     17019|
| signal to waiting high pri task, with timeout                    |     21213|
|-----------------------------------------------------------------------------|
| average lock and unlock mutex                                    |      4432|
|-----------------------------------------------------------------------------|
| average alloc and dealloc memory page                            |      1219|
|-----------------------------------------------------------------------------|
|                M A I L B O X   M E A S U R E M E N T S                      |
|-----------------------------------------------------------------------------|
| Send mailbox message to waiting high priority task and wait                 |
| repeat for  128 times and take the average                                  |
|-----------------------------------------------------------------------------|
|   size(B) |       time/packet (nsec)       |          KB/sec                |
|-----------------------------------------------------------------------------|
|          0|                           31444|                               0|
|          8|                           31442|                             254|
|         16|                           31442|                             508|
|         32|                           31442|                            1017|
|         64|                           31442|                            2035|
|        128|                           31442|                            4070|
|        256|                           31442|                            8141|
|        512|                           31442|                           16283|
|       1024|                           31442|                           32567|
|       2048|                           31442|                           65135|
|       4096|                           31439|                          130284|
|-----------------------------------------------------------------------------|
| message overhead:       31444     nsec/packet                               |
| raw transfer rate:              0 KB/sec (without overhead)                 |
|-----------------------------------------------------------------------------|
|                   P I P E   M E A S U R E M E N T S                         |
|-----------------------------------------------------------------------------|
| Send data into a pipe towards a receiving high priority task and wait       |
|-----------------------------------------------------------------------------|
|                          matching sizes (_ALL_N)                            |
|-----------------------------------------------------------------------------|
|   size(B) |       time/packet (nsec)       |          KB/sec                |
|-----------------------------------------------------------------------------|
| put | get |  no buf  | small buf| big buf  |  no buf  | small buf| big buf  |
|-----------------------------------------------------------------------------|
|    8|    8|     33321|     33321|     33322|       240|       240|       240|
|   16|   16|     33904|     33904|     33905|       471|       471|       471|
|   32|   32|     35071|     35071|     35072|       912|       912|       912|
|   64|   64|     37404|     37404|     37405|      1711|      1711|      1711|
|  128|  128|     42071|     42071|     42072|      3042|      3042|      3042|
|  256|  256|     51404|     51404|     51405|      4980|      4980|      4980|
|  512|  512|     70071|     70071|     70072|      7306|      7306|      7306|
| 1024| 1024|    107404|    107404|    107405|      9534|      9534|      9534|
| 2048| 2048|    182071|    182071|    182073|     11248|     11248|     11248|
|-----------------------------------------------------------------------------|
|                      non-matching sizes (1_TO_N) to higher priority         |
|-----------------------------------------------------------------------------|
|   size(B) |       time/packet (nsec)       |          KB/sec                |
|-----------------------------------------------------------------------------|
| put | get |  no buf  | small buf| big buf  |  no buf  | small buf| big buf  |
|-----------------------------------------------------------------------------|
[... 6 lines cut ...]
[... cut ...]
|  512|   32|    275174|    276661|     49315|      1860|      1850|     10382|
| 1024|   16|    960942|    900885|     86744|      1065|      1136|     11804|
| 2048|    8|   3627479|   3256572|    161760|       564|       628|     12660|
|-----------------------------------------------------------------------------|
|         END OF TESTS                                                        |
|-----------------------------------------------------------------------------|
PROJECT EXECUTION SUCCESSFUL

```

closed 2021-05-20 by hakehuang