zephyrproject-rtos/zephyr #35102: testing.ztest.error_hook: fails on lpcxpresso55s16_ns

hakehuang, 2021-05-11:
**Describe the bug**
assert error

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/ztest/error_hook;mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s16_ns ..
3. make; make flash
4. See error

**Expected behavior**
tes pass

**Impact**
ztest error_hook support

**Logs and console output**

testing.ztest.error_hook

```
    *** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***
    Running test suite error_hook_tests
    ===================================================================
    START - test_catch_assert_fail
    ASSERTION FAIL [a != ((void *)0)] @ WEST_TOPDIR/zephyr/tests/ztest/error_hook/src/main.c:41
    	parameter a should not be NULL!
    Caught assert failed
    Assert error expected as part of test case.
     PASS - test_catch_assert_fail in 0.17 seconds
    ===================================================================
    START - test_catch_fatal_error
    case type is 0
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
    	ESF could not be retrieved successfully. Shall never occur.
    Caught assert failed
    Assert failed was unexpected, aborting...
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
    	ESF could not be retrieved successfully. Shall never occur.
    Caught assert failed
    Assert failed was unexpected, aborting...
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc1

hakehuang, 2021-05-11:
@MaureenHelm , @dleach02 , @mmahadevan108

galak, 2021-05-11:
@hakehuang are you seeing this on lpcxpresso55s69_ns?

hakehuang, 2021-05-12:
> lpcxpresso55s69_ns

@galak  55s69_ns platform does not have required CONFIG_ARCH_HAS_USERSPACE, so I do not test it.

henrikbrixandersen, 2021-05-18:
I cannot reproduce this with the changes in #35416.

galak, 2021-05-19:
Closing this, @hakehuang re-open if you still have an issue since we merged #35416.

ABOSTM, 2021-05-21:
This test is still failed on nucleo_g071rb and nucleo_l073rz despite #35416.
SHA1: 5f61d89233

henrikbrixandersen, 2021-05-21:
> This test is still failed on nucleo_g071rb and nucleo_l073rz despite #35416.
> 
> SHA1: 5f61d89233

Please update the PR (or open a new one?) to reflect which board it is failing on. The problem was fixed for the lpcxpresso55s16.

ABOSTM, 2021-05-21:
OK, I close this issue and opened a new issue: #35540

closed 2021-05-21 by ABOSTM