zephyrproject-rtos/zephyr #35104: arch.interrupt.gen_isr_table.arm_mainline: fails on lpcxpresso55s16_ns

hakehuang, 2021-05-11:
**Describe the bug**
arch.interrupt.gen_isr_table.arm_mainline test failure on lpcxpresso55s16_ns
```
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/gen_isr_table/src/main.c:278: test_build_time_interrupt: (check_sw_isr(isr3, ISR3_ARG, ISR3_OFFSET) is non-zero)
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s16_ns -DCONFIG_CORTEX_M_DEBUG_NULL_POINTER_EXCEPTION_DETECTION_NONE=y ..
3. make; make flash
4. See error

**Expected behavior**
test pass

**Impact**
interrupt support

**Logs and console output**

```
  *** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***

    START - Test gen_isr_tables

    IRQ configuration (total lines 60):

    Running test suite context

    ===================================================================

    START - test_build_time_direct_interrupt

    isr1 isr=0x4ed irq=59

    isr2 isr=0x55d irq=58

    Checking _irq_vector_table entry 59 for irq 59

    triggering irq 59

    Triggering irq : 59

    isr1 ran

    Checking _irq_vector_table entry 58 for irq 58

    triggering irq 58

    Triggering irq : 58

    isr2 ran

     PASS - test_build_time_direct_interrupt in 0.21 seconds

    ===================================================================

    START - test_build_time_interrupt

    _sw_isr_table at location 0x2000006c

    isr3 isr=0x45d irq=57 param=0xb01dface

    Checking _sw_isr_table entry 57 for irq 57

    triggering irq 57

    Triggering irq : 57

    interrupt 57 didn't run once, ran 0 times

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/gen_isr_table/src/main.c:278: test_build_time_interrupt: (check_sw_isr(isr3, ISR3_ARG, ISR3_OFFSET) is non-zero)

    check interrupt isr3 failed

     FAIL - test_build_time_interrupt in 0.36 seconds

    ===================================================================

    START - test_run_time_interrupt

    isr5 isr=0x4a5 irq=55 param=0xf0ccac1a

    Checking _sw_isr_table entry 55 for irq 55

    triggering irq 55

    Triggering irq : 55

    isr5 ran with parameter 0xf0ccac1a

    isr6 isr=0x4c9 irq=54 param=0xba5eba11

    Checking _sw_isr_table entry 54 for irq 54

    triggering irq 54

    Triggering irq : 54

    isr6 ran with parameter 0xba5eba11

     PASS - test_run_time_interrupt in 0.28 seconds

    ===================================================================

    Test suite context failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc1

hakehuang, 2021-05-11:
@MaureenHelm,@dleach02 , @mmahadevan108

henrikbrixandersen, 2021-05-19:
I believe this is identical to #34915.

closed 2021-05-19 by galak