zephyrproject-rtos/zephyr #35107: Atmel SAM E70 / Cortex-M7 fails to boot if CONFIG_NOCACHE_MEMORY=y

mnkp, 2021-05-11:
**Describe the bug**
If the application is compiled with CONFIG_NOCACHE_MEMORY=y it crashes early during the Zephyr's boot process. No log output can be observed. In case of Atmel SAM platform it may reboot successfully after watchdog reset, which happens after about 20 s. After the first successful boot the subsequent reboots often run without any issues.

The crash happens before the logger subsystem is initialized, however analyzing `z_arm_fault` reveals that it takes place when executing
```
        SCB->DCCISW = (((sets << SCB_DCCISW_SET_Pos) & SCB_DCCISW_SET_Msk) |
                       ((ways << SCB_DCCISW_WAY_Pos) & SCB_DCCISW_WAY_Msk)  );
```
in `SCB_CleanInvalidateDCache()` located in `zephyrproject/modules/hal/cmsis/CMSIS/Core/Include/cachel1_armv7.h:306`

The issue was introduced by d6b50233acc7e27030bfbf1bd069d0f6969a4f68 commit. Reverting the commit fixes the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b sam_e70_xplained -- -DCONFIG_NOCACHE_MEMORY=y samples/hello_world/`
2. `west flash`
3. Power cycle the board, i.e. cut off power supply for several seconds and then turn it on.
4. No "Hello World!" message can be seen in the terminal.

**Impact**
This is a show stopper for any application that requires `CONFIG_NOCACHE_MEMORY=y`. The option is required by e.g. Atmel SAM Ethernet driver.

Atmel SAM is using standard ARM MPU module. It is likely that other ARM families face the same issue.

**Environment**
 - OS: Linux
 - Toolchain zephyr 0.12.4
 - Version v2.6.0-rc1

galak, 2021-05-13:
Looks like M7 is having issue in general due to commit  d6b5023

galak, 2021-05-18:
Fixed by #35258

closed 2021-05-18 by galak

yashi, 2021-05-20:
Why do we need to _clean_ the d-cache in `z_arm_mpu_init()`? What data in no-cache area are we trying to write back to the memory?  If there is nothing we need to keep, can we just call `SCB_InvalidateDCache()`?

Just for me to understand the context:
- `CONFIG_NOCACHE_MEMORY` means we have uncached region in the system memory.
- `CONFIG_INIT_ARCH_HW_AT_BOOT` means force Zephyr to initialize the hardware to reset state.  For this case, we explicitly ask Zephyr not to initialize the hardware (which implicitly means that hardware is clean and initialized by PoR, no?)
- The D-Cache is enabled at `init_dcache()` at `PRE_KERNEL_1` after `z_arm_mpu_init()`.
- The commit ac3d9438ed4 added `CB_CleanInvalidateDCache()` in the function by @CristXu .