zephyrproject-rtos/zephyr #35154: [Coverity CID: 233524] Unchecked return value in include/drivers/dma.h

zephyrbot, 2021-05-11:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/0622bde3bf22926544c1aaef4c01dfa23d79f26e/include/drivers/dma.h

Category: Error handling issues
Function: `z_impl_dma_request_channel`
Component: Drivers
CID: [233524](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=233524)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/0622bde3bf22926544c1aaef4c01dfa23d79f26e/include/drivers/dma.h#L405

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

galak, 2021-05-11:
@hakehuang can you look at this coverity bug - we might be treating this as a false positive.

hakehuang, 2021-05-12:
@galak , I think it is not a issue, this return value is not about function sucess or not, but return the former status. here we do not need the former status, as we only need to clear the bit. 

```
/**
 * @brief Atomically test and clear a bit.
 *
 * Atomically clear bit number @a bit of @a target and return its old value.
 * The target may be a single atomic variable or an array of them.
 *
 * @param target Address of atomic variable or array.
 * @param bit Bit number (starting from 0).
 *
 * @return true if the bit was set, false if it wasn't.
 */
```

MaureenHelm, 2021-05-13:
> I think it is not a issue, this return value is not about function sucess or not, but return the former status. here we do not need the former status, as we only need to clear the bit.

Is there a reason not to use `atomic_clear_bit()` instead?

hakehuang, 2021-05-13:
> > I think it is not a issue, this return value is not about function sucess or not, but return the former status. here we do not need the former status, as we only need to clear the bit.
> 
> Is there a reason not to use `atomic_clear_bit()` instead?

right, this api is more friendly for coverity scan. I will create PR to fix it

PR #35243 (2021-05-13): dma: change an atmoic api for Coverity issue 233524
use atomic_clear_bit instead of atomic_test_and_set_bit
which will not return old value, and thus more coverity friendy

Fixes #35154
closed 2021-05-13 by galak