zephyrproject-rtos/zephyr #35169: esp32:    uart_poll_in never ready for UART2 only

aep, 2021-05-11:
for this simple uart poll

```C
#include <zephyr.h>
#include <sys/printk.h>
#include <device.h>
#include <drivers/uart.h>

void main(void)
{
    printk("Hello World! %s\n", CONFIG_BOARD);

    struct device * uart = device_get_binding("UART_2");
    if (!uart) {
        printk("no UART_2\n");
        return;
    }

    printk("going\n");
    unsigned char recvChar;
    for (;;) {
        int r = uart_poll_in(uart, &recvChar);
        if (r == 0) {
            printk("%c", recvChar);
        } else if (r == -1) {
            continue;
        } else {
            printk("\n\nERR %d\n\w", r);
        }
    }

}
```

this dts does not work:

```dts
/dts-v1/;

#include <espressif/esp32.dtsi>

/ {
	model = "esp32";
	compatible = "espressif,esp32";

	aliases {
		uart-0 = &uart0;
		i2c-0 = &i2c0;
	};

	chosen {
		zephyr,sram = &sram0;
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
		zephyr,flash = &flash0;
	};
};

&cpu0 {
	clock-frequency = <ESP32_CLK_CPU_240M>;
};

&cpu1 {
	clock-frequency = <ESP32_CLK_CPU_240M>;
};

&uart0 {
	status = "okay";
	current-speed = <115200>;
	tx-pin = <1>;
	rx-pin = <3>;
	//rts-pin = <22>;
	//cts-pin = <19>;
};
&uart1 {
	status = "okay";
	current-speed = <115200>;

	tx-pin = <17>;
	rx-pin = <16>;
};

```

but change uart2 to uart1 and it works fine

sylvioalves, 2021-05-12:
Hi @aep, are you using ESP32 WROOM or WROVER?
Could you describe the error? Building issues or  `UART_2` device not found?

aep, 2021-05-12:
Wroom. As described in the subject, running the posted code works fine, but poll will just never return a character. Same with the async api.

closed 2021-05-14 by nashif