zephyrproject-rtos/zephyr #35187: Version selection not working

rodrigobrochado, 2021-05-11:
**Describe the bug**

The Cmake `find_package` is not following `EXACT` [version selection](https://cmake.org/cmake/help/latest/command/find_package.html#version-selection). I've modified the CMakeLists.txt of some samples code to versions that don't exist yet (like 2.7.0) or old ones (like 2.4.0) and the build passed where I expected to fail.  

I've then removed the lines 51 to 66 below and it worked as described in the [Zephyr Cmake guide](https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#zephyr-cmake-package-version): https://github.com/zephyrproject-rtos/zephyr/blob/fb88c77ac2baa8f33c53ba1ad72393fadd87e124/share/zephyr-package/cmake/ZephyrConfigVersion.cmake#L51-L66 

**To Reproduce**
1. git checkout v2.6.0-rc1
2. git diff  
\# empty
3. west diff  
\# empty
4. unset ZEPHYR_BASE
5. edit ~/zephyrproject/zephyr/samples/hello_world/CMakeLists.txt, replacing [line 5](https://github.com/zephyrproject-rtos/zephyr/blob/929c573d31d9c4a09f92b0f40cd4fc0ae1c49c69/samples/hello_world/CMakeLists.txt#L5) with the following:  
`find_package(Zephyr 2.7.0 EXACT REQUIRED)`
6. cd ~/zephyrproject/zephyr/
7. west build --board nrf52840dk_nrf52840

**Expected behavior**
Cmake should through a Fatal Error. It builds normally instead.

**Impact**
This can cause accidental builds with improper zephyr features and undesirable consequences.

**Logs and console output**

```bash
$ git diff

diff --git a/samples/hello_world/CMakeLists.txt b/samples/hello_world/CMakeLists.txt
index 3e3d0e8149..897bd1e2a0 100644
--- a/samples/hello_world/CMakeLists.txt
+++ b/samples/hello_world/CMakeLists.txt
@@ -2,7 +2,7 @@
 
 cmake_minimum_required(VERSION 3.13.1)
 
-find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
+find_package(Zephyr 2.7.0 EXACT REQUIRED)
 project(hello_world)
 
 target_sources(app PRIVATE src/main.c)
```

Check `-- Zephyr version` bellow:  
```bash
$ west build --board nrf52840dk_nrf52840
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/rodrigo/github/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/rodrigo/github/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.6.0-rc1 (/home/rodrigo/github/zephyrproject/zephyr), build: v2.6.0-rc1
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.6") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /home/rodrigo/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.11.3 (/home/rodrigo/zephyr-sdk-0.11.3)
-- Found dtc: /home/rodrigo/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/rodrigo/github/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: /home/rodrigo/github/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/rodrigo/github/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/rodrigo/github/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/device_extern.h
Parsing /home/rodrigo/github/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/rodrigo/github/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '/home/rodrigo/github/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/rodrigo/github/zephyrproject/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/rodrigo/github/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/rodrigo/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rodrigo/Documents/c1/github/zephyrproject/zephyr/samples/hello_world/build
-- west build: building application
[1/141] Preparing syscall dependency handling

[134/141] Linking C executable zephyr/zephyr_prebuilt.elf

[141/141] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       17100 B         1 MB      1.63%
            SRAM:        5600 B       256 KB      2.14%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**Environment:**
 - OS: Linux (ubuntu 20.10)
 - Toolchain Zephyr SDK 0.11.3
 - Version 2.6.0-rc1, but same behavior on 2.4.0
- Cmake 3.16.3

**Additional context**
I still don't know much of Cmake or west. This behavior could be related to the fact that `zephyr.base` is always set by `west build` default [configurations](https://docs.zephyrproject.org/latest/guides/west/config.html#west-config-cmd), but if it's always set and following `ZephyrConfigVersion.cmake` logic, how would the version selection work?

tejlmand, 2021-05-18:
Thanks for raising this.
A little bit of history.

Before introducing Zephyr as CMake package, this is what we had (and still have on the current Zephyr 1.14 LTS):
https://github.com/zephyrproject-rtos/zephyr/blob/af56a6b45a3f2998bff5c983f27fb5b46e84b0fb/samples/hello_world/CMakeLists.txt#L5

and user would do `source zephyr-env.sh` to get ZEPHYR_BASE defined:
https://docs.zephyrproject.org/1.14.1/getting_started/index.html?highlight=zephyrrc#build-the-application

When `find_package(Zephyr)` was introduced, the need for having `ZEPHYR_BASE` in environment was removed, but still supported. And to ensure consistent behavior with old principle, then having `ZEPHYR_BASE` in environment would disable all version checking which is the reason for the code you linked.

So, if ZEPHYR_BASE is unset, the check does work, like:
```
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild samples/hello_world/
CMake Error at CMakeLists.txt:5 (find_package):
  Could not find a configuration file for package "Zephyr" that is compatible
  with requested version "5.4.0".

  The following configuration files were considered but not accepted:

    /projects/github/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake, version: 2.6.0
    /projects/bitbucket/sec_test/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake, version: 2.3.99
```

Now, to the second part that you also observed. `west` is specifying `ZEPHYR_BASE` before calling CMake, and the reason for that is to be able to build with current Zephyr LTS.

With `west` you are able to specify `west config zephyr.base-prefer env` and then leave this setting unset, that should ensure that Zephyr base is unset and CMake package check should work when building with `west`:
```
$ west config zephyr.base-prefer env
$ echo $ZEPHYR_BASE

$ west build -bnrf52840dk_nrf52840 -dbuild_version samples/hello_world/
-- west build: generating a build system
CMake Error at CMakeLists.txt:5 (find_package):
  Could not find a configuration file for package "Zephyr" that exactly
  matches requested version "2.7.0".

  The following configuration files were considered but not accepted:

    /projects/github/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake, version: 2.6.0
    /projects/bitbucket/sec_test/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake, version: 2.3.99
```

rodrigobrochado, 2021-05-19:
Thanks for the explanation @tejlmand!!

So, two questions:
1. We use a freestanding application and [it requires](https://docs.zephyrproject.org/latest/guides/west/troubleshooting.html#invalid-choice-build-or-flash-etc) ZEPHYR_BASE to be set in order to work (exporting the workspace was not enough for us); You mentioned that ZEPHYR_BASE presence disables version checking and [Zephyr Cmake guide](https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#zephyr-cmake-package-version) about package version is pointed mainly on freestanding applications.   
So, given all that, should we add a note to that part of the guide advertising that version checking won't work with a freestanding application built with `west` (even though it's a Cmake guide and it works with Cmake)? 

2. I couldn't make `zephyr.base-prefer` to work. After setting it to `env` and assuring that `ZEPHYR_BASE` is unset, `west built` didn't fail as you did. Any tips? I'm using west 0.8.0.

carlescufi, 2021-05-19:
@tejlmand will send a PR in time for 2.6.0

tejlmand, 2021-05-25:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/35643

PR #35643 (2021-05-25): cmake: check zephyr version if specified and ZEPHYR_BASE is set
Fixes: #35187

This extends the Zephyr package to also honor version when ZEPHYR_BASE
is set in environment.

Specifying `find_package(Zephyr 2.x.y)` without using ZEPHYR_BASE will
lookup a Zephyr in following order:
- Current repo, if that is a Zephyr repo
- Current west workspace
- Exported Zephyr CMake package for freestanding application
and ensure that the chosen Zephyr meets the version criteria.

When setting ZEPHYR_BASE in environment the version check is disabled
and the Zephyr referenced by ZEPHYR_BASE will always be use [... cut]
commit b574c4f6 (2021-05-25): cmake: check zephyr version if specified and ZEPHYR_BASE is set

Fixes: #35187

This extends the Zephyr package to also honor version when ZEPHYR_BASE
is set in environment.

Specifying `find_package(Zephyr 2.x.y)` without using ZEPHYR_BASE will
lookup a Zephyr in following order:
- Current repo, if that is a Zephyr repo
- Current west workspace
- Exported Zephyr CMake package for freestanding application
and ensure that the chosen Zephyr meets the version criteria.

When setting ZEPHYR_BASE in environment the version check is disabled
and the Zephyr referenced by ZEPHYR_BASE will always be us [... cut]
closed 2021-05-25 by galak