zephyrproject-rtos/zephyr #35197: Zephyr Project Development with 2 Ethernet Interfaces Supported (eth0, and eth1)

bob2oneil, 2021-05-12:
Is it possible to develop a Zephyr application that supports 2 Ethernet interfaces, that what would be 2 physical NICs on a hardware platform.

One NIC would be a standard interface accepting UDP packets with a standard 1500 byte MTU.

The other NIC would require a raw socket, with an MTU of over 8K (jumboframe).

The secondary NIC would be bridged with other secondary NICS to form a bridge between running instances of Zephyr, with the emulation environment being native_posix_64 and qemu_riscv64.

Is it possible to 2 have two NIC support in Zephyr for a single application?

jukkar, 2021-05-12:
Yes, you can have two or any number of Ethernet controllers in the system. For each network interface, there needs to be a call to `ETH_NET_DEVICE_INIT()`. See for example `drivers/ethernet/eth_native_posix.c` which does exactly this. In that driver user can set `CONFIG_ETH_NATIVE_POSIX_INTERFACE_COUNT` and the driver will create as many network interfaces to the system.

What traffic UDP etc, depends on your application, the Ethernet driver does not really care what data is being transferred.

bob2oneil, 2021-05-12:
Many thanks jukkar

closed 2021-05-12 by bob2oneil