zephyrproject-rtos/zephyr #35198: subsys.pm.device_pm: frdm_k64f leave idel fails

hakehuang, 2021-05-12:
**Describe the bug**
 Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/pm/power_mgmt/src/main.c:161: test_power_state_trans: leave_idle is false

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. See error

**Expected behavior**
case pass

**Impact**
pm features

**Logs and console output**

```
    *** Booting Zephyr OS build zephyr-v2.5.0-3761-g730acbd6ed85  ***

    Running test suite power_management_test

    ===================================================================

    START - test_power_idle

    give way to idle thread

     PASS - test_power_idle in 0.103 seconds

    ===================================================================

    START - test_power_state_trans

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/pm/power_mgmt/src/main.c:161: test_power_state_trans: leave_idle is false

    

     FAIL - test_power_state_trans in 0.112 seconds

    ===================================================================

    START - test_power_state_notification

     PASS - test_power_state_notification in 0.1 seconds

    ===================================================================

    Test suite power_management_test failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc1

hakehuang, 2021-05-12:
@MaureenHelm  ,@dleach02 , @mmahadevan108

galak, 2021-05-12:
@ceolin similar issue on how do we filter platforms that don't support PM.

dleach02, 2021-05-12:
@galak, is this filtering done on the YAML tags line (as well as the other exclude filters)?

ceolin, 2021-05-13:
@galak That is the same problem indeed. To be fair that is not an issue with this particular board, this test should be failing with others platforms too.

ceolin, 2021-05-13:
https://github.com/zephyrproject-rtos/zephyr/pull/35233

closed 2021-05-13 by ceolin