zephyrproject-rtos/zephyr #35203: Options when using CONFIG_THREAD_RUNTIME_STATS

alexsven, 2021-05-12:
**Is your feature request related to a problem? Please describe.**
As far as I can see, the CPU % printed when using CONFIG_THREAD_RUNTIME_STATS is based off of total number of cycles since then system booted. 

**Describe the solution you'd like**
I would like to have an option for seeing the percentage used per thread in the given log intervall, similar to what CONFIG_TRACING_CPU_STATS provided.

koffes, 2021-06-14:
I second this. The way it stands today, the information is not that valuable.

cfriedt, 2021-07-27:
@alexsven, @koffes - is this a feature either of you would be willing to submit a PR for?

koffes, 2021-08-02:
We would like to, but it has a low priority in our backlog.

koffes, 2023-02-10:
We would still like to have a look at this but we cannot make any promises on timeline, and we certainly welcome anyone to contribute.
We have had several people surprised by the results being "from boot". If e.g. running this for 30 mins, then a thread suddenly uses 50% CPU: This will take time to show in the logs with this current setting. I am surprised that more people are not impacted.

cfriedt, 2023-02-11:
Just to clarify - is this in reference to a shell command?

koffes, 2023-12-28:
No, this is in reference to what is printed if CONFIG_THREAD_RUNTIME_STATS is enabled.

koffes, 2024-01-16:
I did have a quick go at this, but the underlying APIs all fetch data since boot. This means that we either need to store values statically in the monitor module to calculate diffs, or create additional variables for time `t-1` in some kernel structs also to calculate diffs.